RfidResearchGroup/proxmark3 #1055: Legic standalone with non-RDV4 PM3

ikarus23, 2020-11-11:
Hi! The [Standalone mode](https://github.com/RfidResearchGroup/proxmark3/wiki/Standalone-mode) wiki page says that the `HF_LEGIC` mode is only compatible with RDV4. However, if I look at [the code](https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/Standalone/hf_legic.c), it looks like everything related to them RDV4 (the flash) is optional and marked as such (`#WITH_FLASH`). Therefore, it should work on devices like a Proxmark3 Easy, right? I only have a RDV4, so I cannot test it. Can someone else test it and update the wiki page if I'm right?

iceman1001, 2020-11-11:
the mode will run on a non-rdv4 but it will not function as intended.
It will read and simulate tags but it will never store.

We never wanted to have  a "PARTIAL" compliance in the compatible list since that will confuse ppl from what they expect.
Keeping things simple

ikarus23, 2020-11-11:
I understand, but still, it is unfortunate. This way people with a pm3 easy won't notice that they can use this mode if the want read/clone. I guess my perspective is a bit different. I don't think of it as partial working. I think of it more like "it is fully working" (read/clone), but "there is an extra with the RDV4" (store).

closed 2020-11-11 by ikarus23

iceman1001, 2020-11-11:
I do understand your take on it.  So either that or we start adding more text explain what ppl can expect and when they don't read , they will ask more questions ...

We can play with words saying,
- the full experience on RDV4  is read/sim/save
- the full experience on GENERIC is read/sim

I, for one, don't think that kind of wording is good.

Now users might start to expect the standalone modes to be maintained,  support all devices,  which we, the maintainers of the repo, have no intention at all to support.  That is the role of the developer of the standalone mode. 

As I see it,  for this particular standalone mode,  it was a bonus to make the stuff guarded to allow for GENERIC devices to use it not its intention.

_Its not like standalone modes needs to be more confusing as it is already._

I updated the wiki,  to clarity this ambigousity you pointed out,  but for the end user I don't think it became easier.
https://github.com/RfidResearchGroup/proxmark3/wiki/Standalone-mode

doegox, 2020-11-11:
What about sth like
[image]

iceman1001, 2020-11-11:
It exemplifies what I wrote before,  The first thing you gonna ask is "..partial" ? ... where do I find out what partial means.
So yes,  adding partial by itself isn't good enough.   The person who wants to use this mode,  now has to read the source code ,  where I would expect a clear documentation inside of it  otherwise its just even more frustrating. 

And yes, it is a tidbit more nicer like that.   Maybe a link to the source code,  where the definition of partial means for that standalone mode.

doegox, 2020-11-11:
ok I changed it, and also rephrase the desc
`Read/simulate Legic Prime tags (RDV4: + save to flashmem)`

iceman1001, 2020-11-11:
Do you think that it became more clear?

doegox, 2020-11-11:
yes, a bit, else I wouldn't have done it

ikarus23, 2020-11-11:
>The person who wants to use this mode, now has to read the source code

We have this problem already. The description is just very basic and short. The user does not know how the mode works exactly, what buttons to press and what the LEDs mean. There even is no info on the Standalone page on how to get into the standalone mode (long button press) or how to activate them (modifying the Makefile.platfrom). ;)

iceman1001, 2020-11-11:
Feel free to update that documentation

ikarus23, 2020-11-11:
Done :)