RfidResearchGroup/proxmark3 #1056: Endianness of UID for hf 15 dump

ikarus23, 2020-11-12:
Hi! I noticed that the endianness  of of the UID on `hf 15 dump` is not consistent with the file created. I'm not sure if this is just a simple inconsistency of if it might cause bugs for `hf 15 sim` or hf 15 restore`.

```
[usb] pm3 --> hf 15 dump

[+]  UID: E0 07 A4 00 00 BF 8D 4F
[+] TYPE: Texas Instrument France
[=] Using UID as filename
[+] Reading memory from tag UID E0 07 A4 00 00 BF 8D 4F
...........................................[-] ⛔ Tag returned Error 16: The specified block is not available (doesn't exist).

block#   | data         |lck| ascii
---------+--------------+---+----------
  0/0x00 | 89 17 3A C3  | 0 | ..:.

...

 42/0x2A | A0 40 58 8E  | 0 | .@X.

[+] saved 172 bytes to binary file hf-15-4F8DBF0000A407E0-dump-1.bin
[+] saved 43 blocks to text file hf-15-4F8DBF0000A407E0-dump-1.eml
[+] saved to json file hf-15-4F8DBF0000A407E0-dump-1.json

iceman1001, 2020-11-12:
A simple inconsistency.  
The generatefilename based on uid fct doesn't take in consideration of endianess,  it used the dump endianess.
Which for 15693 how we present UID.

iceman1001, 2020-11-12:
I pushed a fix,

closed 2020-11-12 by iceman1001

ikarus23, 2020-11-12:
Great! Thanks!

commit a28e5a88 (2020-11-12): fix #1056 - swapendianess for filenames based on uid