RfidResearchGroup/proxmark3 #1059: Do I Need To Update My Bootloader? If so how

Syax242, 2020-11-15:
I am a windows 10 user trying to use my Proxmark3 Easy. Previously I was having problems with the flash finding the proxmark but after rereading the instructions I realized I just needed to use the command:
client/proxmark3 COM5 --flash --unlock-bootloader --image bootrom.elf --image fullimage.elf

However after using this command I got the Following errors while flashing:

================ OBS ! ====================
Note: Your bootloader does not understand the new CMD_BL_VERSION command
It is recommended that you first updated your bootloader alone,
reboot the Proxmark3 then only update main the main firmware

Available memory on this board: UNKNOWN

================ OBS ! ====================
Note: Your bootloader does not understand the new CHIP_INFO command
It is recommended that you first updated your bootloader alone,
reboot the Proxmark3 then only update main the main firmware

Permitted flash range: 0x00100000-0x00140000
Loading ELF file

loading usable ELF segments:
      0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
      1: V 0x00200000 P 0x00100200 (0x00000d50->0x00000d50) [R X] @0x298

Loading ELF file

loading usable ELF segments:
      0: V 0x00102000 P 0x00102000 (0x000468a0->0x000468a0) [R X} @0x94
Error: PHDR is not containedd in Flash
Firmware probably too big for your device
All done

I'm not sure if I need to update the bootloader somehow or what to do to fix this issue but I've also attached a image of what happened for reference.

[image]

iceman1001, 2020-11-15:
I suggest you follow the instructions and flash bootrom first
``
./pm3-flash-bootrom
./pm3-flash-fullimage
```

doegox, 2020-11-15:
See also https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#256kb-versions

Syax242, 2020-11-15:
I actually followed those steps directly after posting and it worked but now its changed my proxmark to COM7 and whenever I try to use client/proxmark3 COM7 it says it cant communicate with the proxmark then disconnects it and when it reconnects it does so as a USB device for some reason.

[image]

[image]

iceman1001, 2020-11-15:
Use the scripts I posted about. 
Do the button-trick (press and hold),     and flash fullimage
and you should be fine

Syax242, 2020-11-15:
I'm note sure how to use @doegox link info but the method you posted flashed the image and showed that the available memory on my board is 512K bytes but its still doing the same thing of reconnecting as the unknown USB device.

doegox, 2020-11-15:
ok if it's 512k, just ignore my link, no need to shrink the fw

Syax242, 2020-11-16:
I'm still having the same issue of it reconnecting as a USB device instead of a COM after it has successfully flashed, any suggestions?

iceman1001, 2020-11-16:
Well,  we can conclude its not a fw/software issue but a user based issue since the compilatio / flashing works.
I will close this issue.  
If you want to ask user related questions I suggest you ask / read the forums or head over to the discord server.

closed 2020-11-16 by iceman1001