RfidResearchGroup/proxmark3 #1060: Bug prevents some cards from cloning correctly

dgood71, 2020-11-15:
**Describe the bug**
34 Bit HID card was cloned to  T5577 card, the resulting data print from the clone is different than the original and the cloned card does not work even though the facility code and card number are correct.  lf Search shows both the original and clone cards as identical, but data print x reveals that the clone is indeed not correct.

**To Reproduce**

Clone a card using this command:
lf hid clone -w N10002 --fc 193 --cn 45193
lf sea
card will appear to be correct; the fc and cn are shown properly.
However, data print x will show:
[+] DemodBuffer: 1D5559655556955A69565659
and the original card's data print x shows this: 1D5559655556955A6956565A
The last character in the hex string is different; the 9 should be an A.
The clone is NOT identical to the original card and does not work.

**Expected behavior**
The data print x from the clone should be the same as the original card : 1D5559655556955A6956565A

**Screenshots**
lf search on the original HID card:
[=] Checking for known tags...
[=]
[+] [N10002] - HID N10002 34-bit;  FC: 193  CN: 45193
[+] [Optus34] - Indala Optus 34-bit;  FC: 137  CN: 193
[+] [Smartpass] - Cardkey Smartpass 34-bit;  FC: 24  CN: 45193  Issue 1

[usb] pm3 --> data print x
[+] DemodBuffer: 1D5559655556955A6956565A

And on my non-working clone, here's what I see:
[=] Checking for known tags...
[=]
[+] [N10002] - HID N10002 34-bit;  FC: 193  CN: 45193
[+] [Optus34] - Indala Optus 34-bit;  FC: 137  CN: 193
[+] [Smartpass] - Cardkey Smartpass 34-bit;  FC: 24  CN: 45193  Issue 1

[usb] pm3 --> data print x
[+] DemodBuffer: 1D5559655556955A69565659

**Desktop (please complete the following information):**
Using Ubuntu OS
 [CLIENT ]
  client: RRG/Iceman/master/v4.9237-1961-gf63cf021-dirty...-unclean 2020-11-06 22:55:51
  compiled with GCC 10.2.0 OS:Linux ARCH:x86_64

[ PROXMARK3 ]
  firmware.................. PM3OTHER

[ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-1961-gf63cf021-dirty...-unclean 2020-11-06 22:56:32
       os: RRG/Iceman/master/v4.9237-1961-gf63cf021-dirty...-unclean 2020-11-06 22:56:51
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

[ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

**Additional context**
See this link in the Proxmark [forums:](http://www.proxmark.org/forum/viewtopic.php?pid=40260#p40260) for complete info.
Iceman's response in the forum regarding this issue:
there seems to be a bug in the wiegand decoding part, where it have to detect the sentinel bit vs length of data.

iceman1001, 2020-11-15:
Thanks for reporting it.
Its a known bug,   I have been looking at it since two weeks.

iceman1001, 2021-04-27:
Having looked into N10002,   it seem to be a confusing on the internet how this format works.
I found two different definitions,
and to top it off,  I also found a bug in the source that affects this one.

iceman1001, 2021-04-27:
@dgood71   do you have more sample credentials to share?

dgood71, 2021-04-27:
I don't, at least right now.  I'll have to see whether I can get my hands
on any other cards that have the same issue.  Some cards work just fine,
but I have a very limited sample to choose from and this is the only one
I've found to date that has this issue.

iceman1001, 2021-04-28:
Interesting,  then this make the need for more samples even more important.
Need to figure out how N10002 actually does its PACS data and verify it against other findings.

dgood71, 2021-04-28:
When I said "some cards work just fine" I wasn't referring specifically to
N10002 cards - just RFID cards in general.  Sorry if I caused confusion
with that statement.  I'll have to see whether any of the other cards I
have access to are N10002 cards, and if so get samples from them
specifically.

iceman1001, 2021-04-28:
ok,  since it will be hard to hunt a bug with weak debug information and confusing answers,  I will close this issue.
Once you find something,  document it properly and open a new issue.

closed 2021-04-28 by iceman1001