RfidResearchGroup/proxmark3 #1085: Hardnested segfaults on authentication errors

socram8888, 2020-12-03:
While running the hardnested against some (probably) original Mifare Plus 1K 4-byte NUID cards, if there's any kind of authentication error, there's a high chance that the attack will fail with:
 - Leaving the attack with no result message
 - Time and states going NaN
 - Catastrophically giving a segmentation fault.

Hw version:
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-2195-ga03240a0d 2020-12-03 00:25:44
  compiled with GCC 8.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3OTHER

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-2195-ga03240a0d 2020-12-03 00:26:22
       os: RRG/Iceman/master/v4.9237-2195-ga03240a0d-dirt...-unclean 2020-12-03 02:44:14
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 270742 bytes (52%) Free: 253546 bytes (48%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```
Hw status:
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............44020
[#]   Available memory........41712
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............0
[#]   dma8 memory.............-2110148
[#]   dma16 memory............-2110148
[#]   toSend memory...........41712
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0
[#] LF Sampling Stack
[#]   Max stack usage.........4880 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[... 209 lines cut ...]
      23 |     444 | Apply bit flip properties                               |               0 |    0s
[#] AcquireNonces: Auth1 error
      24 |     556 | Apply bit flip properties                               |               0 |    0s
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth2 error len=1
[#] AcquireNonces: Auth1 error
      25 |     668 | Apply bit flip properties                               |               0 |    0s
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
      26 |     779 | Apply bit flip properties                               |               0 |    0s
[#] AcquireNonces: Auth1 error
      27 |     891 | Apply bit flip properties                               |               0 |    0s
[#] AcquireNonces: Auth2 error len=1
      28 |    1003 | Apply bit flip properties                               |               0 |    0s
      29 |    1113 | Apply bit flip properties                               |               0 |    0s
      30 |    1225 | Apply bit flip properties                               |               0 |    0s
[#] AcquireNonces: Can't select card (UID)
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
      32 |    1335 | Apply Sum property. Sum(a0) = 96                        |               0 |    0s
      32 |    1335 | (Ignoring Sum(a8) properties)                           |               0 |    0s
Segmentation fault (core dumped)
```

The hardware does not seem to blame, as by repeatedly running the attack, it got each and every key. Also interesting is that, between a failed and a successful attack, the position of the tag itself didn't change, so it is probably some software-related timing issue that is not being gracefully handled.

iceman1001, 2020-12-03:
Interesting, 
run `gdb` and see if there is a memory leak leading to the segmentation fault

socram8888, 2020-12-03:
Running with GDB and debug on the same card:
```
[usb] pm3 --> hf mf hardnested 8 a 749934CC8ED3 0 a s
[=] Target block no:  0, target key type:A, known target key: 0x000000000000 (not set)
[=] File action: none, Slow: Yes, Tests: 0
[+] Using AVX2 SIMD core.
[New Thread 0x7ffffda90700 (LWP 23329)]
[New Thread 0x7ffff7ff0700 (LWP 23330)]
[New Thread 0x7ffffd280700 (LWP 23331)]
[New Thread 0x7ffffca70700 (LWP 23332)]
[Thread 0x7ffffda90700 (LWP 23329) exited]
[Thread 0x7ffff7ff0700 (LWP 23330) exited]
[Thread 0x7ffffca70700 (LWP 23332) exited]

[Thread 0x7ffffd280700 (LWP 23331) exited]

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 4 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 132 million (2^27.0) keys/s      | 140737488355328 |   12d
      21 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   12d
[#] AcquireNonces: Auth1 error
[New Thread 0x7ffffca70700 (LWP 23333)]
[New Thread 0x7ffffd280700 (LWP 23334)]
[New Thread 0x7ffff7ff0700 (LWP 23335)]
[New Thread 0x7ffffda90700 (LWP 23336)]
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[Thread 0x7ffffca70700 (LWP 23333) exited]
[Thread 0x7ffffda90700 (LWP 23336) exited]
[Thread 0x7ffffd280700 (LWP 23334) exited]
      32 |     112 | Apply bit flip properties                               |   2847574327296 |    6h
[Thread 0x7ffff7ff0700 (LWP 23335) exited]
[New Thread 0x7ffffda90700 (LWP 23337)]
[New Thread 0x7ffff7ff0700 (LWP 23338)]
[New Thread 0x7ffffd280700 (LWP 23339)]
[New Thread 0x7ffffca70700 (LWP 23340)]
[#] AcquireNonces: Auth2 error len=1
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth2 error len=1
[Thread 0x7ffffca70700 (LWP 23340) exited]
[Thread 0x7ffffda90700 (LWP 23337) exited]
[Thread 0x7ffffd280700 (LWP 23339) exited]
      33 |     223 | Apply bit flip properties                               |   2248744894464 |    5h
[Thread 0x7ffff7ff0700 (LWP 23338) exited]
[New Thread 0x7ffffca70700 (LWP 23341)]
[New Thread 0x7ffffd280700 (LWP 23342)]
[New Thread 0x7ffff7ff0700 (LWP 23343)]
[New Thread 0x7ffffda90700 (LWP 23344)]
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth2 error len=1
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[Thread 0x7ffff7ff0700 (LWP 23343) exited]
[Thread 0x7ffffca70700 (LWP 23341) exited]
[Thread 0x7ffffda90700 (LWP 23344) exited]
      34 |     334 | Apply bit flip properties                               |   2095387901952 |    4h
[Thread 0x7ffffd280700 (LWP 23342) exited]
[New Thread 0x7ffffda90700 (LWP 23345)]
[New Thread 0x7ffff7ff0700 (LWP 23346)]
[New Thread 0x7ffffd280700 (LWP 23347)]
[New Thread 0x7ffffca70700 (LWP 23348)]
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
[... 207 lines cut ...]
      55 |    2311 | Apply bit flip properties                               |               0 |    0s
[Thread 0x7ffff7ff0700 (LWP 23415) exited]
[New Thread 0x7ffffda90700 (LWP 23417)]
[New Thread 0x7ffff7ff0700 (LWP 23418)]
[New Thread 0x7ffffd280700 (LWP 23419)]
[New Thread 0x7ffffca70700 (LWP 23420)]
[Thread 0x7ffffca70700 (LWP 23420) exited]
[Thread 0x7ffffd280700 (LWP 23419) exited]
[Thread 0x7ffff7ff0700 (LWP 23418) exited]
      56 |    2419 | Apply bit flip properties                               |               0 |    0s
[Thread 0x7ffffda90700 (LWP 23417) exited]
[New Thread 0x7ffffca70700 (LWP 23421)]
[New Thread 0x7ffffd280700 (LWP 23422)]
[New Thread 0x7ffff7ff0700 (LWP 23423)]
[New Thread 0x7ffffda90700 (LWP 23424)]
[Thread 0x7ffffca70700 (LWP 23421) exited]
[Thread 0x7ffffd280700 (LWP 23422) exited]
[Thread 0x7ffff7ff0700 (LWP 23423) exited]
      57 |    2527 | Apply bit flip properties                               |               0 |    0s
[Thread 0x7ffffda90700 (LWP 23424) exited]

Thread 1 "proxmark3" received signal SIGSEGV, Segmentation fault.
0x0000000008189856 in count_bitarray_AND_AVX2 (A=0x2f5af080, B=0xe) at hardnested_bitarray_core.c:212
212             A[i] &= B[i];
```

socram8888, 2021-01-22:
I've stumbled upon this bug again with another tag. Pinging @pwpiwi as per @iceman1001's recommendation. Using:
 - [This nonce dump](https://github.com/RfidResearchGroup/proxmark3/files/5858698/hf-mf-02520397-nonces_8_b_crashed.zip) (inside a ZIP because GitHub refuses to let me upload a plain .bin file)
 - The command `hf mf hardnested r f hf-mf-02520397-nonces_8_b_crashed.bin`

The result seems to be vary between yielding NaNs and crashing:
[image]
[image]

iceman1001, 2021-01-22:
AVX2...

On my WSL-1 , Ubuntu 20.04,   I get the NaN but it doesn't crash.
[image]

socram8888, 2021-01-22:
For the record, this issue seems to also affect [azcid's original bitslice implementation](https://github.com/aczid/crypto1_bs.git) from which I think Proxmark's is based. Running that program with this trace causes also a segmentation fault:
[image]

socram8888, 2021-01-22:
[Nonce pack](https://github.com/RfidResearchGroup/proxmark3/files/5859092/hf-mf-02520397-nonces_8_b.zip)

Here's one ZIP with three set of nonces. One succeeds and yields the correct password, other fails "gracefully" with NaNs. The third one, **if ran after either of the first two**, will yield crash 100% of the times.

Gator96100, 2021-01-24:
I was a little bored and did some digging. The crash is caused by a wrong `VECTOR_SIZE` at hardnested_bf_core.c:207 where `bs_ones.bytes` only has 8 elements, but `VECTOR_SIZE` is 16. When fixing this hardnested will go into NaN. This is what @iceman1001 also noticed. 

The reason why it goes into NaN is that the every ` states_bitarray` is empty, but I have very limited knowledge on how hardnested works. My understanding is that this bug is caused by a faulty nonces collection. Maybe caused by `VECTOR_SIZE`? 
@socram8888 is would be interesting if this issue happens on a arm device, as it would force skip all CPU optimizations.

pwpiwi, 2021-01-24:
Not sure if I can help here - I cannot confirm the issue on Official Repo:

```
proxmark3> hf mf hard r f hf-mf-02520397-nonces_8_b_crashed.bin
--target block no:  0, target key type:A, known target key: 0x000000000000 (not set), file action: read, Slow: No, Tests: 0
Using AVX2 SIMD core.

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 832 million (2^29.6) keys/s      | 140737488355328 |    2d
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    2d
       3 |       0 | Reading nonces from file nonces.bin...                  | 140737488355328 |    2d
       9 |    2576 | (1. guess: Sum(a8) = 256)                               |     43343982592 |   52s
      11 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |     22471231488 |   27s
      11 |    2576 | Starting brute force...                                 |     43343982592 |   52s
      13 |    2576 | (2. guess: Sum(a8) = 224)                               |    168287371264 |  3min
      15 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |    162218934272 |  3min
      15 |    2576 | Starting brute force...                                 |    168287371264 |  3min
      16 |    2576 | (3. guess: Sum(a8) = 192)                               |    266080583680 |  5min
      19 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |    226682142720 |  5min
      19 |    2576 | Starting brute force...                                 |    266080583680 |  5min
      21 |    2576 | Brute force phase completed. Key found: 26940b21ff5d    |               0 |    0s
proxmark3> hf mf hard r f hf-mf-02520397-nonces_8_b_failed.bin
--target block no:  0, target key type:A, known target key: 0x000000000000 (not set), file action: read, Slow: No, Tests: 0
Using AVX2 SIMD core.

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 682 million (2^29.3) keys/s      | 140737488355328 |    2d
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    2d
       3 |       0 | Reading nonces from file nonces.bin...                  | 140737488355328 |    2d
       9 |    2576 | (1. guess: Sum(a8) = 256)                               |     96781238272 |  2min
      10 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |     75909160960 |  2min
      12 |    2576 | (2. guess: Sum(a8) = 224)                               |    574836310016 | 14min
      14 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |    568768790528 | 14min
      14 |    2576 | Starting brute force...                                 |    574836310016 | 14min
      16 |    2576 | (3. guess: Sum(a8) = 192)                               |    934388105216 | 23min
      19 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |    894989238272 | 22min
      20 |    2576 | Brute force phase completed. Key found: 26940b21ff5d    |               0 |    0s
proxmark3> hf mf hard r f hf-mf-02520397-nonces_8_b_success.bin
--target block no:  0, target key type:A, known target key: 0x000000000000 (not set), file action: read, Slow: No, Tests: 0
Using AVX2 SIMD core.

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 738 million (2^29.5) keys/s      | 140737488355328 |    2d
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    2d
       3 |       0 | Reading nonces from file nonces.bin...                  | 140737488355328 |    2d
       9 |    2576 | (1. guess: Sum(a8) = 256)                               |    185844023296 |  4min
      11 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |    164971036672 |  4min
      11 |    2576 | Starting brute force...                                 |    185844023296 |  4min
      13 |    2576 | (2. guess: Sum(a8) = 224)                               |   1252417994752 | 28min
      15 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |   1246351982592 | 28min
      16 |    2576 | (3. guess: Sum(a8) = 192)                               |   2048233570304 | 46min
      19 |    2576 | Apply Sum(a8) and all bytes bitflip properties          |   2008834244608 | 45min
      21 |    2576 | Brute force phase completed. Key found: 26940b21ff5d    |               0 |    0s
```

@Gator96100:  Nope. `bs_ones.bytes` has `MAX_BITSLICES / 8` elements. And we have `#define VECTOR_SIZE (MAX_BITSLICES/8)`. Therefore no issue at hardnested_bf_core.c:234

iceman1001, 2021-01-24:
hw Are you running a other version of the offical client?  I don't see a "f" param in hardnested.   Seem to be printing "nonces.bin"  too. 

One my WSL-1 / Ubuntu 20.02,  GCC 9.3.0  with latest official repo I get NaN  for the failed. 

```
iceman@TAU:~/offical$ uname -a
Linux TAU 4.4.0-19041-Microsoft #488-Microsoft Mon Sep 01 13:43:00 PST 2020 x86_64 x86_64 x86_64 GNU/Linux

iceman@TAU:~/offical$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

iceman@TAU:~/offical$ git status
On branch master
Your branch is up to date with 'origin/master'.

iceman@TAU:~/offical$ cp ../pm3_rrg/hf-mf-02520397-nonces_8_b_failed.bin nonces.bin
iceman@TAU:~/offical$ ./client/proxmark3 /dev/ttyS3
proxmark3> hf mf hard r
--target block no:  0, target key type:A, known target key: 0x000000000000 (not set), file action: read, Slow: No, Tests: 0
Using AVX2 SIMD core.

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 2692 million (2^31.3) keys/s     | 140737488355328 |   15h
       2 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   15h
       4 |       0 | Reading nonces from file nonces.bin...                  | 140737488355328 |   15h
       4 |    2352 | Read 2352 nonces from file. cuid=02520397               | 140737488355328 |   15h
      10 |    2352 | (1. guess: Sum(a8) = 256)                               |       172903488 |    0s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |       172903488 |    0s
      10 |    2352 | (2. guess: Sum(a8) = 224)                               |       662762368 |    0s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |       660299072 |    0s
      10 |    2352 | (3. guess: Sum(a8) = 192)                               |      1602485504 |    1s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |      1598877696 |    1s
      10 |    2352 | Starting brute force...                                 |      1602485504 |    1s
      10 |    2352 | (4. guess: Sum(a8) = 160)                               |      6465369600 |    2s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |      4884802048 |    2s
      10 |    2352 | (5. guess: Sum(a8) = 176)                               |      9887611904 |    4s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |      9791420416 |    4s
      10 |    2352 | (6. guess: Sum(a8) = 128)                               |     13637745664 |    5s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |      1706415872 |    1s
      10 |    2352 | (7. guess: Sum(a8) = 144)                               |      2447330560 |    1s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |       774127744 |    0s
      10 |    2352 | Starting brute force...                                 |      2447330560 |    1s
      10 |    2352 | (8. guess: Sum(a8) = 200)                               |      1332744064 |    0s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |      1328216576 |    0s
      10 |    2352 | (9. guess: Sum(a8) = 136)                               |      2176605184 |    1s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |       336858208 |    0s
      10 |    2352 | (10. guess: Sum(a8) = 152)                              |       708820160 |    0s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |       497411904 |    0s
      10 |    2352 | (11. guess: Sum(a8) = 120)                              |      2315717376 |    1s
      10 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |       476113632 |    0s
      10 |    2352 | Starting brute force...                                 |      2315717376 |    1s
      10 |    2352 | (12. guess: Sum(a8) = 112)                              |      1853415040 |    1s
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |       180533184 |    0s
      11 |    2352 | (13. guess: Sum(a8) = 96)                               |      1654045440 |    1s
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |        73756920 |    0s
      11 |    2352 | (14. guess: Sum(a8) = 104)                              |       217532320 |    0s
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |         6125606 |    0s
      11 |    2352 | (15. guess: Sum(a8) = 80)                               |       163280608 |    0s
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |        67121448 |    0s
      11 |    2352 | (16. guess: Sum(a8) = 64)                               |       549649344 |    0s
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |            5084 |    0s
      11 |    2352 | (17. guess: Sum(a8) = 56)                               |         5270172 |    0s
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |          736812 |    0s
      11 |    2352 | (18. guess: Sum(a8) = 32)                               |        95058088 |    0s
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |               0 |    0s
      11 |    2352 | (19. guess: Sum(a8) = 0)                                |            -nan | -nand
      11 |    2352 | Apply Sum(a8) and all bytes bitflip properties          |            -nan | -nand
proxmark3>
```

Gator96100, 2021-01-24:
@pwpiwi You are correct that `bs_ones.bytes` and `VECTOR_SIZE` have the same size, it was a visual bug that let me think they are not the same size. However, it does have a segmentation fault at hardnested_bf_core.c:234.

iceman1001, 2021-01-24:
@pwpiwi   feel free to join the discord server,   https://discord.gg/8yYTKGVf

pwpiwi, 2021-01-25:
Indeed, f option doesn't exist on official repo and is silently ignored. After copying to nonces.bin I can confirm the error on official repo. It is some time ago since I had looked into the code but I will have a look...

I am anonymously and "read only" on discord.

iceman1001, 2021-01-25:
I can imagine the sheer amount of ppl trying to get hold of you,  but so far the discord server has been cordial to all the famous ones who are there.  And still,  we all love to see you active again :)

doegox, 2021-01-25:
So the error is not on the memcpy line but the alloc just before.
`p->len[EVEN_STATE] becomes = 0` and the `bucket_size` alloc on the stack becomes for too large

doegox, 2021-01-25:
if in `brute_force_bs` I check for len too, it doesn't crash anymore
```
        if (p->states[ODD_STATE] != NULL && p->states[EVEN_STATE] != NULL &&
            p->len[ODD_STATE] != 0 && p->len[EVEN_STATE] != 0) {
```
but warning
[image]

Having defined states pointers while len is zero is weird...

iceman1001, 2021-01-25:
when I add a breakpoint there and since I am manually stepping,   the failed nonce file acts different.  Its almost like a race condition...

doegox, 2021-01-25:
the failed doesn't have these len=0, only the crashed

iceman1001, 2021-01-25:
...is that  p->  pointer thread safe??   I get a sneaky feeling the crash is threads based.

iceman1001, 2021-01-25:
the len = 0 patch,  I am wrong,  Its not thread concurrency  its with the candidate generation.  
And with that I give up.

doegox, 2021-01-25:
@socram8888 please try latest. It should solve the crash at least.

socram8888, 2021-01-25:
@doegox indeed, the crash has been solved.

@pwpiwi I see you managed to crack all three files. If you were running an older version, do you know from when or the specific commit? We could try diffing the hardnested files to find if there's any regression.

doegox, 2021-01-26:
> @pwpiwi I see you managed to crack all three files. If you were running an older version, do you know from when or the specific commit? We could try diffing the hardnested files to find if there's any regression.

No, he confirmed  the error is on official repo too (https://github.com/RfidResearchGroup/proxmark3/issues/1085#issuecomment-766735189)

socram8888, 2021-01-26:
> > @pwpiwi I see you managed to crack all three files. If you were running an older version, do you know from when or the specific commit? We could try diffing the hardnested files to find if there's any regression.
> 
> No, he confirmed the error is on official repo too ([#1085 (comment)](https://github.com/RfidResearchGroup/proxmark3/issues/1085#issuecomment-766735189))

Not sure, in https://github.com/RfidResearchGroup/proxmark3/issues/1085#issuecomment-766408559 he did manage to crack them

pwpiwi, 2021-01-26:
@socram8888: all three nonce files in your nonce pack belong to the same card with key bb1502c29e80 ?

socram8888, 2021-01-26:
> @socram8888: all three nonce files in your nonce pack belong to the same card with key bb1502c29e80 ?

Yes, same card and same sector. It's a public transport card, a Mifare Plus in SL1.

iceman1001, 2021-07-15:
@socram8888   some news?  or can we close this one as _not resolved_

socram8888, 2021-07-15:
Can't say anything, I've not tried this attack since last time where it was broken. Unless somebody has re-checked the atack's code, it should be still broken.

iceman1001, 2021-07-15:
ok, closing because of no progress.

closed 2021-07-15 by iceman1001