RfidResearchGroup/proxmark3 #1100: Libjansson4 failure on 'make all'

SolomonGilbert, 2020-12-14:
Hi all,

Sadly having trouble compiling proxmark3 client (git pull yesterday 13th Dec 2020). I'm running Ubuntu 20.04, with libjansson4 version 2.12-1build1. Everything going smoothly, 'make accessrights' worked just fine, and - not that it's relevant - all comms work perfectly with the proxmark on /dev/ttyACM0. The official repo for the proxmark also works magnificently, managed to compile that just fine.

When compiling, the compiler gets tetchy at src/jansson_path.c, with errors described in screenshot:

[image]

Here's what the compiler's working with:

Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library found
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.8 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

All the pre-requisites are installed and up-to-date as per instructions, I've double checked to ensure they're coming from the right sources (had kali repo installed on here) and everything is indeed coming from official Ubuntu repo.

Followed the installation and compilation instructions to a tee, double checked everything, and I run other stuff reliant on libjansson4, which is a pretty innocuous library really. I'd have expected there to be more Qt5 problems!

I can't spot the problem and would definitely appreciate all the help you can offer. I hope I've provided enough initial information, I've looked online and I seem to be the first one experiencing this specific issue. Asked about with other proxmark users and nobody has experienced it either, with one of them compiling for Raspian the other day.

Let me know what else you need from me, and keep up the good work.

Solomon xx

iceman1001, 2020-12-14:
Interesting that it detects you have a system install of libjansson already but it still tries to compile the included libjansson inside client/deps..

```
Jansson library: system library found 
``` 

Have you installed the "libjansson-dev"  on your system?  or did you just install the libjansson4 package ?

iceman1001, 2020-12-14:
hold on,   that jansson_path.c  isn't libjansson dep,  its customized pm3 client.
So your GCC 9.3 is having some issue.   Which flags was used?

SolomonGilbert, 2020-12-14:
Both libjansson-dev and libjansson4 installed. Didn't realise that jansson_path isn't libjansson (sorry!).

No flags used, just simple ol' make clean && make all.

iceman1001, 2020-12-14:
lets see, I can't trigger your fault,  no gcc 9.3.0.. I have only gcc 9.2.1.

with that said lets try stuff:

```  
Move this to be the first include to be load in  jansson_path.c
#include "jansson_path.h"
```` 

and include a ref inside  jansson_path.h
```
#include "jansson.h"
```

SolomonGilbert, 2020-12-14:
[image]
[image]

Alas not it seems.

iceman1001, 2020-12-14:
make clean before?

SolomonGilbert, 2020-12-14:
Yup, just tried again twice, still the same problem. I updated to 20.04 last night, and pulled the git shortly after. It may be that there's some gcc problem, so might be worth trying to roll back gcc to 9.2.1 and see whether I get the same error? Don't really want to do that though...

Doesn't really make sense to me. I'm not a huge c guy though so I'm in your hands on this one.

Thank you for the quick responses by the way - it's immensely appreciated as this may be relied on during an upcoming pentest soon.

iceman1001, 2020-12-14:
We have CI and build on several OS / compiler versions and it looks okay. 
We can't keep track of latest versions of compilers. :(

SolomonGilbert, 2020-12-14:
Yeah that's fair enough. I wouldn't expect you guys to!

Well let me give it a go on 9.2.1 and see what happens. If it works alright shall we call it closed?

iceman1001, 2020-12-14:
Let me know if it works on 9.2.1, 
but we keep this one since we will have to sort it out eventual

SolomonGilbert, 2020-12-14:
Can't find a reliable 9.2.1 source, but tried it on 8.4 and 7.5 and got the same results. Double checked as well. I honestly have no idea what the issue is. I'm starting to fear the worst:

"It's not you it's me"
"This has never happened to me before I swear"

You know; standard compiler issues.

doegox, 2020-12-14:
There is a docker config in the repo with Ubuntu 20.04 and gcc 9.3.0.
I tried and it works fine.
Maybe you can try it on your side, maybe it can help you tracking the issue you're having.
https://github.com/RfidResearchGroup/proxmark3/tree/master/docker/ubuntu

doegox, 2020-12-14:
Also if you can reproduce the issue with the original code (removing the modification in jansson_path.h) with `make V=1` to see the actual commands...

SolomonGilbert, 2020-12-14:
Well, building it on docker worked, nice! That clearly begs the question what the hell have I done to my poor forgiving laptop :')

Reproducing for you now.

cmolson, 2020-12-14:
I just wanted to add I am also running Ubuntu 20.04 with gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)  I have not had this issue.

Here is my output:

Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.8 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

I tried installing libjansson-dev, the build then picks it up as a system library found but there are no issues compiling still (make clean each time).

doegox, 2020-12-14:
Ok let's close this now. @SolomonGilbert you're welcome to ask for help on the Discord if you've still issues on your setup but apparently it's not a bug of this repo.

closed 2020-12-14 by doegox