RfidResearchGroup/proxmark3 #1117: eMRTD Checklist, issues and feature requests

aveao, 2020-12-17:
This issue is intended to be a meta progress tracker and discussion space for the `hf emrtd` commands.

# Checklist

- [x] ISO 14a/14b support
- [x] Non-BAC passport support (untested)
- [x] BAC passport support
- [x] `hf emrtd dump`
- [x] `hf emrtd info` that displays basic info (perhaps just based on EF_DG1, tested)
- [x] `hf emrtd dump` dumping more detailed data from the files by parsing them, such as extracting the JPG and the cert file
- [ ] `hf emrtd info` that displays extended info (EF_DG1, EF_DG2, EF_DG11, EF_DG12?)
- [ ] `hf emrtd info` that displays a GUI with extended info (EF_DG1, EF_DG2, EF_DG11, EF_DG12?)
- [ ] PACE support
- [ ] Bruteforce support (might be interesting to look at bruteforcing CAN when we add PACE support)
- [ ] Various vulnerability checks as suggested by doegox
- [ ] Cert verification as suggested by iceman
- [ ] Dumping EF_CardAccess when doing `hf emrtd dump`, as suggested by doegox
- [ ] Automatically changing document number to uppercase and checking for length of DOB/Expiry, as suggested by doegox
- [ ] Ability to `hf emrtd dump` to a different folder, as suggested by doegox
- [ ] Ability to view info from a dump folder, as suggested by doegox

# Test results

## Working

- Turkish IDs (14a)
- Turkish ePassports (14a, 14b)
- Swedish ePassports (14b)
- Danish ePassports (2020, 14b)
- Lithuanian ePassports (14a)
- United States ePassports (14a)
- United Kingdom ePassports (14a)
- Canada ePassports (14a)
- Ukranian ePassports (14a)
- Serbian ePassports (14a)

## Not working

- Phillipines ePassports (14b) -> Does not detect (coupling issue?)
- Old Danish ePassports  -> Does not detect (coupling issue?)
- Belgian ePassports (14b) -> Not stable due to increasing timeout after an incorrect MRZ input. Being worked on by doegox.

aveao, 2020-12-17:
Whoops, accidentally picked the wrong option that added the wrong labels and assignees, please feel free to correct those.

doegox, 2020-12-18:
Notes about passports not working at first sight:
* tag antenna can be at different places: front cover, back cover, second page with ID (page is thick), last page (page is thick)
* some passports require to be opened to unshield the antenna

nickthegreek, 2020-12-21:
Greek passports work as well. Awesome work!

How easy would it be to implement emulation of a dumped passport? We have this client for example who want us to test some passport scanning app and obviously they're not going to ship real passports via mail. So using proxamrk to emulate a real one would be a great feature.

doegox, 2021-01-08:
Just to keep track of it, there is an interesting bit of code here to be able to parse the PKCS#7 certificate with mbedtls
https://github.com/qemu/skiboot/commit/9e7a4b327050101ec629b4028c2f7a452d7d5c0c

iceman1001, 2021-04-23:
Pushed a fix for 14b,  which influences emrtd cmds,  I tested on my passport and it works.

zachary822, 2021-09-12:
This also works on Taiwan passports (14a).

team-orangeBlue, 2023-09-16:
I've checked 14a Russian passports (random ID, 0008-20). Works well

zhiyan114, 2024-04-08:
Confirmed China's ePassport system worked, though, it doesn't automatically generate a .jpg of the facial image for DG2.

iceman1001, 2024-04-08:
I guess the image is because the pm3 client tries to show both facial image and signature image.    

And we only have one image object in the client for now.

saltymango2619, 2024-12-20:
I can also confirm chinese passport works. Though I had to go out of my way to decode the EF_DG2.BIN file.

I used this: https://github.com/Fethbita/emrtd

AliBello, 2025-07-17:
Looks like new ePassports issued by the Netherlands don't work anymore since they changed manufacturers to IDEMIA (https://web.archive.org/web/20250323181414/https://www.idemia.com/press-release/netherlands-introduces-new-dutch-identity-documents-incorporating-lasink-helios-idemia-smart-identity-2024-06-11/). Likely now uses PACE.

iceman1001, 2026-09-10:
The `hf emrtd info`   has gotten some love last month,  PACE/CAN.   A python3/kviy app under `tools/epassport`  also exists. 

Closing.

closed 2026-09-10 by iceman1001