RfidResearchGroup/proxmark3 #1136: pm3-flash-all or manual result in "Waiting for Proxmark3 to appear on /dev/ttyACM0"

mercertom, 2020-12-27:
***Things to try before submitting bug report***
read the [troubleshooting guide](/doc/md/Installation_Instructions/Troubleshooting.md) I have read. It is of not help.

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.
This says "nothing to be done for ____" for everything.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to RRG/Iceman repo.

proxmark3 /dev/ttyACM0 --flash --unlock-bootloader --image bootrom.elf --image fullimage.elf

pm3-flash-bootrom gives same exact problem. Waiting for proxmark3 to appear on /dev/ttyACM0.

**Describe the bug**
I hold/press the button in different ways, and get 3 different not-normal white led states, at least 1 of which is the "ready to flash" state.

**To Reproduce**
Steps to reproduce the behavior:
1. pm3-flash-all or proxmark3 /dev/ttyACM0 --flash --unlock-bootloader --image bootrom.elf --image fullimage.elf
2. hold the button while plugging in your pm3
3. watch it go into 1 of 3 not-normal modes, at least 1 of which is the "flash me" mode.
4. Watch it wait for over 30s and see nothing on /dev/ttyACM0.

**Expected behavior**
see the pm3. flash it.

**Desktop (please complete the following information):**
 - Manjaro

iceman1001, 2020-12-27:
how about you write which steps you did before trying to flash?

mercertom, 2020-12-27:
I followed https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md right up to the point where it recommends

pm3-flash-bootrom

and, if that doesn't work,

proxmark3 /dev/ttyACM0 --flash --unlock-bootloader --image bootrom.elf --image fullimage.elf

It's a 256k easy. The guide really should not call setting makefile.platform "Advanced compilation parameters". Advanced sounds optional. It should be on the https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md Compilation Instructions page, and not its own page. Pretty much everyone has to set the "advanced" parameters to succeed.

iceman1001, 2020-12-27:
This repo focuses on the RDV4 and the instructions matches that.  
There seem not to be any software related issue here,  so I am closing.

closed 2020-12-27 by iceman1001

mercertom, 2020-12-27:
Well, do you support the old iceman1001 repo? On that repo, it says it's deprecated, and to use this one. I followed the setup guide step by step, and it's not able to detect a proxmark3.

iceman1001, 2020-12-27:
This is an open source project,  there are no _support_

iceman1001, 2020-12-27:
You can ask questions on the forum or the discord server and hope for the best.

mercertom, 2020-12-27:
The project claims to work for easy 256k pm3s. It doesn't. That's an issue. Yet you decided to close the issue because the project is focused on rdv4 non-easy pm3s. I don't understand. Where should a person with a pm3 256k easy turn for functional code? Why isn't this a valid issue? All 3 repos used to work (rrg, iceman1001, proxmark3). Something about changes to my system or changes to these 3 codebases has made it so none of the 3 repos work.

What is the discord server?

iceman1001, 2020-12-27:
Its all well documented how to make it work on generic devices.   You seem to have a problem reading the documentation and is therefore not a software related issue.   You seem to confuse three different repositories.   I recommend you take time to read the existing documentation.

mercertom, 2020-12-27:
> Its all well documented how to make it work on generic devices. You seem to have a problem reading the documentation

I literally followed the documentation step by step.

> You seem to confuse three different repositories.

Not sure what makes you think I'm confusing three different repos just because I mention them. Is there anything in particular that makes you think I'm confusing them?

> I recommend you take time to read the existing documentation.

I recommend you update the setup guide to not call makefile.platform "advanced". It should be marked "essential", and be part of the standard setup steps because it is essential for almost everyone, and certainly essential for all non-512k pm3s, and the iceman1001 repo documentation says I should use the rrg repo. Do you have any specific documentation that you think I missed? I've done the setup guide for linux step by step. I've read the troubleshooting.md, and it has nothing for me. I've googled far and wide, reading similar errors on this repo, other repos, proxmark forum, and dangerousthings. None of them were pertinent to my issue. Why do you think this isn't a software issue or an issue with the setup guide documentation? I can't imagine that it isn't either.

mercertom, 2020-12-27:
Also, what is the discord?

iceman1001, 2020-12-27:
You insist of not providing any logs of your steps, nor your setup,  and refuse to take advice.  
Nor do you seem to understand the given documentation and insists that its wrong.   You will now be blocked.