RfidResearchGroup/proxmark3 #1151: hf mf chk (and others) report wrong "OBS"

ikarus23, 2020-12-30:
When an factory formatted tag is checked for keys there is a wrong "OBS" message (last line). There were no unknown keys! I'm not sure where this message is all displayed. For example, there is this same wrong "OBS" message in `hf mf autopwn`.

On a side not: If "OBS" is referring to an unexpected error it should be spelled "oobs" (https://en.wikipedia.org/wiki/Linux_kernel_oops).
```
[usb] pm3 --> hf mf chk *1 ? d
[=] No key specified, trying default keys
[ 0] ffffffffffff
[ 1] 000000000000
[ 2] a0a1a2a3a4a5
[ 3] b0b1b2b3b4b5
[ 4] c0c1c2c3c4c5
[ 5] d0d1d2d3d4d5
[ 6] aabbccddeeff
[ 7] 1a2b3c4d5e6f
[ 8] 123456789abc
[ 9] 010203040506
[10] 123456abcdef
[11] abcdef123456
[12] 4d3a99c351dd
[13] 1a982c7e459a
[14] d3f7d3f7d3f7
[15] 714c5c886e97
[16] 587ee5f9350f
[17] a0478cc39091
[18] 533cb6c723f6
[19] 8fd0a4f256e9
[20] 0000014b5c31
[21] b578f38a5c61
[22] 96a301bce267
................................
[=] time in checkkeys 2 seconds

[=] testing to read key B...

[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 001 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 002 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 003 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 004 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 005 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 006 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 007 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 008 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 009 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 010 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 011 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 012 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 013 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 014 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] | 015 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] |-----|----------------|---|----------------|---|
[+] ( 0:Failed / 1:Success)
[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-2A459E43-key.bin
[=]  OBS! --> 0xFFFFFFFFFFFF <-- has been inserted for unknown keys.
```

iceman1001, 2020-12-30:
It is an informative line.  It is printed regardless of success since we don't keep track of number of found keys and how many keys that were to be found.     Where we can't readily detect number of sectors for every single MFC tag, The only reliable way is when user specify how many sectors they expect to find.    Which gave the decision to just print it every time,  which has been fine for all the years.

So there is _nothing_ wrong with the message per se that it is printed every time. 

however,  I do agree that It could be better phrased with explaining that it only is of interest when there is a **0** in the **res**  column.

ikarus23, 2020-12-30:
Oh, ok! Makes sense! I was just confused by the "OBS". So better phrasing is most welcome! Thanks!

iceman1001, 2020-12-30:
Lets see if @tcprst  can come up with a better sentence for us to express the message we are trying to express.

tcprst, 2020-12-30:
I always thought it should be FYI as in For Your Information.

Something like: `FYI! --> 0xFFFFFFFFFFFF <-- has been inserted for unknown keys where res is 0`

iceman1001, 2020-12-30:
... if you add yellow to "0xFFFFFF"  ,  res,  and 0,   and that will be a more informative message.

commit 037fd476 (2020-12-30): fix #1151 - improved feedback text
closed 2020-12-30 by iceman1001