RfidResearchGroup/proxmark3 #1163: FPC USART for BT add-on... absent

Einstein2150, 2021-01-11:
***Things to try before submitting bug report***
read the [troubleshooting guide](/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to RRG/Iceman repo.

-

**Describe the bug**
FPC USART for BT add-on is still absent after:
```
brew remove  proxmark
brew install --with-blueshark --HEAD rfidresearchgroup/proxmark3/proxmark3
```

A connection fails:

proxmark3 -p /dev/tty.PM3_RDV40-DevB 
[=] Session log /Users/a1121460/.proxmark3/logs/log_20210111.txt
[+] loaded from JSON file /Users/a1121460/.proxmark3/preferences.json
[=] Using UART port /dev/tty.PM3_RDV40-DevB
[!!] 🚨 ERROR: cannot communicate with the Proxmark

**To Reproduce**
Steps to reproduce the behavior:
```
brew remove  proxmark
brew install --with-blueshark --HEAD rfidresearchgroup/proxmark3/proxmark3
```

connect bluetooth

proxmark3 -p /dev/tty.PM3_RDV40-DevB 
[=] Session log /Users/a1121460/.proxmark3/logs/log_20210111.txt
[+] loaded from JSON file /Users/a1121460/.proxmark3/preferences.json
[=] Using UART port /dev/tty.PM3_RDV40-DevB
[!!] 🚨 ERROR: cannot communicate with the Proxmark

**Expected behavior**
BT Aden should be active

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS Big Sur
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/b60daea 2021-01-11 12:58:49
  compiled with Clang/LLVM Apple LLVM 12.0.0 (clang-1200.0.32.28) OS:OSX ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: RRG/Iceman/master/b60daea 2021-01-11 13:00:47
       os: RRG/Iceman/master/b60daea 2021-01-11 13:01:19
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 309738 bytes (59%) Free: 214550 bytes (41%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

 - hw status
usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............42360
[#]   Available memory........42360
[#] Tracing
[#]   tracing ................1
[#]   traceLen ...............0
[#]   dma8 memory.............-2111808
[#]   dma16 memory............-2111808
[#]   toSend memory...........-2111808
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD5690C23DF636E2A
[#] Smart card module (ISO 7816)
[#]   version.................v3.10
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........4096 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override....std    : follow standard 
[#]   [b] BCC override........std    : follow standard
[#]   [2] CL2 override........std    : follow standard
[#]   [3] CL3 override........std    : follow standard
[#]   [r] RATS override.......std    : follow standard 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......252928
[#]   Transfer Speed PM3 -> Client = 505856 bytes/s
[#] Various
[#]   Max stack usage.........4096 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............-1
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31395 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#]   Mifare..................920 keys
[#]   T55x7...................109 keys
[#]   iClass..................7 keys

 - data tune
[usb] pm3 --> data tune
[=] REMINDER: 'hw tune' doesn't actively tune your antennas, it's only informative
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 70.16 V - 125.00 kHz
[+] LF antenna: 32.26 V - 134.83 kHz
[+] LF optimal: 70.16 V - 125.00 kHz
[+] Approx. Q factor (*): 11.9 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12.2 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 47.75 V - 13.56 MHz
[+] Approx. Q factor (*): 8.3 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

No GUI in this build!

**Additional context**
Bluetooth connection is enabled and established from Settings-Menue in macOS

proxmark3 -p /dev/tty.PM3_RDV40-DevB 
[=] Session log /Users/a1121460/.proxmark3/logs/log_20210111.txt
[+] loaded from JSON file /Users/a1121460/.proxmark3/preferences.json
[=] Using UART port /dev/tty.PM3_RDV40-DevB
[!!] 🚨 ERROR: cannot communicate with the Proxmark

iceman1001, 2021-01-11:
.... Have you flashed the device?

tcprst, 2021-01-11:
I also see that a `brew upgrade proxmark3` is needed. With the recent changes to the formula, blueshark should be fixed and the GCC toolchain should be updated.

iceman1001, 2021-01-11:
if the toolchain is updated,  you might need uninstall and reinstall

Einstein2150, 2021-01-11:
> .... Have you flashed the device?

yes - pm3-flash-all

iceman1001, 2021-01-11:
what was the output from the compilation?   and the flashing?  
Since your device doesn't have the bt-addon fw on it ,  somewhere before is the cause...

Einstein2150, 2021-01-11:
I got it. A complete update of all brew-components and a reinstallation and refresh made it :-)

`a1121460@E112-MAC003 ~> brew remove  proxmark3
Uninstalling /usr/local/Cellar/proxmark3/HEAD-b60daea... (676 files, 41.0MB)
a1121460@E112-MAC003 ~> brew install --with-blueshark --HEAD rfidresearchgroup/proxmark3/proxmark3
==> Installing proxmark3 from rfidresearchgroup/proxmark3
==> Cloning https://github.com/RfidResearchGroup/proxmark3.git
Updating /Users/a1121460/Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at b60daea lf t55xx sniff - now uses cliparser
==> make clean
==> make all PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON BREW_PREFIX=/usr/local
==> make install PREFIX=/usr/local/Cellar/proxmark3/HEAD-b60daea PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON BREW_PREFIX=/usr/local
==> Install success!
==> The latest bootloader and firmware binaries are ready and waiting in the current homebrew Cellar within share/firmware.
🍺  /usr/local/Cellar/proxmark3/HEAD-b60daea: 676 files, 41MB, built in 3 minutes 10 seconds
a1121460@E112-MAC003 ~> pm3-flash-all
[=] Session log /Users/a1121460/.proxmark3/logs/log_20210111.txt
[+] loaded from JSON file /Users/a1121460/.proxmark3/preferences.json
[+] About to use the following files:
[+]    /usr/local/Cellar/proxmark3/HEAD-b60daea/bin/../share/proxmark3/firmware/bootrom.elf
[+]    /usr/local/Cellar/proxmark3/HEAD-b60daea/bin/../share/proxmark3/firmware/fullimage.elf
[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodemiceman1
 🕑  59 found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading ELF file /usr/local/Cellar/proxmark3/HEAD-b60daea/bin/../share/proxmark3/firmware/bootrom.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d50->0x00000d50) [R X] @0x298

[+] Loading ELF file /usr/local/Cellar/proxmark3/HEAD-b60daea/bin/../share/proxmark3/firmware/fullimage.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00102000 P 0x00102000 (0x000487a0->0x000487a0) [R X] @0x94
[+]    1: V 0x00200000 P 0x0014a7a0 (0x0000176c->0x0000176c) [RW ] @0x48834
[=] Note: Extending previous segment from 0x487a0 to 0x49f0c bytes

[+] Flashing...
[+] Writing segments for file: /usr/local/Cellar/proxmark3/HEAD-b60daea/bin/../share/proxmark3/firmware/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. OK
[+]  0x00100200..0x00100f4f [0xd50 / 7 blocks]
....... OK

[+] Writing segments for file: /usr/local/Cellar/proxmark3/HEAD-b60daea/bin/../share/proxmark3/firmware/fullimage.elf
[+]  0x00102000..0x0014bf0b [0x49f0c / 592 blocks]
...................................................................
        @@@  @@@@@@@ @@@@@@@@ @@@@@@@@@@   @@@@@@  @@@  @@@
        @@! !@@      @@!      @@! @@! @@! @@!  @@@ @@!@!@@@
        !!@ !@!      @!!!:!   @!! !!@ @!@ @!@!@!@! @!@@!!@!
        !!: :!!      !!:      !!:     !!: !!:  !!! !!:  !!!
        :    :: :: : : :: :::  :      :    :   : : ::    : 
        .    .. .. . . .. ...  .      .    .   . . ..    . 
...................................................................
....................................................................................... OK

[+] All done

Have a nice day!
a1121460@E112-MAC003 ~> pm3
[=] Session log /Users/a1121460/.proxmark3/logs/log_20210111.txt
[+] loaded from JSON file /Users/a1121460/.proxmark3/preferences.json
[=] Using UART port /dev/tty.usbmodemiceman1
[=] Communicating with PM3 over USB-CDC

  ██████╗ ███╗   ███╗█████╗ 
  ██╔══██╗████╗ ████║╚═══██╗
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║ ╚══██╗
  ██║     ██║ ╚═╝ ██║█████╔╝       Iceman ☕
  ╚═╝     ╚═╝     ╚═╝╚════╝    ❄️ bleeding edge

  https://github.com/rfidresearchgroup/proxmark3/

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/b60daea 2021-01-11 16:58:51
  compiled with Clang/LLVM Apple LLVM 12.0.0 (clang-1200.0.32.28) OS:OSX ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: RRG/Iceman/master/b60daea 2021-01-11 16:59:49
       os: RRG/Iceman/master/b60daea 2021-01-11 17:00:29
  compiled with GCC 10.2.1 20201103 (release)

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 311047 bytes (59%) Free: 213241 bytes (41%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

`

closed 2021-01-11 by Einstein2150