RfidResearchGroup/proxmark3 #1168: 'lf em 4x05 write' seems to write random data

xstff78, 2021-01-19:
I observed a strange behaviour when writing data to an EM4305 tag. It also occurs when no tag is attached to the proxmark. Therefore I assume it is a client issue.
```
 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-2866-g0e18443b0 2021-01-14 21:28:43
  compiled with MinGW-w64 10.2.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-2866-g0e18443b0 2021-01-14 21:30:23
       os: RRG/Iceman/master/v4.9237-2866-g0e18443b0 2021-01-14 21:31:10
  compiled with GCC 8.4.0
```

The command
```[usb] pm3 --> lf em 4x05 write -a 10 -d DEADBEEF```

is answered by
```[=] Writing address 10 data 01031222 using password 00000000```

This data (0x01031222) also shown when no tag is attached. If so, it is actually written and can be read from the tag. Sending the same write command again results in the same (wrong?) data.

After restarting the client and sending the DEADBEEF write command again, the confirmed and actually written data is
```[=] Writing address 10 data 01741D72 using password 00000000```
The confimed data changes on each restart of the client.

Can someone reproduce this? Any ideas?

iceman1001, 2021-01-19:
Thanks,
its a known bug introduced with the cli-parser changes.

iceman1001, 2021-01-27:
Try pulling latest and see if that sorted it out.

https://github.com/RfidResearchGroup/proxmark3/commit/fdcc4b741c0f6debb621fe7d981a8336dff8a50f

closed 2021-01-27 by iceman1001

xstff78, 2021-01-28:
I can confirm that the command now works as intended.

Thank you for the quick help and all your work on this great project! :)