RfidResearchGroup/proxmark3 #1173: raspberry pi not support '__builtin_cpu_init();'

menshiyun, 2021-01-26:
***Compilation problems***

```
pi@raspberrypi:~/proxmark3 $ make STANDALONE= PLATFORM=PM3GENERIC PLATFORM_SIZE=256 SKIP_HITAG=1 VERBOSE=1

===================================================================
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_SIZE:     256
Platform extras:   No extra selected
Included options:  LF EM4x50 EM4x70 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   No standalone mode selected
===================================================================
[*] MAKE client/all
===================================================================
Client platform:   Linux
GUI support:       QT not found, disabled
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 not found, disabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
compiler version:  gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
===================================================================
[*] MAKE deps/amiitool/libamiibo.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/cliparser/libcliparser.a
PLATFORM Linux
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/hardnested/libhardnested.a
[-] CC hardnested_bf_core.c
hardnested_bf_core.c: In function ‘GetSIMDInstr’:
hardnested_bf_core.c:562:5: error: implicit declaration of function ‘__builtin_cpu_init’; did you mean ‘__builtin_irint’? [-Werror=implicit-function-declaration]
     __builtin_cpu_init();
     ^~~~~~~~~~~~~~~~~~
     __builtin_irint
hardnested_bf_core.c:562:5: error: nested extern declaration of ‘__builtin_cpu_init’ [-Werror=nested-externs]
cc1: all warnings being treated as errors
make[2]: *** [../../../Makefile.host:80: obj/hardnested_bf_core.o] Error 1
make[1]: *** [Makefile:723: deps/hardnested/libhardnested.a] Error 2
make: *** [Makefile:137: client/all] Error 2
```

**Desktop (please complete the following information):**
 - OS: RaspiOS

iceman1001, 2021-01-26:
Pull latest and try if it resolved.

closed 2021-01-26 by iceman1001

menshiyun, 2021-01-26:
Compiled successfully

commit 05c59850 (2021-01-26): fix #1173  - added guards