RfidResearchGroup/proxmark3 #1175: Memory leak on "lf search"

socram8888, 2021-01-26:
Running a `lf search` command with no tag in latest master, having SANITIZE enabled causes an stack overflow:

```marcos@DESKTOP-3ABQRGH:/mnt/c/Users/Marcos/Documents/Programacion/proxmark3$ client/proxmark3 /dev/ttyS11
[=] Session log /home/marcos/.proxmark3/logs/log_20210126.txt
[+] loaded from JSON file /home/marcos/.proxmark3/preferences.json
[=] Using UART port /dev/ttyS11
[=] Communicating with PM3 over USB-CDC

  ██████╗ ███╗   ███╗█████╗
  ██╔══██╗████╗ ████║╚═══██╗
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║ ╚══██╗
  ██║     ██║ ╚═╝ ██║█████╔╝       Iceman ☕
  ╚═╝     ╚═╝     ╚═╝╚════╝    ❄️ bleeding edge

  https://github.com/rfidresearchgroup/proxmark3/

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-2920-gfad365860 2021-01-26 23:46:24
  compiled with GCC 8.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-2898-g2829e20d4 2021-01-22 14:12:38
       os: RRG/Iceman/master/v4.9237-2920-gfad365860 2021-01-26 23:47:01
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 276571 bytes (53%) Free: 247717 bytes (47%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

[usb] pm3 --> lf search

[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
=================================================================
==4095==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7fffd0ca6dbf at pc 0x7fc36b0ac92f bp 0x7fffd0ca6aa0 sp 0x7fffd0ca6a98
READ of size 1 at 0x7fffd0ca6dbf thread T0
    #0 0x7fc36b0ac92e in askdemod_ext ../common/lfdemod.c:1705
    #1 0x7fc36ad9119e in ASKbiphaseDemod src/cmddata.c:780
    #2 0x7fc36af31961 in demodFDXB src/cmdlffdxb.c:494
    #3 0x7fc36aeffe36 in CmdLFfind src/cmdlf.c:1486
    #4 0x7fc36afafee5 in CmdsParse src/cmdparser.c:273
    #5 0x7fc36af00386 in CmdLF src/cmdlf.c:1605
    #6 0x7fc36afafee5 in CmdsParse src/cmdparser.c:273
    #7 0x7fc36afae4f4 in CommandReceived src/cmdmain.c:322
    #8 0x7fc36b0678bf in main_loop src/proxmark3.c:422
    #9 0x7fc36b06aed0 in main src/proxmark3.c:1074
    #10 0x7fc368ee409a in __libc_start_main ../csu/libc-start.c:308
    #11 0x7fc36ad7fc69 in _start (/mnt/c/Users/Marcos/Documents/Programacion/proxmark3/client/proxmark3+0x1b9c69)

Address 0x7fffd0ca6dbf is located in stack of thread T0 at offset 351 in frame
    #0 0x7fc36ad90fbc in ASKbiphaseDemod src/cmddata.c:769

  This frame has 6 object(s):
    [32, 36) 'startIdx'
    [96, 100) 'offset'
    [160, 164) 'clk'
    [224, 228) 'invert'
    [288, 296) 'size'
    [352, 131424) 'BitStream' <== Memory access at offset 351 underflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism or swapcontext
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow ../common/lfdemod.c:1705 in askdemod_ext
Shadow bytes around the buggy address:
  0x10007a18cd60: 00 00 f1 f1 f1 f1 01 f2 f2 f2 f2 f2 f2 f2 04 f2
  0x10007a18cd70: f2 f2 f2 f2 f2 f2 04 f2 f2 f2 f3 f3 f3 f3 00 00
  0x10007a18cd80: 00 00 00 00 00 00 00 00 00 00 00 00 f1 f1 f1 f1
  0x10007a18cd90: 04 f2 f2 f2 f2 f2 f2 f2 04 f2 f2 f2 f2 f2 f2 f2
  0x10007a18cda0: 04 f2 f2 f2 f2 f2 f2 f2 04 f2 f2 f2 f2 f2 f2 f2
=>0x10007a18cdb0: 00 f2 f2 f2 f2 f2 f2[f2]00 00 00 00 00 00 00 00
  0x10007a18cdc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007a18cdd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007a18cde0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007a18cdf0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007a18ce00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==4095==ABORTING```

iceman1001, 2021-01-26:
So the call to ASKbiphaseDemod triggers it.

```
ASKbiphaseDemod src/cmddata.c:769

 [352, 131424) 'BitStream' <== Memory access at offset 351 underflows this variable
 ```
 
 What kind of tag where you running the `lf search`on?
 
 ```
 lf read
 data save -f lf_search_bug_biphase.pm3
 ```
 and upload/share it?

socram8888, 2021-01-26:
There was none. I was trying to scan a mysterious 13.56MHz that crashes the Proxmark (will open another issue later when I have more info) but also happens with no tag.

iceman1001, 2021-01-26:
... empty...  should be a length check so no demodulation fcts is called with no data.

iceman1001, 2021-01-27:
yeah,   `hf search`  seems to crash on `14b reader` ...

iceman1001, 2021-01-27:
Try pulling latest,  I think there is no underflow longer

closed 2021-01-27 by iceman1001

socram8888, 2021-01-28:
Indeed, works just fine, thanks!

commit 02c1034c (2021-01-27): fix #1175 - biphase demod identification of value clocks now fails if no changes detected