RfidResearchGroup/proxmark3 #1201: HF field not activating for DESFire commands

VortixDev, 2021-02-15:
**Describe the bug**
The HF field is not being activated for transmission when running the `hf mfdes createaid` command, according to the warning (I can also replicate this with `hf mfdes formatpicc`, and have not tested others):
```
[usb] pm3 --> hf mfdes createaid -a 123456 -f 1111 -k 0E -l 2E --name Test
[=] Creating AID using:
[=] AID      563412
[=] Key set1 0x0E
[=] Key Set2 0x2E
[=] FID      1111
[=] DF Name  Test
[#] Warning: HF field is off, ignoring TransmitFor14443a command
[!!] 🚨 APDU: No APDU response.
[!] ⚠     Can't create aid ->
```

**To Reproduce**
Run the example command for `hf mfdes createaid` (`hf mfdes createaid -a 123456 -f 1111 -k 0E -l 2E --name Test`).

**Expected behavior**
The field should be activated prior to any attempt to transmit.

**Additional information**
Using `hf 14a raw -sk 00` to activate the field before running the `hf mfdes createaid` command leads to the command executing successfully. It should be noted that, although the field stays on after the `hf 14a raw` command is run, the command will still fail with "No APDU response" if the card is withdrawn from the field before `hf mfdes createaid` is ran (even if brought back into the field before executing). A successful run is shown below.

```
[usb] pm3 --> hf mfdes createaid -a 123456 -f 1111 -k 0E -l 2E --name Test
[=] Creating AID using:
[=] AID      563412
[=] Key set1 0x0E
[=] Key Set2 0x2E
[=] FID      1111
[=] DF Name  Test
[#] Warning: HF field is off, ignoring TransmitFor14443a command
[!!] 🚨 APDU: No APDU response.
[!] ⚠     Can't create aid -> 
[usb] pm3 --> hf 14a raw -sk 00
[+] Card selected. UID[7]:
[+] 04 59 45 32 0B 4E 80 
[+] received 0 bytes
[usb] pm3 --> hf mfdes createaid -a 123456 -f 1111 -k 0E -l 2E --name Test
[=] Creating AID using:
[=] AID      563412
[=] Key set1 0x0E
[=] Key Set2 0x2E
[=] FID      1111
[=] DF Name  Test
[+] Successfully created aid.
```

**Diagnostic information**
 - OS: Kali Linux

 - hw version
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-2982-g3a051b326 2021-01-31 13:43:36
  compiled with GCC 10.2.1 20201224 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-2982-g3a051b326 2021-01-31 13:44:23
       os: RRG/Iceman/master/v4.9237-2982-g3a051b326 2021-01-31 13:44:47
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 309679 bytes (59%) Free: 214609 bytes (41%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

 - hw status
```
[usb] pm3 --> hw status
[#] Memory                                                                                     
[#]   BigBuf_size.............42336
[#]   Available memory........33552
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............0
[#]   dma8 memory.............-2111832
[#]   dma16 memory............-2111832
[#]   toSend memory...........33552
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD5690C23DF535B2A
[#] Smart card module (ISO 7816)
[#]   version.................v3.10
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........4768 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override....std    : follow standard 
[#]   [b] BCC override........std    : follow standard
[#]   [2] CL2 override........std    : follow standard
[#]   [3] CL3 override........std    : follow standard
[#]   [r] RATS override.......std    : follow standard 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......274944
[#]   Transfer Speed PM3 -> Client = 549888 bytes/s
[#] Various
[#]   Max stack usage.........4768 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............75
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31722 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
```

 - data tune
```
[usb] pm3 --> data tune
[=] REMINDER: 'hw tune' doesn't actively tune your antennas, it's only informative
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 74.02 V - 125.00 kHz
[+] LF antenna: 35.57 V - 134.83 kHz
[+] LF optimal: 74.46 V - 126.32 kHz
[+] Approx. Q factor (*): 12.6 by frequency bandwidth measurement
[+] Approx. Q factor (*): 13.0 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 45.90 V - 13.56 MHz
[+] Approx. Q factor (*): 8.0 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

iceman1001, 2021-02-15:
That is the process for `hf mfdes createaid` . You need to authenticate first and the auth command turns on the field.
Most desfire commands needs a auth first.  So this is by design. 
Either we implement auth for each command,  which means a messy command line, 
or we split it up and might suffer from ppl who forget they have the field on.   

Its not super well documented the way that we operated the DESFire on a Pm3.  Feel free to make a note / blogpost about it.

closed 2021-02-15 by VortixDev

iceman1001, 2021-02-15:
I do mean that a note_on_desfire.md  would be great.