RfidResearchGroup/proxmark3 #1216: Error with RSA sign

K7cl, 2021-03-03:
-

**Describe the bug**
RSA sign fail in `mem info`

**To Reproduce**
Steps to reproduce the behavior:
1. clone the master branch
2. edit client/src/cmdflashmem.c at line 459, change ·bool shall_sign = false, shall_write = false;` to `bool shall_sign = true, shall_write = true;`
3. make clean && make all %% make install
4. connect to pm3 and run comman `mem info`

**Expected behavior**
sign then write to pm3, to override empty sign. To fix wrong return in rdv4 device detection.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Kali 2021.1
 - inside proxmark3 client run the following commands and paste the output here.
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3239-g5657e392f-dirt...-unclean 2021-03-03 03:46:26
  compiled with GCC 10.2.1 20210110 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  device.................... RDV4   // I edit the cmdhw.c at 972 from `if (rdv4_validate(&mem) == PM3_SUCCESS) {` to `if (rdv4_validate(&mem) == PM3_EFAILD) {` for find the error raeson, so it shows the right return this time.
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3239-g5657e392f-dirt...-unclean 2021-03-03 03:46:57
       os: RRG/Iceman/master/v4.9237-3239-g5657e392f-dirt...-unclean 2021-03-03 03:47:13
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............40248
[#]   Available memory........40248
[#] Tracing
[#]   tracing ................1
[#]   traceLen ...............0
[#]   dma8 memory.............-2113920
[#]   dma16 memory............-2113920
[#]   toSend memory...........-2113920
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD567A882A76AC526
[#] Smart card module (ISO 7816)
[#]   version.................v3.11
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........3960 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override....std    : follow standard 
[#]   [b] BCC override........std    : follow standard
[#]   [2] CL2 override........std    : follow standard
[#]   [3] CL3 override........std    : follow standard
[#]   [r] RATS override.......std    : follow standard 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......282112
[#]   Transfer Speed PM3 -> Client = 564224 bytes/s
[#] Various
[#]   Max stack usage.........4104 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............-1
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............30364 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#]   Mifare..................1018 keys
[#]   T55x7...................113 keys
[#]   iClass..................9 keys
[usb] pm3 --> data tune
[=] REMINDER: 'hw tune' doesn't actively tune your antennas, it's only informative
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 36.89 V - 125.00 kHz
[+] LF antenna: 27.45 V - 134.83 kHz
[+] LF optimal: 36.89 V - 125.00 kHz
[+] Approx. Q factor (*): 5.6 by frequency bandwidth measurement
[+] Approx. Q factor (*): 6.4 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 46.09 V - 13.56 MHz
[+] Approx. Q factor (*): 8.0 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

iceman1001, 2021-03-03:
That code is disabled by choice,    of course the private key isn't there.

closed 2021-03-03 by iceman1001

K7cl, 2021-03-03:
Thanks! So the private key coded under the pub key is not real?

iceman1001, 2021-03-03:
As mentioned in the source
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/src/cmdflashmem.c#L42-L43