RfidResearchGroup/proxmark3 #1229: Client always shows "device / fw mismatch"

ikarus23, 2021-03-30:
Hi, I just wanted you guys to know that the client always shows "device / fw mismatch", even after updating everything. What a I did was:

```
git pull
make clean
make -j
./pm3-flash-all
./pm3
```

And I get:

```
 [ PROXMARK3 ]
  device.................... device / fw mismatch
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present
```

I'm using my PM3 RDV 4.01.

doegox, 2021-03-30:
This sounds like the signature stored in your external flash got erased.
I'd suggest to contact RRG and provide them your flash Unique ID (cf `hw status`) to get and restore the signature data.

iceman1001, 2021-03-30:
The output from  `mem info`   is better.

ikarus23, 2021-03-31:
What has this to do with a signature on the flash? I think I might interpret this message all wrong, as it is not that clear to me. I thought this message tells me of the popular issue of having the client software not matching the firmware on the PM3. I guess I got this wrong. For what is this message then? Just originality checks?

Yes, my signature seems to be gone. Did not even know there was one. Is it important for any other thing than originality check? If not, I don't think I need it back.

```
[usb] pm3 --> mem info

[=] --- Flash memory Information ---------
[=] ID................... 259C99A782A867D5
[=] SHA1................. 986B6D2487CD190042C6E746395EE7B6137DBF7B
[-] ⛔ Signature............ ( fail )

[=] --- RDV4 RSA signature ---------------
[=]  FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
[=]  FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
[=]  FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
[=]  FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF

[=] --- RDV4 RSA Public key --------------
[=] Len.................. 128
[=] Exponent............. 010001
[=] Public key modulus N
[=]  E28D809BF323171D11D1ACA4C32A5B7E0A8974FD171E75AD120D60E9B76968FF
[=]  4B0A6364AE50583F9555B8EE1A725F279E949246DF0EFCE4C02B9F3ACDCC623F
[=]  9337F21C0C066FFB703D8BFCB5067F309E056772096642C2B1A8F50305D5EC33
[=]  DB7FB5A3C8AC42EB635AE3C148C910750ABAA280CE82DC2F180F49F30A1393B5

[+] RSA public key validation.... ( ok )
[-] ⛔ RSA private key validation... ( fail )
[-] ⛔ RSA verification..... ( fail )
```

iceman1001, 2021-03-31:
its being used to verify that you haven't by mistake load RRG/Iceman fw compiled for RDV4 device on a _generic_ proxmark3 device. 
The signature exists only on RDV4,  and if the signature fails,  we can use that as a indication your hw/fw is mismatched.

All been discussed on discord previously,  

closing.

closed 2021-03-31 by iceman1001

ikarus23, 2021-03-31:
ok

LVWILHITE, 2021-06-11:
I have the same issue 

when proxmark3 loads....

> [ PROXMARK3 ]

> device.................... device / fw mismatch

mem info shows...

> ⛔ Signature............ ( fail )

> ....

> ⛔ Genuine Proxmark3 RDV4 signature detected ... (X) <--red X

I have tried to find the "All been discussed on discord previously," as noted by @ iceman1001 but was unsuccessful.  Can someone tell me how I might, "contact RRG and provide them your flash Unique ID (cf hw status) to get and restore the signature data." as noted by doegox? 

@ikarus23 - did you get this corrected?  If so, how?

Thanks so much in advance to all!

iceman1001, 2021-06-12:
You find the channel under _newbie_ category. 
For your second question,  use your google-foo?

ikarus23, 2021-06-13:
@LVWILHITE Hi. No, I did not do anything about it. It is just a cosmetic issue. I know I have an RDV4 and I know I have flashed the correct firmware. So the error message is just a false positive due to the missing signature in the flash. For me getting the correct signature from Proxgrid is not worth the trouble. I just live with the error message. :)