RfidResearchGroup/proxmark3 #1232: Hitag2 simu

RaspExt, 2021-04-03:
**Description**

After to have started the hitag2 simulation through the following command « lf hitag sim 2 car.json », I can see the configured EEPROM displayed on the console and simulation is well started. However, impossible to decode correctly the signal in order to be able to detect/see the START_AUTH command despite the fact that I can see the signal with the commands: data plot and lf sniff -s 3000 -@ in oscilloscope mode.

There may be some adjustments to be made. Any idea ?

**Expected behavior**

Be able to emulate a hitag2 tag in order to replace the hitag2 keyfob.

iceman1001, 2021-04-03:
#124 
#240
#878 
#889
#551
#764

The `lf hitag` commands needs more love.  I am currently looking into the `lf hitag sniff`   
If you go back to an older version of the repo like two years,  or rather before I did some refactoring of the hitag2 stuff you find simulation working.

dk5ras, 2022-02-04:
HITAG in general seems to be a mistery :) Quite hard to start playing with it without at least some basic tools...

pispo1, 2023-07-03:
> #124 #240 #878 #889 #551 #764
> 
> The `lf hitag` commands needs more love. I am currently looking into the `lf hitag sniff` If you go back to an older version of the repo like two years, or rather before I did some refactoring of the hitag2 stuff you find simulation working.

I have been looking the versions around April 2019 but it is no clear to me identify a working version of **lf hitag sim**. @iceman1001 Could you tell me a working commit of tag sim?

iceman1001, 2023-07-05:
I have no clue,   try git bisect,  from 2018 to 2020 in order to find a working hitag sim.

iceman1001, 2024-09-09:
Closing since lack of interest

closed 2024-09-09 by iceman1001