RfidResearchGroup/proxmark3 #1233: Communication fail when reading a strix4k

AndColla, 2021-04-03:
**Describe the bug**
Proxmark hw disconnect after a `hf search` with a strix4k tag on the hf antenna.
With the original firmware I don´t have any issue and gat the correct answer.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch `hf search` with a strix4k tag on the hf antenna
2. See screenshot 1

**Expected behavior**
The hardware should not disconnect and I should get the correct infos.
See screenshot 2 for the output from the original fw/client

**Screenshots**
Error:
[image]
Original fw/client:
[image]

**Desktop (please complete the following information):**
 - OS: Arch Linux x86_64 Kernel: 5.11.11-arch1-1
 - hw version
  [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3428-gb6ab16637 2021-04-03 10:53:31
  compiled with GCC 10.2.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3428-gb6ab16637 2021-04-03 10:54:04
       os: RRG/Iceman/master/v4.9237-3428-gb6ab16637 2021-04-03 10:54:25
  compiled with GCC 10.2.0

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )
 - hw status
[#] Memory
[#]   BigBuf_size.............44248
[#]   Available memory........44248
[#] Tracing
[#]   tracing ................1
[#]   traceLen ...............0
[#]   dma8 memory.............-2109920
[#]   dma16 memory............-2109920
[#]   toSend memory...........-2109920
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........3928 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override....std    : follow standard 
[#]   [b] BCC override........std    : follow standard
[#]   [2] CL2 override........std    : follow standard
[#]   [3] CL3 override........std    : follow standard
[#]   [r] RATS override.......std    : follow standard 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......308736
[#]   Transfer Speed PM3 -> Client = 617472 bytes/s
[#] Various
[#]   Max stack usage.........4088 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............-1
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............32351 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
 - data tune
=] ---------- LF Antenna ----------
[+] LF antenna: 40,82 V - 125,00 kHz
[+] LF antenna: 40,41 V - 134,83 kHz
[+] LF optimal: 47,96 V - 129,03 kHz
[+] Approx. Q factor (*): 8,4 by frequency bandwidth measurement
[+] Approx. Q factor (*): 13,9 by peak voltage measurement
[!] ⚠️  Contradicting measures seem to indicate you're running a PM3_OTHER firmware on a RDV4, please check your setup
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 33,94 V - 13.56 MHz
[+] Approx. Q factor (*): 9,9 by peak voltage measurement
[+] HF antenna is OK

**Additional context**
`data tune` says that I´m running a PM3_OTHER firmware on a RDV4, but I´m using a Proxmark3 easy
I´m currently running the latest version from github (commit b6ab166377f47dbfd92993f76ff6edbfb0a9d496)

AndColla, 2021-04-03:
I´ve read issue #921 and in theory ISO14443B should work (FeliCa crashes, all others OK), but it only works in release v4.9237

iceman1001, 2021-04-03:
Thanks for reporting it. 
Here I thought it was stable enough to enable it in `hf search`  ,   I reverted it and your 14b should be detected nice again if you pull latest

AndColla, 2021-04-03:
I just pulled the latest commit and got the same problem

iceman1001, 2021-04-03:
You got the same????   That doesn't make any sense.  What are you doing?

AndColla, 2021-04-03:
I´m just doing a `hf search`, nothing special
With original firmware or with release v4.9237 I don´t have any issue

iceman1001, 2021-04-03:
What is the output when you run `hf search`  on latest source?

AndColla, 2021-04-03:
This is the output on latest source (commit 160941c280d642874c85444fe5db366254079d44)
[image]

iceman1001, 2021-04-04:
Ok,  confirmed,   I found my 14b tag,  and `hf search` get stuck.

iceman1001, 2021-04-23:
try pulling latest and test.  I pushed a fix  https://github.com/RfidResearchGroup/proxmark3/commit/2119c4e7bd918290266f45a1823dd7dc8c0809f0

AndColla, 2021-04-24:
I confirm that it's working on 0d04c14be34e4f732282fe5605ccc3f1b0e6f3fe

closed 2021-04-24 by AndColla