RfidResearchGroup/proxmark3 #1244: Unable to use or update Rysccorp Proxmark Pro after updating bootrom

dan2600, 2021-04-13:
After flashing my Proxmark pro with the newest released bootrom, the Proxmark pro enters "unchained mode" on boot, the keypad does not respond, and it does not mount on the PC.

No direction on the proxmark pro forces DFU mode

I cannot find documentation to place the Proxmark Pro into DFU mode through the hardware.

Since it is still "booting" it does not seem that the bootloader is corrupt, just unable to enter DFU mode since it goes into Unchained mode.

Installed with pm3-flash-bootrom.bat

**Expected behavior**
After flashing bootrom unit should enter DFU or standard USB mode.

**LOG**
```
[+] About to use the following file:
[+]    bootrom.elf
[+] Waiting for Proxmark3 to appear on COM5
 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on COM5
 found
[!!] ====================== OBS ! ===========================================
[!!] Note: Your bootloader does not understand the new CMD_BL_VERSION command
[!!] It is recommended that you first update your bootloader alone,
[!!] reboot the Proxmark3 then only update the main firmware

[!!] Follow these steps :
[!!]  1)   ./pm3-flash-bootrom
[!!]  2)   ./pm3-flash-all
[!!]  3)   ./pm3
[=] --------------------------------------------------------
[=] Available memory on this board: UNKNOWN

[!!] ====================== OBS ! ======================================
[!!] Note: Your bootloader does not understand the new CHIP_INFO command
[=] Permitted flash range: 0x00100000-0x00140000
[+] Loading ELF file bootrom.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d28->0x00000d28) [R X] @0x298

[+] Flashing...
[+] Writing segments for file: bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
 OK
[+]  0x00100200..0x00100f27 [0xd28 / 7 blocks]
 OK

[+] All done

[=] Have a nice day!

```

**Desktop (please complete the following information):**
 - OS: Windows 10
   Unable to connect to Proxmark Pro after flashing bootrom.

iceman1001, 2021-04-13:
I guess you didn't read the notes before where we mention that `proxmark pro` most likely will mess up.

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#platform

[image]

closed 2021-04-13 by iceman1001

lain-d, 2021-04-13:
@iceman1001 Thanks Iceman, indeed I did miss that note. Please refer to my other suggestion above. From other repos i've seen with baremetal support (often for ARM or Xtensa) it seems common practice in the readme to separate devices with full support from ones with limited or non-functioning support, rather than list it as supported and put "but it doesn't work" in the known issues. 

love this repo! working great on my proxmark3 cheers!

doegox, 2021-04-13:
done

iceman1001, 2021-04-13:
Yeah,  there is much to read.   Hard to make it foolproof, since we have noticed a large amount of newcomers tends not to read anything. 
The supported section has been clarified beyond doubt.