RfidResearchGroup/proxmark3 #1245: Proxmark3 easy issue

koszyk1980, 2021-04-13:
Hi, I starting fun with this amazing device . I did all steps from this video and when I press FLASH - all.bat  I get this message on screen ? I using proxmark3 easy bought  on AliExpress . Any advice?? 

Link video : https://youtu.be/ROIplLNklpo
Screenshot: https://www.dropbox.com/s/40hgsqia1sisq06/screen.jpg?dl=0

iceman1001, 2021-04-13:
Have you followed the installation instructions on this repo ?  
The instructions here is up-to-date and proven to work.

koszyk1980, 2021-04-13:
> Have you followed the installation instructions on this repo ?  
> 
> The instructions here is up-to-date and proven to work.   
> 
> 
> 
> 

I did everything like on video on YouTube https://youtu.be/ROIplLNklpo

Can you send me link with instructions up to date . I will try again 🤔

iceman1001, 2021-04-13:
So you didn't follow the instructions on this repo.  ok.  
You find all relevant links in the readme / first page of this repo.

closed 2021-04-13 by iceman1001