RfidResearchGroup/proxmark3 #1246: Proxmark3 easy disconnect after command  pm3 flash all ?

koszyk1980, 2021-04-13:
Hi any idea why i can’t communicate with proxmark3 now ? I did all steps from rfidresearchgroup . After command pm3-flash -all my device is disconnected:( https://twitter.com/messages/media/1382076885149093901

nick-draffen-tfs, 2021-04-13:
Are you using precompiled or did you compile yourself?  Make sure you are using the `PLATFORM=PM3GENERIC`

Gator96100, 2021-04-14:
1. Did you compile with PLATFORM=PM3GENERIC?
2. When coming from an unknown firmware firmware pm3-flash-all will result in soft bricking your device. You’ll need to hold down the button while plugging your proxmark in and only release it after all your flashing is done. You need to run pm3-flash-bootrom first.

koszyk1980, 2021-04-14:
My steps : 

Driver Installation ( Windows 7 ) my port is com6 

Download the Gator96100 ProxSpace package from https://github.com/Gator96100/ProxSpace/releases

Extract 'ProxSpace' to a location path without spaces.

Run runme64.bat.

Clone the RRG/Iceman repository

cd
git clone https://github.com/RfidResearchGroup/proxmark3.git
cd proxmark3

I can’t understand this ?? So I skip it 

Compile and use the project

To use the compiled client, the only differences are that executables end with .exe (e.g. proxmark3.exe) and that the Proxmark3 port is one of your comX ports where "X" is the com port number assigned to proxmark3 under Windows, so commands like proxmark3 /dev/ttyACMX become proxmark3.exe comX.

After i did :

Get the latest commits

cd proxmark3
git pull

Clean and compile everything

make clean && make all

An finally I did 

pm3-flash-all

After this command my device is disconnected no visible fir windows ?

iceman1001, 2021-04-14:
Did you follow the instructions on this repository on how to install?

iceman1001, 2021-04-14:
and I am closing, since this isn't a repo issue,  but a user issue which can / should be resolved on forum or discord in a more convenient way.

closed 2021-04-14 by iceman1001