RfidResearchGroup/proxmark3 #1252: Cannot save and load data using "trace" command

wh201906, 2021-04-22:
**Describe the bug**
When I use "hf mf sniff" then "trace save" for many times, it will only save the first trace data, even I use different filename.  
Also, the "trace load" cannot load the file into the trace buffer.

**To Reproduce**
Just "hf mf sniff" and "trace save". The logs might be better to show how to reproduce.

**Expected behavior**
The trace files should be different.

**Screenshots**
For the bug in "trace save":

    [usb] pm3 --> hf 14a sniff -c
    
    [#] Starting to sniff. Press PM3 Button to stop.
    [#] trace len = 2298
    [usb] pm3 --> trace list
    
    Annotate trace buffer with selected protocol data
    You can load a trace from file (see `trace load -h`) or it be downloaded from device by default
    
    
    usage:
        trace listoptions:
    
    examples/notes:
        trace list -t raw              -> just show raw data without annotations
        trace list -t 14a              -> interpret as ISO14443-A communications
        trace list -t thinfilm         -> interpret as Thinfilm communications
        trace list -t topaz            -> interpret as Topaz communications
        trace list -t mf               -> interpret as MIFARE Classic communications and decrypt crypto1 stream
        trace list -t des              -> interpret as MIFARE DESFire communications
        trace list -t 14b              -> interpret as ISO14443-B communications
        trace list -t 7816             -> interpret as ISO7816-4 communications
        trace list -t 15               -> interpret as ISO15693 communications
        trace list -t iclass           -> interpret as iCLASS communications
        trace list -t legic            -> interpret as LEGIC communications
        trace list -t felica           -> interpret as ISO18092 / FeliCa communications
        trace list -t hitag1           -> interpret as Hitag1 communications
        trace list -t hitag2           -> interpret as Hitag2 communications
        trace list -t hitags           -> interpret as HitagS communications
        trace list -t lto              -> interpret as LTO-CM communications
        trace list -t cryptorf         -> interpret as CryptoRF communitcations
        trace list -t mf --dict <mfc_default_keys>         -> use dictionary keys file
        trace list -t 14a -f                               -> show frame delay times
        trace list -t 14a -1                               -> use trace buffer
    
    [usb] pm3 --> trace
    [usb] pm3 --> trace save
    
    Save protocol data from trace buffer to binary file
    File extension is <.trace>
    
    usage:
        trace saveoptions:
    
    examples/notes:
        trace save -f mytracefile      -> w/o file extension
    
    [usb] pm3 --> trace save -f 111.trc
    [=] downloading tracelog data from device
    [+] saved 2298 bytes to binary file 111.trc.trace
    [usb] pm3 --> hf 14a sniff -c
    
    [#] Starting to sniff. Press PM3 Button to stop.
    [#] trace len = 2517
    [usb] pm3 --> trace save -f 112.trc
    [+] saved 2298 bytes to binary file 112.trc.trace
    [usb] pm3 --> hf 14a sniff -c
[... 115 lines cut ...]
        [=] ---------- Reminder ------------------------
        [=] `hw tune` doesn't actively tune your antennas,
        [=] it's only informative.
        [=] Measuring antenna characteristics, please wait...
         🕛   9
        [=] ---------- LF Antenna ----------
        [+] LF antenna: 29.13 V - 125.00 kHz
        [+] LF antenna: 28.46 V - 134.83 kHz
        [+] LF optimal: 30.99 V - 127.66 kHz
        [+] Approx. Q factor (*): 5.4 by frequency bandwidth measurement
        [+] Approx. Q factor (*): 9.0 by peak voltage measurement
        [!] ⚠️  Contradicting measures seem to indicate you're running a PM3_GENERIC firmware on a RDV4
        [!] ⚠️  False positives is possible but please check your setup
        [+] LF antenna is OK
        [=] ---------- HF Antenna ----------
        [+] HF antenna: 27.65 V - 13.56 MHz
        [+] Approx. Q factor (*): 8.0 by peak voltage measurement
        [+] HF antenna is OK
        
        (*) Q factor must be measured without tag on the antenna
        
        [+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
        
        [!] ⚠️  You appear to be on an environment without an X11 server or without DISPLAY environment variable set.
        [!] ⚠️  Plot may not work until you resolve these issues.

iceman1001, 2021-04-22:
No bug.   Its related to user handling error.  

If you read the help text,  you find that you need to specify that you don't want to download a trace from device per automatic by specifying `-1` param.

You also seem to struggle with the params in general when loading file.   Its quite simple. 

```
 trace load -f traces/hf_14a_reader_4b_rats.trace
 trace list -1 -t 14b
 `` `

closed 2021-04-22 by iceman1001

wh201906, 2021-04-22:
@iceman1001 I didn't check the help text carefully. Sorry about that. I know how to load trace files now.

How about saving trace files? I called "hf 14a sniff" then "trace save -f xxx" for many times with different filenames, but the traces files are all the same(I used cmp command and it returned nothing, the size of them are also the same). You can look at the logs above. The traceLen was changing after each "hf 14a sniff" but the "trace save" always returned "saved 2298 bytes..."
I just went through the help text about "trace load" and "trace save", I don't know whether I am operating wrongly.
Thanks for your help!

wh201906, 2021-04-22:
I got the point.
After "hf 14a sniff", I need to call "trace list -t xx" to dump the trace from device to the client buffer then list it, and "trace save" only save the data from the client buffer.
"trace list --buffer -t xx" will NOT dump the trace from the device, it will use the data from client buffer directly.

iceman1001, 2021-04-22:
`Trace save`  will download if graph buffer is empty.   If you already downloaded something, it will save that.
```
    if (g_traceLen == 0) {
        download_trace();
        if (g_traceLen == 0) {
            PrintAndLogEx(WARNING, "trace is empty, nothing to save");
            return PM3_SUCCESS;
        }
    }
```

`data clear` will empty both graphbuffer and bigbuf on device side. 

```
data clear
hf 14a sniff -cr

trace save -f xxxx
or 
trace list -t 14a
trace save -f xxx
```