RfidResearchGroup/proxmark3 #1261: lf hitag info not finding key

will-caruana, 2021-05-01:
**Describe the bug**
lf hitag info not finding key

**To Reproduce**
lf hitag info
or
lf search 
or 
auto

**Expected behavior**
I was able to confirm the key worked using an old branch:
```[usb] pm3 --> lf hitag info

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]      UID: 0E04718C
[+]     TYPE:
[=] -------------------------------------------------------------

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3070-g667e2bb9f 2021-02-12 11:29:06
  compiled with GCC 9.3.0 OS:Linux ARCH:x86_64

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3070-g667e2bb9f 2021-02-12 11:30:39
       os: RRG/Iceman/master/v4.9237-3070-g667e2bb9f 2021-02-12 11:31:18```

**Current Desktop Setup:**
 ```[ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3733-g3bdc9d183 2021-05-01 10:56:23
  compiled with GCC 9.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3733-g3bdc9d183 2021-05-01 10:56:44
       os: RRG/Iceman/master/v4.9237-3733-g3bdc9d183 2021-05-01 10:56:57
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )
```

iceman1001, 2021-05-01:
Try pulling latest and flash,  and see if it works again

will-caruana, 2021-05-01:
That worked thank you.

closed 2021-05-01 by will-caruana