RfidResearchGroup/proxmark3 #1270: Can't compile under arm-none-eabi-gcc 11.1.0

xiashali, 2021-05-13:
**Describe the bug**

>compiler version:  arm-none-eabi-gcc (Arch Repository) 11.1.0
[-] CC lfops.c
lfops.c: In function 'printT55xxConfig':
lfops.c:257:9: error: this 'else' clause does not guard... [-Werror=misleading-indentation]
  257 |         else
      |         ^~~~
lfops.c:260:13: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'else'
  260 |             if (T55xx_Timing.m[i].write_3 != 0xFFFF && i == T55XX_DLMODE_1OF4)
      |             ^~
cc1: all warnings being treated as errors

After fix indents of `lfops.c`, I got

>  optimized_ikeys.c:310:28: error: argument 1 of type 'uint8_t *' {aka 'unsigned char *'} declared as a pointer [-Werror=array-parameter=]
  310 | void diversifyKey(uint8_t *csn, uint8_t *key, uint8_t *div_key) {
      |                   ~~~~~~~~~^~~
In file included from optimized_ikeys.c:59:
optimized_ikeys.h:62:27: note: previously declared as an array 'uint8_t[8]' {aka 'unsigned char[8]'}
   62 | void diversifyKey(uint8_t csn[8], uint8_t key[8], uint8_t div_key[8]);
      |                   ~~~~~~~~^~~~~~
optimized_ikeys.c:310:42: error: argument 2 of type 'uint8_t *' {aka 'unsigned char *'} declared as a pointer [-Werror=array-parameter=]
  310 | void diversifyKey(uint8_t *csn, uint8_t *key, uint8_t *div_key) {
      |                                 ~~~~~~~~~^~~
In file included from optimized_ikeys.c:59:
optimized_ikeys.h:62:43: note: previously declared as an array 'uint8_t[8]' {aka 'unsigned char[8]'}
   62 | void diversifyKey(uint8_t csn[8], uint8_t key[8], uint8_t div_key[8]);
      |                                   ~~~~~~~~^~~~~~
optimized_ikeys.c:310:56: error: argument 3 of type 'uint8_t *' {aka 'unsigned char *'} declared as a pointer [-Werror=array-parameter=]
  310 | void diversifyKey(uint8_t *csn, uint8_t *key, uint8_t *div_key) {
      |                                               ~~~~~~~~~^~~~~~~
In file included from optimized_ikeys.c:59:
optimized_ikeys.h:62:59: note: previously declared as an array 'uint8_t[8]' {aka 'unsigned char[8]'}
   62 | void diversifyKey(uint8_t csn[8], uint8_t key[8], uint8_t div_key[8]);
      |                                                   ~~~~~~~~^~~~~~~~~~
cc1: all warnings being treated as errors

**Desktop**
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.2 in /usr/lib)
native BT support: Bluez found, enabled
Jansson library:   system library found
Lua library:       system library found
Python3 library:   Python3 v3.9 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  gcc (GCC) 10.2.0

iceman1001, 2021-05-13:
Always new compilers.   Feel free to make a Pull Request with fixes

iceman1001, 2021-05-13:
never mind,  pull latest and see if  you get warnings

xiashali, 2021-05-13:
```
> BigBuf.c: In function 'LogTrace':
> BigBuf.c:260:8: error: array subscript 'tracelog_hdr_t[0]' is partly outside array bounds of 'uint8_t[1]' {aka 'unsigned char[1]'} [-Werror=array-bounds]
>   260 |     hdr->timestamp = timestamp_start;
>       |        ^~
> BigBuf.c:17:30: note: while referencing '__bss_end__'
>    17 | extern uint8_t _stack_start, __bss_end__;
>       |                              ^~~~~~~~~~~
> BigBuf.c:261:8: error: array subscript 'tracelog_hdr_t[0]' is partly outside array bounds of 'uint8_t[1]' {aka 'unsigned char[1]'} [-Werror=array-bounds]
>   261 |     hdr->duration = duration & 0xFFFF;
>       |        ^~
> BigBuf.c:17:30: note: while referencing '__bss_end__'
>    17 | extern uint8_t _stack_start, __bss_end__;
>       |                              ^~~~~~~~~~~
> BigBuf.c:262:8: error: array subscript 'tracelog_hdr_t[0]' is partly outside array bounds of 'uint8_t[1]' {aka 'unsigned char[1]'} [-Werror=array-bounds]
>   262 |     hdr->data_len = iLen;
>       |        ^~
> BigBuf.c:17:30: note: while referencing '__bss_end__'
>    17 | extern uint8_t _stack_start, __bss_end__;
>       |                              ^~~~~~~~~~~
> BigBuf.c:263:8: error: array subscript 'tracelog_hdr_t[0]' is partly outside array bounds of 'uint8_t[1]' {aka 'unsigned char[1]'} [-Werror=array-bounds]
>   263 |     hdr->isResponse = !readerToTag;
>       |        ^~
> BigBuf.c:17:30: note: while referencing '__bss_end__'
>    17 | extern uint8_t _stack_start, __bss_end__;
>       |                              ^~~~~~~~~~~
> cc1: all warnings being treated as errors
> 
```

iceman1001, 2021-05-13:
well,   that makes it.  You have to wait some months before support for that compiler will be.

doegox, 2021-05-13:
Meanwhile, you can compile with `make NOERROR=1` to ignore the warning.

doegox, 2021-05-13:
FTR gcc 11.1.0 brings some more issues on the host side too:
```
src/cmdhf14b.c: In function 'CmdHF14BDump':
src/cmdhf14b.c:685:27: error: array subscript 1020 is outside array bounds of 'uint8_t[516]' {aka 'unsigned char[516]'} [-Werror=array-bounds]
  685 |             if ((lockbytes[0] & mask)) {
      |                  ~~~~~~~~~^~~
src/cmdhf14b.c:1349:13: note: referencing a variable length array of size between 68 and 516
 1349 |     uint8_t data[cardsize];
      |             ^~~~
src/cmdhf14b.c:575:27: error: array subscript 1021 is outside array bounds of 'uint8_t[516]' {aka 'unsigned char[516]'} [-Werror=array-bounds]
  575 |             if ((lockbytes[1] & mask) == 0) {
      |                  ~~~~~~~~~^~~
src/cmdhf14b.c:1349:13: note: referencing a variable length array of size between 68 and 516
 1349 |     uint8_t data[cardsize];
      |             ^~~~

src/cmdhffelica.c: In function 'CmdHFFelicaAuthentication2':
src/cmdhffelica.c:819:5: warning: 'mbedtls_des3_crypt_ecb' reading 8 bytes from a region of size 1 [-Wstringop-overread]
  819 |     mbedtls_des3_crypt_ecb(&des3_ctx_dec, cc, p3c);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdhffelica.c:819:5: note: referencing argument 2 of type 'const unsigned char *'
In file included from src/cmdhffelica.c:23:
../common/mbedtls/des.h:291:5: note: in a call to function 'mbedtls_des3_crypt_ecb'
  291 | int mbedtls_des3_crypt_ecb(mbedtls_des3_context *ctx,
      |     ^~~~~~~~~~~~~~~~~~~~~~

src/loclass/ikeys.c:352:28: warning: argument 1 of type 'uint8_t *' {aka 'unsigned char *'} declared as a pointer [-Warray-parameter=]
  352 | void diversifyKey(uint8_t *csn, uint8_t *key, uint8_t *div_key) {
      |                   ~~~~~~~~~^~~
In file included from src/loclass/ikeys.c:61:
src/loclass/ikeys.h:63:27: note: previously declared as an array 'uint8_t[8]' {aka 'unsigned char[8]'}
   63 | void diversifyKey(uint8_t csn[8], uint8_t key[8], uint8_t div_key[8]);
      |                   ~~~~~~~~^~~~~~
src/loclass/ikeys.c:352:42: warning: argument 2 of type 'uint8_t *' {aka 'unsigned char *'} declared as a pointer [-Warray-parameter=]
  352 | void diversifyKey(uint8_t *csn, uint8_t *key, uint8_t *div_key) {
      |                                 ~~~~~~~~~^~~
In file included from src/loclass/ikeys.c:61:
src/loclass/ikeys.h:63:43: note: previously declared as an array 'uint8_t[8]' {aka 'unsigned char[8]'}
   63 | void diversifyKey(uint8_t csn[8], uint8_t key[8], uint8_t div_key[8]);
      |                                   ~~~~~~~~^~~~~~
src/loclass/ikeys.c:352:56: warning: argument 3 of type 'uint8_t *' {aka 'unsigned char *'} declared as a pointer [-Warray-parameter=]
  352 | void diversifyKey(uint8_t *csn, uint8_t *key, uint8_t *div_key) {
      |                                               ~~~~~~~~~^~~~~~~
In file included from src/loclass/ikeys.c:61:
src/loclass/ikeys.h:63:59: note: previously declared as an array 'uint8_t[8]' {aka 'unsigned char[8]'}
   63 | void diversifyKey(uint8_t csn[8], uint8_t key[8], uint8_t div_key[8]);
      |                                                   ~~~~~~~~^~~~~~~~~~

cmac.c:405:45: warning: argument 5 of type 'unsigned char *' declared as a pointer [-Warray-parameter=]
  405 |                              unsigned char *output) {
      |                              ~~~~~~~~~~~~~~~^~~~~~
In file included from cmac.c:53:
../mbedtls/cmac.h:190:44: note: previously declared as an array 'unsigned char[16]'
  190 |                              unsigned char output[16]);
      |                              ~~~~~~~~~~~~~~^~~~~~~~~~

ht2crack5gpu.c: In function 'main':
ht2crack5gpu.c:288:9: warning: unused variable 'gpu' [-Wunused-variable]
  288 |     int gpu = 1;
      |         ^~~
```

doegox, 2021-05-13:
Compilation with gcc 11.1 for both ARM and Host should be fixed now, please test @xiashali

xiashali, 2021-05-14:
It works now, thank you. @doegox

closed 2021-05-14 by doegox