RfidResearchGroup/proxmark3 #1274: Application failed to start due to no Qt platform plugin could be initialized??

TiekHan, 2021-05-15:
I encounter this issue for both the proxmark3 command line, and the universal gui. Strangely the cross-compiled gui doesnt have this kind of problem. Any idea how to fix this? Thanks.
[image]
[image]

**One more thing, the pm3.bat is working fine for me.**
[image]

iceman1001, 2021-05-15:
When it comes to the GUI,  we don't support other software in this repo.  
When it comes to pre-compiled binaries,  same thing, we don't support it in this repo.  You have to ask @Gator96100  nicely.

And since the last image shows it to be working,   then there seem _not_ to be any problem with this repo.  

If you don't show relevant information to this repo, I will close this issue

TiekHan, 2021-05-15:
alright, let me figure out what caused this . Thanks @iceman1001

closed 2021-05-15 by TiekHan

Gator96100, 2021-05-15:
The reason pm3.bat works is because it calls setup.bat. 
Hint: look at what variables are sets by setup.bat

TiekHan, 2021-05-16:
Thanks @Gator96100 , i managed to make it works by adding a system variable = "QT_QPA_PLATFORM_PLUGIN_PATH" , variable value = "%QT_PLUGIN_PATH%".