RfidResearchGroup/proxmark3 #1278: 03a7109 compile error on ubuntu 18.04 with arm-none-eabi-gcc

JohnDoeIsKIng, 2021-05-19:
rev 03a7109 brings on a compile error on ubuntu 18.04:

[*] MAKE bootrom/all
/bin/sh: 1: arm-none-eabi-gcc: not found
compiler version:  
[-] CC ../common/mbedtls/version.c
make[1]: arm-none-eabi-gcc: Command not found
../common_arm/Makefile.common:123: recipe for target 'obj/version.o' failed
make[1]: *** [obj/version.o] Error 127
Makefile:130: recipe for target 'bootrom/all' failed
make: *** [bootrom/all] Error 2

Obviously, the binary isn't found.
It's definitely in my path:
$ echo "$PATH"
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/john/gcc-arm-none-eabi-10-2020-q4-major/bin/

What else could I try ?
Regards

JD.

doegox, 2021-05-19:
the arm-none-eabi-gcc binary is in your *user* PATH but you're compiling with `sudo`, so with the *root* PATH.

there is no reason to run the compilation as sudo, I would recommend to never do that. (what if by mistake a `rm -rf /${UNKNOWN_VAR}` becomes a `sudo rm -rf /`...)

closed 2021-05-19 by doegox

JohnDoeIsKIng, 2021-05-19:
Ouh ...

Your'e right of course.
A chmod 777 -c -R di the job to do the make without sudo.
Sorry for that.
Best regards

JD.