RfidResearchGroup/proxmark3 #1283: android phone + termux + proxmark not compiling on Pixel3A

c0decave, 2021-05-22:
**Describe the bug**
Compiling the proxmark3 repository ends with an error message:

```
/data/data/com.termux/files/home/proxmark3/client/src/cmdlf.c:267:17: error: variable length array folded to constant array as an extension [-Werror,-Wgnu-folding-constant]
        uint8_t data[PM3_CMD_DATA_SIZE - payload_header_size];
                ^
1 error generated.
make[2]: *** [CMakeFiles/proxmark3.dir/build.make:1326: CMakeFiles/proxmark3.dir/src/cmdlf.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: CMakeFiles/proxmark3.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the Steps for termux here: https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/termux_notes.md
2. Choose to compile proxmark3 from the git repository
3. Compile it
4. See error

**Expected behavior**
Compiling through without error

**Phone:**
 - HW: Pixel3A
 - OS: Android 11 (Rooted)
 - Patch Level: May, 2021

**Additional context**
My workaround:  

In src/cmdlf.c around line 267. I guess that the version of clang or cc in termux is older and simply does not like the former construction although it is defined const properbly the variable LF_CMDREAD_EXTRA_SYMBOLS is troubleing it. As LF_CMDREAD... is defined as 4, i simply replaced the payload_header_size with the result of 12 + (3*4).  

   
```
const uint8_t payload_header_size = 12 + (3 * LF_CMDREAD_MAX_EXTRA_SYMBOLS);
struct p {
        uint32_t delay;
        uint16_t period_0;
        uint16_t period_1;
        uint8_t  symbol_extra[LF_CMDREAD_MAX_EXTRA_SYMBOLS];
        uint16_t period_extra[LF_CMDREAD_MAX_EXTRA_SYMBOLS];
        uint32_t samples : 31;
        bool     verbose : 1;
 //       uint8_t data[PM3_CMD_DATA_SIZE - payload_header_size ];
        uint8_t data[PM3_CMD_DATA_SIZE - 24];

```

Now proxmark compiles fine and is useable on my phone.

doegox, 2021-05-22:
Please provide the info visible at the beginning of the compilation with compiler version, thanks.

c0decave, 2021-05-22:
```
~/proxmark3 $ git tag
v4.9237

```

cleanup and make client

```
~/proxmark3 $ make clean && make client

===================================================================
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/clean
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE mf_nonce_brute/clean

===================================================================
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all

```
clang 
```
clang --version
clang version 12.0.0
Target: aarch64-unknown-linux-android24
Thread model: posix
InstalledDir: /data/data/com.termux/files/usr/bin

```

doegox, 2021-05-22:
Thank you, I could reproduce on host with clang 12

doegox, 2021-05-22:
should be fixed.
please confirm.
thanks again four your bugreport.

closed 2021-05-24 by iceman1001