RfidResearchGroup/proxmark3 #1285: Missing amiibo commands or wrong heading

ikarus23, 2021-05-24:
Hi. Probably just a cosmetic thing, but for `hf mfu` the last heading is "amiibo". However, there are no amiibo commands there.
```
[usb] pm3 --> hf mfu
help             This help
-----------      ----------------------- recovery -------------------------
keygen           Generate 3DES MIFARE diversified keys
pwdgen           Generate pwd from known algos
otptear          Tear-off test on OTP bits
-----------      ----------------------- operations -----------------------
cauth            Authentication - Ultralight-C
dump             Dump MIFARE Ultralight family tag to binary file
info             Tag information
ndef             Prints NDEF records from card
rdbl             Read block
restore          Restore a dump onto a MFU MAGIC tag
wrbl             Write block
---------        ----------------------- simulation -----------------------
eload            load Ultralight .eml dump file into emulator memory
eview            View emulator memory
sim              Simulate MIFARE Ultralight from emulator memory
---------        ----------------------- magic ----------------------------
setpwd           Set 3DES key - Ultralight-C
setuid           Set UID - MAGIC tags only
---------        ----------------------- amiibo ----------------------------
```

iceman1001, 2021-05-24:
part of the skeleton commands,  currently commented out,   until someone implements them.
I will comment out the last line, so it doesn't confuse people.

iceman1001, 2021-05-24:
done https://github.com/RfidResearchGroup/proxmark3/commit/d9f1b977525631a9b92506916c2d645bd7a4267e

closed 2021-05-24 by iceman1001

ikarus23, 2021-05-24:
That is what I though. Thanks for the quick fix.