RfidResearchGroup/proxmark3 #1287: Error in filename produced by "hf mf autopwn"

ikarus23, 2021-05-25:
Hi! As you all know `hf mf autopwn` produces files with the card content and the keys. I just used the command and noticed one file name got corrupted.

```
[...]

[+] | 013 | ffffffffff00   | R | 00ffffffffff   | R |
[+] | 014 | ffffffffff00   | R | 00ffffffffff   | R |
[+] | 015 | ffffffffff00   | R | 00ffffffffff   | R |
[+] |-----|----------------|---|----------------|---|
[=] ( D:Dictionary / S:darkSide / U:User / R:Reused / N:Nested / H:Hardnested / C:statiCnested / A:keyA  )

[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-A4훜��u.bin
[=] FYI! --> 0xFFFFFFFFFFFF <-- has been inserted for unknown keys where res is 0
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file hf-mf-A49FA432-dump.bin
[+] saved 64 blocks to text file hf-mf-A49FA432-dump.eml
[+] saved to json file hf-mf-A49FA432-dump.json
[=] autopwn execution time: 32 seconds
[usb] pm3 --> exit

ikarus@moon ~/rfid/proxmark3-rdv4 (git)-[master] % ls -l hf-mf-A4*
-rw-r--r-- 1 ikarus users  1024 May 25 18:43  hf-mf-A49FA432-dump.bin
-rw-r--r-- 1 ikarus users  2111 May 25 18:43  hf-mf-A49FA432-dump.eml
-rw-r--r-- 1 ikarus users 11642 May 25 18:43  hf-mf-A49FA432-dump.json
-rw-r--r-- 1 ikarus users   192 May 25 18:43 'hf-mf-A4훜'$'\265\314''u.bin'
```

iceman1001, 2021-05-25:
Interesting,
it looks like the key file name got corrupted somehow

doegox, 2021-05-25:
I can't reproduce here.
@ikarus23 can you reproduce and dig into it ?

ikarus23, 2021-05-25:
Just to be transparent:
I've talked to @doegox. I could reproduce it. The error occurs if I crack using the known key from sector 17 (signature sector) of a EV1 1k tag. So `hf mf auto -s 17 -a -k 75ccb59c9bed` will lead to this sector.

@doegox found the issue. Typical overflow because PM3 thinks MFC EV1 1k is 16 sectors big.

doegox, 2021-05-25:
yes so I could reproduce it too :)
@ikarus23 I can close the issue ?

ikarus23, 2021-05-25:
Yes, it is fixed. Thank you!.

closed 2021-05-25 by ikarus23