RfidResearchGroup/proxmark3 #1288: Weired sector calcualtion in "hf mf chk"

ikarus23, 2021-05-25:
MIFARE Classic tags have memory which is logically divided into sectors and block. The biggest MFC tag has 4k memory. For this tag, the first 32 sectors have 4 blocks and the remaining 8 sectors have 16 blocks. This adds to a total of 256 blocks.

When using `hf mf chk` it is possible to specify a block. By doing so, and especially if a high block is chosen (e.g. 252), there are two strange output lines, that I can not make sens of.
 
```
[usb] pm3 --> hf mf chk -a --blk 252 -f mfc_default_keys.dic
[+] Loaded 1094 keys from mfc_default_keys.dic
[=] Start check for keys...
[=] ..
[=] time in checkkeys 0 seconds

[=] testing to read key B...
[=] Sector 25, First block of sector 100, Num of block 4
[=] Reading block 103
Data:FF FF FF FF FF FF 

[+] found keys:

[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 025 | ffffffffffff   | 1 | ffffffffffff   | 1 |
[+] |-----|----------------|---|----------------|---|
[+] ( 0:Failed / 1:Success )
```

Look at 
```
[=] Sector 25, First block of sector 100, Num of block 4
[=] Reading block 103
```
The sectors number seams wrong. Also, what is is with "First block of sector 100"? And  "Reading block 103" seems wrong too. 
It looks like there are some strange block2sector calculations.

A good implementation might be taken from the Android source code:
```
    public int blockToSector(int blockIndex) {
        validateBlock(blockIndex);
        if (blockIndex < 32 * 4) {
            return blockIndex / 4;
        } else {
            return 32 + (blockIndex - 32 * 4) / 16;
        }
    }
```

iceman1001, 2021-05-26:
I think you are mis-reading the output. 
Sector 25,  first block is 100,  will read the sector trailer at block 103, to see if key B can be read out.

Closing since there is no bug.

ikarus23, 2021-05-26:
I thought so too, but why "Sector 25"?. As you can see I from the input, I was doing `hf mf chk` for block 252. This block is in sector 39 (of a 4k MFC tag).

mwalker33, 2021-05-26:
Just having a quick look.
I think the "GetSectorFromBlockNo" function has a bug.
```
    if (blockNo < 128)
        return blockNo / 4;
    else
        return 32 + ((128 - blockNo) / 16);
    
block 252 would result in 
        32 + ((128 - 252) / 16) 
     => 32 + (-124/16)
     => 32 + (-7.75)
     => 24.25
    
```     
But if we swap the order of 128 - blockNo, which to me makes more sense.
i.e. we want to know how many 16 block sectors to add to the 32 * 4 block sectors.
```
          32 + ((252 - 128) / 16)
    => 32 + (124/16)
    => 32 +  7.75
    => 39.75  
```

doegox, 2021-05-26:
seems to make sense indeed

doegox, 2021-05-26:
Fixed in https://github.com/RfidResearchGroup/proxmark3/commit/b260cc19cfae6a1cc1054a5af65b8f95e1a3a215
Sounds ok for you ?

ikarus23, 2021-05-26:
Yes, now it makes sense!. Thank you.
(Why is this in the future branch? should' it be part of the master?)

closed 2021-05-26 by ikarus23

doegox, 2021-05-26:
> (Why is this in the future branch? should' it be part of the master?)

it will be but @iceman1001 is on travel and will give trainings so master needs to be stable and I don't push anything without his acknowledge, therefore "future" as staging branch.

doegox, 2021-05-26:
now merged in master

iceman1001, 2021-05-28:
so what happens if the block number is less than 128 now?

doegox, 2021-05-28:
Same as before