RfidResearchGroup/proxmark3 #1289: mf_nonce_brute fails to find first bits of key

datatags, 2021-05-26:
**Describe the bug**
mf_nonce_brute is not able to find the first 16 bits of the key in "phase 2"

**To Reproduce**
Steps to reproduce the behavior:
1. CD into tools/mf_nonce_brute/ and run the second command example given by `./mf_nonce_brute`
2. When it completes, it will say "failed to find key"

**Expected behavior**
mf_nonce_brute to find the correct key.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Raspberry Pi OS
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3948-g95dfe5491 2021-05-25 03:22:56
  compiled with GCC 8.3.0 OS:Linux ARCH:arm

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3948-g95dfe5491 2021-05-25 03:28:29
       os: RRG/Iceman/master/v4.9237-3948-g95dfe5491 2021-05-25 03:31:50
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )
```
 - hw status
```
[#] Memory
[#]   BigBuf_size............. 44208
[#]   Available memory........ 44208
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#] LF Sampling Stack
[#]   Max stack usage......... 3944 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 109056
[#]   Transfer Speed PM3 -> Client... 218112 bytes/s
[#] Various
[#]   Max stack usage......... 4104 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31569 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#]
```
 - data tune
```
[=] ---------- LF Antenna ----------
[+] LF antenna: 38.45 V - 125.00 kHz
[+] LF antenna: 29.98 V - 134.83 kHz
[+] LF optimal: 38.45 V - 125.00 kHz
[+] Approx. Q factor (*): 6.2 by frequency bandwidth measurement
[+] Approx. Q factor (*): 11.2 by peak voltage measurement
[!] ⚠️  Contradicting measures seem to indicate you're running a PM3_GENERIC firmware on a RDV4
[!] ⚠️  False positives is possible but please check your setup
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 33.29 V - 13.56 MHz
[+] Approx. Q factor (*): 9.7 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

**Additional context**
The issue seems to be caused by [this line](https://github.com/RfidResearchGroup/proxmark3/blob/54153153b8e7e63eb02334cdd1b0cf4af43717f8/tools/mf_nonce_brute/mf_nonce_brute.c#L474):
```
key |= (count << 32);
```
The issue being that the upper bits of `key` are set from `count` but never un-set. Adding a debug message to the loop confirmed this, where there were many, many repeated lines like this:
```
...
Checking key fffc...
Checking key fffe...
Checking key fffd...
Checking key ffff...
Checking key fffc...
Checking key fffe...
Checking key fffd...
Checking key ffff...
...
```
where those four digits are just the top four digits of `key`. I was able to fix the issue by changing the previously mentioned line to this:
```
key = args->part_key | (count << 32);
```
which gave the expected output:
`Valid Key found [ 3b7e4fd575ad ]`

Thanks!

iceman1001, 2021-05-26:
Interesting,  I thought I pushed that fix.  Good finding it and good write up.

doegox, 2021-05-26:
Thanks @datatags 
I added your fix with some others along the way, in https://github.com/RfidResearchGroup/proxmark3/commit/13ae226a601bf6745fbbc52db532acc69714c5b7

Note that it's currently in a "future" branch because of temporary freeze of master.
It'll go to master as soon as @iceman is ok with it.

doegox, 2021-05-26:
now merged in master

closed 2021-05-26 by doegox