RfidResearchGroup/proxmark3 #1292: hf mfdes getuid - not return correct value of UID

ah01, 2021-05-29:
**Description**

I have Mifare DESFire EV1 and EV2 with enabled random UID (otherwise in default state including default keys). When I authenticate with default key (succesfully) and try to call `hf mfdes getuid` I will get 16 bytes of data as UID that are definitely not UID of the card (at least not in readable form).

I do not posses datasheet for DESFire but at least I compare PM3 implementation and libfreefare ([source](https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_desfire.c#L1204)) implementation. It looks like PM3 return direct response of PICC, but the data are actually enciphered.

**Log**

```
 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3956-g3fe6c417b 2021-05-26 19:10:25
  compiled with MinGW-w64 10.2.0 OS:Windows (64b) ARCH:x86_64
```

Example of info of EV1 (EV2 behaves same):

```
[usb] pm3 --> hf mfdes info

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]               UID: 80 72 6B E8
[+]      Batch number: B9 0C 19 49 50
[+]   Production date: week 25 / 2019

[=] --- Hardware Information
[=]    raw: 04010101001805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 1.0 ( DESFire EV1 )
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )

[=] --- Software Information
[=]    raw: 04010101041805
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 1.4
[=]   Storage size: 0x18 ( 4096 bytes )
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )

[=] --- Card capabilities
[=]     1.4 - DESFire Ev1 MF3ICD21/41/81, EAL4+
[+]    Number of Masterkeys                  : 1
[+]    Operation of PICC master key          : (3)DES
[+]    PICC Master key Version               : 0 (0x00)
[=]    ----------------------------------------------------------
[+]    [0x0A] Authenticate      : YES
[+]    [0x1A] Authenticate ISO  : YES
[=] -------------------------------------------------------------
[=]  Key setting: 0x0F [1111]
[+]    [1...] CMK Configuration changeable   : YES
[+]    [.1..] CMK required for create/delete : NO
[+]    [..1.] Directory list access with CMK : NO
[+]    [...1] CMK is changeable              : YES

[=] --- Free memory
[+]    Available free memory on card         : 4864 bytes
```

Authenticate with default key:

```
[usb] pm3 --> hf mfdes auth -m 2 -t 2 -a 000000 -n 0
[+]   Key        : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[+]   SESSION    : 44 BA 4F DE 0D 95 75 26 44 BA 4F DE 28 29 FA 60
[=] -------------------------------------------------------------
```

Get UID:

```
[usb] pm3 --> hf mfdes getuid
[+]     UID: EE 5E C8 83 E6 36 B3 13 29 90 1E 6D 95 0B 4C 9C
```

**Expected result**

UID of this card is `04 0F 46 5A D5 62 80` (read by Elatec TWN4 USB reader).

mwalker33, 2021-05-29:
Thanks for the report.
Please note that desire is a work in progress and we are working though it atm.
I note that you use the default key of all 00's and have authenticated with ISO 3Des.

While my response is not the long term answer, it may help isolate a little.
Can you try to authenticate with standard des and see if you get the correct UID.

hf mfdes auth -m 1 -t 1 -a 000000 -n 0

Thanks

ah01, 2021-05-30:
There is no difference:
```
[usb] pm3 --> hf mfdes auth -m 1 -t 1 -a 000000 -n 0
[+]   Key        : 00 00 00 00 00 00 00 00
[+]   SESSION    : 94 B3 9F 54 35 C4 B6 68
[=] -------------------------------------------------------------
[=] -------------------------------------------------------------
[usb] pm3 --> hf mfdes getuid
[+]     UID: E3 6E A8 C2 0E E9 FC 60 AF 25 41 E4 16 2C D6 DB
```

mwalker33, 2021-05-30:
Thanks for running the test.
I have tracked down part of it and think its mostly working (in my test code) for native Single des.
i.e. in my test code it gives the same UID as my omnikey reader.
So will keep working on it.

mwalker33, 2021-06-20:
This should now be working in the latest master.

Please test and confirm if its working as you expect.

ah01, 2021-06-20:
It works perfectly. Tested with DESFire EV1 and EV2 with both defaults DES and 3DES keys.

Great work 👍

closed 2021-06-20 by iceman1001