RfidResearchGroup/proxmark3 #1296: Windows 10 ANSI color not working

Gator96100, 2021-05-31:
I created this issue to help me track down the reason why ANSI color support fails on some Windows 10 PCs. As I am not able to replicate this issue, I do require the help of users who have this issue.

This issue only affects people using Windows 10 with ProxSpace. WSL users are not affected.

**Symptoms:**
- No colors in the Proxmark3 client
- “strange” symbols at the start of every line, see screenshot
[image]

### How to help me find the cause of this issue:
I do need a lot of information about your system to find why some people are affected and others aren’t. To do this, I did create a script to automatically collect information about your: system, environment, processes, software and updates, as well as registry entries that affect your terminal. I do know that is a lot of information to share and if you are not comfortable with sharing it feel free to remove some lines of `collectinfo.txt`, just tell me if you did modify it. 

#### If you do have this issue and want to help do the following:
1. Run this command within ProxSpace:
`bash <(wget -O - 'https://www.proxmarkbuilds.org/experimental/collectinfo.sh')`
2. Make a screenshot of the output.
3. Upload the file `collectinfo.txt` somewhere (e.g. https://pastebin.com/, or attach it to this issue) or send it to me directly over Discord (Gator96100#2719).

PercyJax, 2021-06-01:
I’ve experienced this issue as well. It seems to correlate with window maximizing and snapping, though not 100%.

Gator96100, 2021-06-01:
> 
> 
> I’ve experienced this issue as well. It seems to correlate with window maximizing and snapping, though not 100%.

Thank you so much! Resizing is the problem. I can replicate the issue now.

Gator96100, 2021-06-01:
#1297 should fix it

closed 2021-06-01 by Gator96100