RfidResearchGroup/proxmark3 #1298: AWID Cloning to T55xx Card Hangs Client

PercyJax, 2021-06-01:
**Describe the bug**
Cloning an AWID token to a T55xx card succeeds (the Proxmark blinks, and the card is verified afterwards to have the right data), but the client hangs and console control is not returned.

**To Reproduce**
Steps to reproduce the behavior:
1. Place T55xx card on antenna.
2. Run `lf awid clone --fmt 26 --fc 123 --cn 45678`.
3. The following is output, then the console is hung:
```[=] Preparing to clone AWID 26 to T55x7 with FC: 123 CN: 45678
[+] Blk | Data
[+] ----+------------
[+]  00 | 00107060
[+]  01 | 011DB7DE
[+]  02 | 722BD811
[+]  03 | 11111111```

**Expected behavior**
Card is written to, then control is returned to the console.

**Screenshots**
![image](https://user-images.githubusercontent.com/9935391/120346245-32b6fc00-c2c9-11eb-9b41-8c92fc8375a9.png)

**Desktop (please complete the following information):**
 - OS: Debian 10 on WSL in Windows 10 Pro
 - inside proxmark3 client run the following commands and paste the output here.

```[usb] pm3 --> hw version
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3969-gc45331e1e 2021-06-01 02:43:11
  compiled with GCC 8.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3969-gc45331e1e 2021-06-01 02:43:46
       os: RRG/Iceman/master/v4.9237-3969-gc45331e1e 2021-06-01 02:44:01
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 52% used )```

```[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 44464
[#]   Available memory........ 44464
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#] LF Sampling Stack
[#]   Max stack usage......... 4048 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 240128
[#]   Transfer Speed PM3 -> Client... 480256 bytes/s
[#] Various
[#]   Max stack usage......... 4136 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 10
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30096 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#]```

```[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 21.83 V - 125.00 kHz
[+] LF antenna: 27.67 V - 134.83 kHz
[+] LF optimal: 28.20 V - 136.36 kHz
[+] Approx. Q factor (*): 4.1 by frequency bandwidth measurement
[+] Approx. Q factor (*): 8.2 by peak voltage measurement
[!] ⚠️  Contradicting measures seem to indicate you're running a PM3_GENERIC firmware on a RDV4
[!] ⚠️  False positives is possible but please check your setup
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 34.85 V - 13.56 MHz
[+] Approx. Q factor (*): 10.1 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

[!] ⚠️  You appear to be on an environment without an X11 server or without DISPLAY environment variable set.
[!] ⚠️  Plot may not work until you resolve these issues.
```

iceman1001, 2021-06-01:
Is that on latest master?  (compiled / flashed) ?

PercyJax, 2021-06-01:
This is last tested on a compiled and flashed `commit c45331e1e2ee2411b4158bfb40590c51ec01b6a3`

```percy@Voyager:~/proxmark3$ cat Makefile.platform
# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
# Run 'make PLATFORM=' to get an exhaustive list of possible parameters for this file.

#PLATFORM=PM3RDV4
PLATFORM=PM3GENERIC
# If you want more than one PLATFORM_EXTRAS option, separate them by spaces:
#PLATFORM_EXTRAS=BTADDON
STANDALONE=LF_SAMYRUN
```

Also tested on the nightly binary and had similar issue

PercyJax, 2021-06-01:
Client Debug = Full:

```
[usb] pm3 --> lf awid clone --fmt 26 --fc 123 --cn 45678
[#] awid raw bits:
[#]  0000000100011101101101111101111001110010001010111101100000010001000100010001000100010001

[=] Preparing to clone AWID 26 to T55x7 with FC: 123 CN: 45678
[+] Blk | Data
[+] ----+------------
[+]  00 | 00107060
[+]  01 | 011DB7DE
[+]  02 | 722BD811
[+]  03 | 11111111
[#] LF signal properties:
[#]   high..........255
[#]   low...........10
[#]   mean..........128
[#]   amplitude.....127
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] DEBUG: threshold Count reached at index 36, count: 3
[#] DEBUG (fskdemod) got 1419 bits
[#] DEBUG (aggregate_bits) FSK startIdx 50, fclow*idx 64, n*clk 50
[#] DEBUG (aggregate_bits) extra bits in the end
[#] DEBUG (fskdemod) got 238 bits
[#] DEBUG: (setClockGrid) demodoffset 50, clk 50
[#] DEBUG: (FSKrawDemod) using clock:50, inverted, fc high:10, fc low:8

[+] FSK2a decoded bitstream
[=] -----------------------
[+] DemodBuffer:
[+] 10011111111011111000111110011111
[+] 10011111111011111000111110011111
[+] 10011111111011111000111110011111
[+] 10011111111011111000111110011111
[+] 10011111111011111000111110011111
[+] 10011111111011111000111110011111
[+] 10011111111011111000111110011111
[+] 10011111111011
```

doegox, 2021-06-01:
I tried with same platform and same commit on a native Debian 10, I can't reproduce the bug.
I tried with a T5577 or even without tag, same.
FTR full log should look like this
```
[usb] pm3 --> lf awid clone --fmt 26 --fc 123 --cn 45678
[#] awid raw bits:
[#]  0000000100011101101101111101111001110010001010111101100000010001000100010001000100010001 

[=] Preparing to clone AWID 26 to T55x7 with FC: 123 CN: 45678
[+] Blk | Data 
[+] ----+------------
[+]  00 | 00107060
[+]  01 | 011DB7DE
[+]  02 | 722BD811
[+]  03 | 11111111
[#] LF signal properties:
[#]   high..........255
[#]   low...........1
[#]   mean..........130
[#]   amplitude.....125
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] DEBUG: threshold Count reached at index 20, count: 3
[#] DEBUG (fskdemod) got 1441 bits
[#] DEBUG (aggregate_bits) FSK startIdx -4, fclow*idx 576, n*clk 600
[#] DEBUG (aggregate_bits) extra bits in the end
[#] DEBUG (fskdemod) got 239 bits
[#] DEBUG: (setClockGrid) demodoffset -4, clk 50
[#] DEBUG: (FSKrawDemod) using clock:50, inverted, fc high:10, fc low:8

[+] FSK2a decoded bitstream
[=] -----------------------
[+] DemodBuffer:
[+] 11111111111101111100011111001111
[+] 11111111111101111100011111001111
[+] 11111111111101111100011111001111
[+] 11111111111101111100011111001111
[+] 11111111111101111100011111001111
[+] 11111111111101111100011111001111
[+] 11111111111101111100011111001111
[+] 111111111111011

[#] LF signal properties:
[#]   high..........254
[#]   low...........1
[#]   mean..........129
[#]   amplitude.....125
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] DEBUG: threshold Count reached at index 20, count: 3
[#] DEBUG (fskdemod) got 1341 bits
[#] DEBUG (aggregate_bits) FSK startIdx -4, fclow*idx 376, n*clk 400
[#] DEBUG (fskdemod) got 239 bits
[#] DEBUG: (setClockGrid) demodoffset -4, clk 50
[#] DEBUG: (FSKrawDemod) using clock:50, inverted, fc high:10, fc low:8

[+] FSK2a decoded bitstream
[=] -----------------------
[+] DemodBuffer:
[+] 11111111011100010010010000010000
[+] 11111111011100010010010000010000
[+] 11111111011100010010010000010000
[+] 11111111011100010010010000010000
[+] 11111111011100010010010000010000
[+] 11111111011100010010010000010000
[+] 11111111011100010010010000010000
[+] 111111110111000

[#] LF signal properties:
[#]   high..........254
[#]   low...........1
[#]   mean..........129
[#]   amplitude.....125
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] DEBUG: threshold Count reached at index 20, count: 3
[#] DEBUG (fskdemod) got 1364 bits
[#] DEBUG (aggregate_bits) FSK startIdx 0, fclow*idx 80, n*clk 100
[#] DEBUG (fskdemod) got 239 bits
[#] DEBUG: (setClockGrid) demodoffset 0, clk 50
[#] DEBUG: (FSKrawDemod) using clock:50, inverted, fc high:10, fc low:8

[+] FSK2a decoded bitstream
[=] -----------------------
[+] DemodBuffer:
[+] 11000110111010100001001111110111
[+] 01000110111010100001001111110111
[+] 01000110111010100001001111110111
[+] 01000110111010100001001111110111
[+] 01000110111010100001001111110111
[+] 01000110111010100001001111110111
[+] 01000110111010100001001111110111
[+] 010001101110101

[#] LF signal properties:
[#]   high..........255
[#]   low...........1
[#]   mean..........130
[#]   amplitude.....125
[#]   is Noise......No
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] DEBUG: threshold Count reached at index 20, count: 3
[#] DEBUG (fskdemod) got 1421 bits
[#] DEBUG (aggregate_bits) FSK startIdx -4, fclow*idx 176, n*clk 200
[#] DEBUG (aggregate_bits) extra bits in the end
[#] DEBUG (fskdemod) got 239 bits
[#] DEBUG: (setClockGrid) demodoffset -4, clk 50
[#] DEBUG: (FSKrawDemod) using clock:50, inverted, fc high:10, fc low:8

[+] FSK2a decoded bitstream
[=] -----------------------
[+] DemodBuffer:
[+] 11110111011101110111011101110111
[+] 01110111011101110111011101110111
[+] 01110111011101110111011101110111
[+] 01110111011101110111011101110111
[+] 01110111011101110111011101110111
[+] 01110111011101110111011101110111
[+] 01110111011101110111011101110111
[+] 011101110111011

[+] Data written and verified
[+] Done
[?] Hint: try `lf awid reader` to verify
```
so it seems it hangs on your side after the first demodulation

PercyJax, 2021-06-01:
What is the expected behavior if you do not have a tag on the antenna? I tried with another tag and it went through fine. I tried with no tag and it hung in the same place.

doegox, 2021-06-01:
with no tag it looks like this
```
[usb] pm3 --> lf awid clone --fmt 26 --fc 123 --cn 45678
[#] awid raw bits:
[#]  0000000100011101101101111101111001110010001010111101100000010001000100010001000100010001 

[=] Preparing to clone AWID 26 to T55x7 with FC: 123 CN: 45678
[+] Blk | Data 
[+] ----+------------
[+]  00 | 00107060
[+]  01 | 011DB7DE
[+]  02 | 722BD811
[+]  03 | 11111111
[#] LF signal properties:
[#]   high..........132
[#]   low...........126
[#]   mean..........129
[#]   amplitude.....3
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] LF signal properties:
[#]   high..........131
[#]   low...........126
[#]   mean..........129
[#]   amplitude.....2
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] LF signal properties:
[#]   high..........132
[#]   low...........127
[#]   mean..........129
[#]   amplitude.....3
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] LF signal properties:
[#]   high..........132
[#]   low...........127
[#]   mean..........129
[#]   amplitude.....3
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] LF signal properties:
[#]   high..........132
[#]   low...........127
[#]   mean..........129
[#]   amplitude.....3
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] LF signal properties:
[#]   high..........131
[#]   low...........127
[#]   mean..........129
[#]   amplitude.....2
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] LF signal properties:
[#]   high..........132
[#]   low...........126
[#]   mean..........129
[#]   amplitude.....3
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[#] LF signal properties:
[#]   high..........132
[#]   low...........127
[#]   mean..........129
[#]   amplitude.....3
[#]   is Noise......Yes
[#]   THRESHOLD noise amplitude......8
[#] DEBUG: (setClockGrid) clear settings
[+] Done
[?] Hint: try `lf awid reader` to verify

```

doegox, 2021-06-01:
What type of Proxmark do you have ?
I tried the PM3GENERIC on amyRDV4 but range of values might depend on the model...

PercyJax, 2021-06-01:
I have a KKmoon Proxmark3 Easy, with 512 K chip

doegox, 2021-06-01:
"good" news: I can reproduce on a Pm3 Easy and no tag. Thanks.

doegox, 2021-06-01:
unbounded loop fixed on master, cc @iceman1001 
it triggers only on noisy envs, that's why it's harder to trigger on the RDV4. (`getSignalProperties()->isnoise` in `AcquireData`)

closed 2021-06-01 by doegox