RfidResearchGroup/proxmark3 #1305: Choosing the Platform on MacOS with Homebrew

CaseyB, 2021-06-08:
I’m following the instructions [here](https://forum.dangerousthings.com/t/getting-started-with-proxmark3-on-mac/10156) but it keeps defaulting the platform to `PM3RDV4` when I do brew install.

Here I’m setting the var and verifying that it’s set
```
$ export HOMEBREW_PROXMARK3_PLATFORM=PM3GENERIC
$ echo $HOMEBREW_PROXMARK3_PLATFORM
PM3GENERIC
```
This is what I see when I try to install:
```
$ brew install --HEAD proxmark3
==> Installing proxmark3 from rfidresearchgroup/proxmark3
==> Cloning https://github.com/RfidResearchGroup/proxmark3.git
Updating /Users/<user>/Library/Caches/Homebrew/proxmark3--git
From https://github.com/RfidResearchGroup/proxmark3
   3fe6c417b..899f269a9  master     -> origin/master
==> Checking out branch master
Already on 'master'
Your branch is behind 'origin/master' by 14 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at 899f269a9 Fix wtf bug in t55xxAquireAndCompareBlock0
==> make clean
==> make all BREW_PREFIX=/usr/local PLATFORM=PM3RDV4
```
I also tried editing the Makefile.platform just to see if that changed anything and it didn't. I can get it to work when I use
```
make clean
make PLATFORM=PM3GENERIC
```

iceman1001, 2021-06-08:
are you following the instructions in the https://github.com/RfidResearchGroup/homebrew-proxmark3 repo? 
You should start an issue in that repo instead of here.

CaseyB, 2021-06-08:
You are correct, sorry about that!

iceman1001, 2021-06-08:
however,  you are correct in the fact that the OSX instructions in this repo is in need for some and rewriting to match the updated homebrew repo instructions.

iceman1001, 2021-06-15:
Textual update has been merged.

closed 2021-06-15 by iceman1001