RfidResearchGroup/proxmark3 #1307: missing comands

jithan10, 2021-06-10:
***Things to try before submitting bug report***
read the [troubleshooting guide](/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to RRG/Iceman repo.

-

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

jithan10, 2021-06-10:
when i run the command i get this 

[usb] pm3 --> lf hid clone -w H10301 --fc 118 --cn 1603
[=] Preparing to clone HID tag with ID: 000000000
[+] Done

iceman1001, 2021-06-10:
too little relevant information.  Like I said before,   take time to fill in a proper issue .

jithan10, 2021-06-10:
***Things to try before submitting bug report***
read the [troubleshooting guide](/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to RRG/Iceman repo.
____________________________________________________________________________________________

Last login: Thu Jun 10 11:22:45 on ttys002
jy@jts-MacBook-Pro ~ % brew tap RfidResearchGroup/proxmark3
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> New Formulae
maturin                                  tmuxp
==> Updated Formulae
Updated 45 formulae.

jy@jts-MacBook-Pro ~ % export HOMEBREW_PROXMARK3_PLATFORM=xxxxxx
jy@jts-MacBook-Pro ~ % brew install proxmark3
Warning: rfidresearchgroup/proxmark3/proxmark3 4.9237 is already installed and up-to-date.
To reinstall 4.9237, run:
  brew reinstall proxmark3
jy@jts-MacBook-Pro ~ % brew reinstall proxmark3
==> Downloading https://github.com/RfidResearchGroup/proxmark3/archive/v4.9237.t
Already downloaded: /Users/jy/Library/Caches/Homebrew/downloads/8d6348999984c4412e8cb081c1b3fa53b78eb10062c506ff1d8e16344dd22763--proxmark3-4.9237.tar.gz
==> Reinstalling rfidresearchgroup/proxmark3/proxmark3 
==> make clean
==> make all BREW_PREFIX=/usr/local PLATFORM=PM3RDV4
==> make install PREFIX=/usr/local/Cellar/proxmark3/4.9237 BREW_PREFIX=/usr/loca
==> Install success!
==> The latest bootloader and firmware binaries are ready and waiting in the cur
🍺  /usr/local/Cellar/proxmark3/4.9237: 582 files, 16.5MB, built in 1 minute 22 seconds
jy@jts-MacBook-Pro ~ % brew upgrade --fetch-HEAD proxmark3

Warning: rfidresearchgroup/proxmark3/proxmark3 4.9237 already installed
jy@jts-MacBook-Pro ~ % pm3-flash-all

[=] Session log /Users/jy/.proxmark3/log_20210610.txt
[=] Loading Preferences...
[+] loaded from JSON file /Users/jy/.proxmark3/preferences.json
[+] About to use the following files:
[+]    /usr/local/Cellar/proxmark3/4.9237/bin/../share/proxmark3/firmware/bootrom.elf
[+]    /usr/local/Cellar/proxmark3/4.9237/bin/../share/proxmark3/firmware/fullimage.elf
[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodemiceman1
 🕑  59 found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading ELF file /usr/local/Cellar/proxmark3/4.9237/bin/../share/proxmark3/firmware/bootrom.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d50->0x00000d50) [R X] @0x298

[+] Loading ELF file /usr/local/Cellar/proxmark3/4.9237/bin/../share/proxmark3/firmware/fullimage.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00102000 P 0x00102000 (0x00043430->0x00043430) [R X] @0x94
[+]    1: V 0x00200000 P 0x00145430 (0x00001314->0x00001314) [RW ] @0x434c4
[=] Note: Extending previous segment from 0x43430 to 0x44744 bytes

[+] Flashing...
[+] Writing segments for file: /usr/local/Cellar/proxmark3/4.9237/bin/../share/proxmark3/firmware/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. OK
[+]  0x00100200..0x00100f4f [0xd50 / 7 blocks]
....... OK

[+] Writing segments for file: /usr/local/Cellar/proxmark3/4.9237/bin/../share/proxmark3/firmware/fullimage.elf
[... 223 lines cut ...]
    lf hid clone -r 01400076000c86                       -> write raw value (HID Corporate 48 bit)
    lf hid clone -w H10301 --fc 118 --cn 1603            -> write raw value (HID 10301 26 bit)
    lf hid clone -w H10301 --fc 118 --cn 1603 --q5       -> HID 10301 26 bit, encode for Q5/T5555 tag
    lf hid clone -w H10301 --fc 118 --cn 1603 --em       -> HID 10301 26 bit, encode for EM4305/4469

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

usb] pm3 --> lf hid clone -w H10301 --fc 118 --cn 1603
[=] Preparing to clone HID tag
[+] [H10301] - HID H10301 26-bit;  FC: 118  CN: 1603    parity: valid
[=] Done

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.