RfidResearchGroup/proxmark3 #1325: Hitag eload segmentation fault

mekhalleh, 2021-06-20:
**Describe the bug**
The command: `lf hitag eload -f filename.bin` produce a segmentation fault

**To Reproduce**
The binary dump used:
[image]

The JSON format is:
```json
{
    "Created": "proxmark3",
    "FileType": "hitag",
    "Card": {
      "UID": "12345678"
    },
    "blocks": {
      "0": "12345678",
      "1": "4D494B52",
      "2": "20F04F4E",
      "3": "06AA4854",
      "4": "575F4F4B",
      "5": "55555555",
      "6": "AAAAAAAA",
      "7": "55555555",
      "8": "00000000",
      "9": "00000000",
      "10": "00000000",
      "11": "00000000"
    }
}
```

Steps to reproduce the behavior:
1. Go to pm3
2. Load dump `lf hitag eload -2 -f /home/mekhalleh/Desktop/hitag2/lf-hitag-12345678-dump.bin`

[image]

This crash is on the memcpy call, then, I have tested to modify by the following code.

[image]

[image]

However, the command `lf hitag sim -2` does not reflect the dump now.

[image]

Any suggestion?

**Desktop (please complete the following information):**
 - OS: ArchLinux
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/fix-hitag_eload/v4.9237-4101-g6db86...-unclean 2021-06-21 00:23:18
  compiled with GCC 11.1.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-4055-ga03e8cb25 2021-06-15 21:01:54
       os: RRG/Iceman/fix-hitag_eload/v4.9237-4101-g6db86...-unclean 2021-06-21 00:23:18
  compiled with GCC 11.1.0

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 60% used )
```

 - hw status
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40240
[#]   Available memory........ 40240
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD5690C23DF4EA02A
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] LF Sampling Stack
[#]   Max stack usage......... 3952 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 307712
[#]   Transfer Speed PM3 -> Client... 615424 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30353 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#] 
```

 - data tune
```
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 72.35 V - 125.00 kHz
[+] LF antenna: 38.41 V - 134.83 kHz
[+] LF optimal: 75.67 V - 126.32 kHz
[+] Approx. Q factor (*): 12.8 by frequency bandwidth measurement
[+] Approx. Q factor (*): 13.2 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 45.57 V - 13.56 MHz
[+] Approx. Q factor (*): 7.9 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

iceman1001, 2021-06-20:
`lf hitag eload`  isn't really implemented yet.   The `lf hitag *` commands need much more love.

iceman1001, 2021-06-20:
I pushed a fix for the crash,  but  `lf hitag simulation` doesn't work regardless.
There are quite a number of open issues in terms of hitag commands.

closed 2021-06-24 by iceman1001