RfidResearchGroup/proxmark3 #1326: lf em 4x50 brute not work

massi1990, 2021-06-21:
Hello,

I wish made a script for the brute force em4x50 function in order to find the password.

the function already exists but does not work :
[usb] pm3 --> lf em 4x50 brute --first 00000000 --last 00009999
[=] Trying 39322 passwords in range [0x00000000, 0x00009999]
[=] Estimated duration: 0h24m16s
[=] You can cancel this operation by pressing the pm3 button

[!] Communicating with Proxmark3 device failed

my wish is a sript which launches the command lf em 4x50 brute --first 00000000 --last 00000150 then increment the same function by +150 (lf em 4x50 brute --first 00000000 --last 00000300) until the password is found.

and save a log file.

thanks a lot for your help

iceman1001, 2021-06-21:
Looks like a auto-translated text,  so its a bit hard to follow your issue. 

`lf em 4x50 brute`  should work just fine.   However it looks like your Proxmark3 device doesn't understand it.
Which make me believe you have the wrong firmware flashed on your device.    Client / Firmware version _must_ match.

If you want a script that calls the brute command?   Its an open source project where you can modify the source code yourself and even contribute it by making a pull request (PR) to this repository.

massi1990, 2021-06-21:
[ CLIENT ]
  client: RRG/Iceman/master/v4.9237-2898-g2829e20d4 2021-01-21 11:30:20
  compiled with MinGW-w64 10.2.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-2898-g2829e20d4 2021-01-21 11:30:08
       os: RRG/Iceman/master/v4.9237-2898-g2829e20d4 2021-01-21 11:30:15
  compiled with GCC 8.4.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 276058 bytes (53%) Free: 248230 bytes (47%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2021-06-21:
and do you have a bug  or not?
If you have a bug please post details how to trigger it.

iceman1001, 2021-06-22:
There is a bug in the current `lf em 4x50` commands,   I will push out a fix soon.

iceman1001, 2021-06-22:
Pull latest master and `lf em 4x50 brute` should be working again

closed 2021-06-24 by iceman1001