RfidResearchGroup/proxmark3 #1336: mfp info crash

will-caruana, 2021-06-24:
**Describe the bug**
Segmentation fault (core dumped) when running hf mfp info

**To Reproduce**
Steps to reproduce the behavior:
1. Run hf mfp info
4. See error

**Expected behavior**
It wouldn't crash.

**Screenshots**
``` [usb] pm3 --> hf mfp info

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
Segmentation fault (core dumped)
```
**Desktop (please complete the following information):**
 - OS: 
 ``` Ubuntu 20.04.2 LTS (GNU/Linux 4.4.0-19041-Microsoft x86_64) ```

 - hw version
``` [ Proxmark3 RFID instrument ]                                                                                                                                                                                                                                                                                                                                                                             [ CLIENT ]                                                                                                                                                                                            client: RRG/Iceman/master/v4.9237-4134-g40f08108f 2021-06-23 19:13:01                                                                                                                                compiled with GCC 9.3.0 OS:Linux ARCH:x86_64                                                                                                                                                                                                                                                                                                                                                             [ PROXMARK3 ]                                                                                                                                                                                         device.................... RDV4                                                                                                                                                                      firmware.................. RDV4                                                                                                                                                                      external flash............ present                                                                                                                                                                   smartcard reader.......... present                                                                                                                                                                   FPC USART for BT add-on... absent                                                                                                                                                                                                                                                                                                                                                                        [ ARM ]                                                                                                                                                                                               bootrom: RRG/Iceman/master/v4.9237-4134-g40f08108f 2021-06-23 19:14:55                                                                                                                                    os: RRG/Iceman/master/v4.9237-4134-g40f08108f 2021-06-23 19:15:39                                                                                                                               compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]                                                                                                                                                                                                                                                                                                                            [ FPGA ]                                                                                                                                                                                              LF image built for 2s30vq100 on 2020-07-08 at 23:08:07                                                                                                                                               HF image built for 2s30vq100 on 2020-07-08 at 23:08:19                                                                                                                                               HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30                                                                                                                                                                                                                                                                                                                                            [ Hardware ]                                                                                                                                                                                          --= uC: AT91SAM7S512 Rev A                                                                                                                                                                           --= Embedded Processor: ARM7TDMI                                                                                                                                                                     --= Internal SRAM size: 64K bytes                                                                                                                                                                    --= Architecture identifier: AT91SAM7Sxx Series                                                                                                                                                      --= Embedded flash memory 512K bytes ( 60% used )         
 ```
 - hw status
``` [usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 42352
[#]   Available memory........ 42352
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 147
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A7BEA325
[#] Smart card module (ISO 7816)
[#]   version................. v3.11
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#] LF Sampling Stack
[#]   Max stack usage......... 4040 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[... 15 lines cut ...]
``` [usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕖   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 35.72 V - 125.00 kHz
[+] LF antenna: 28.90 V - 134.83 kHz
[+] LF optimal: 36.21 V - 126.32 kHz
[+] Approx. Q factor (*): 5.6 by frequency bandwidth measurement
[+] Approx. Q factor (*): 6.3 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 45.75 V - 13.56 MHz
[+] Approx. Q factor (*): 8.0 by peak voltage measurement
[+] HF antenna is OK ```

**Additional context**
hf search is able to see the tag. As seen below:
``` hf search
 🕕  Searching for ISO14443-A tag...
[+]  UID: CF F1 E4 54
[+] ATQA: 00 04
[+]  SAK: 20 [1] ```

iceman1001, 2021-06-24:
interesting,    I don't get a crash when running it against my mfp card.

will-caruana, 2021-06-24:
I just tried it on another mfp card I was able to get hf mfp info to work then going back and trying the original card I still get this crash.

merlokk, 2021-06-24:
I have tested several mines. no luck( all is ok

will-caruana, 2021-06-24:
Well seeing as I and others have been able to get it to work its clearly just the card I am using.

closed 2021-06-24 by will-caruana

iceman1001, 2021-06-24:
What is the  ```hf 14a info``` output?

will-caruana, 2021-06-24:
```
[usb] pm3 --> hf 14a info

[+]  UID: CF F1 E4 54
[+] ATQA: 00 04
[+]  SAK: 20 [1]
[+] Possible types:
[+]    MIFARE Plus EV1 2K/4K in SL3
[+]    MIFARE Plus S 2K/4K in SL3
[+]    MIFARE Plus X 2K/4K in SL3
[+]    MIFARE Plus SE 1K
[+]    NTAG 4xx
[=] -------------------------- ATS --------------------------
[+] ATS: 11 78 77 E1 03 00 53 4B 46 57 73 97 61 C0 00 90 00 [ 61 00 ]
[=]      11...............  TL    length is 17 bytes
[=]         78............  T0    TA1 is present, TB1 is present, TC1 is present, FSCI is 8 (FSC = 256)
[=]            77.........  TA1   different divisors are supported, DR: [2, 4, 8], DS: [2, 4, 8]
[=]               E1......  TB1   SFGI = 1 (SFGT = 8192/fc), FWI = 14 (FWT = 67108864/fc)
[=]                  03...  TC1   NAD is supported, CID is supported

[=] -------------------- Historical bytes --------------------
[+]    00534B4657739761C0009000
[?] Hint: try `hf mfp info`
[?] Hint: try `hf mfdes info` ```

iceman1001, 2021-06-24:
You sure its MFP and not NTAG4xx?

will-caruana, 2021-06-24:
Not sure I didn't think NTAG's had U2F support which this card should be able to do.
```
[=] -----------FIDO Info---------------------------------
[=] FIDO U2F authenticator detected. Version: U2F_V2
[!!] 🚨 FIDO2 version doesn't exist (6d00 - Instruction code not supported or invalid).
[usb] pm3 --> hf fido reg
[!!] 🚨 ERROR execute register command. APDU response status: 6700 - Wrong length
[usb] pm3 --> hf fido auth
-- hlen=0
[!!] 🚨 ERROR execute authentication command. APDU response status: 6a86 - Incorrect P1 or P2 parameter. ```

iceman1001, 2021-06-24:
It looks like it found something `U2F_V2`

merlokk, 2021-06-24:
`6d00 - Instruction code not supported or invalid` -- there is no fido2/u2f application

iceman1001, 2021-06-24:
In that case, this line is quite misleading
```
[=] FIDO U2F authenticator detected. Version: U2F_V2
```

merlokk, 2021-06-24:
yes, strange...
it has U2F but strange U2F

iceman1001, 2021-06-24:
Sounds like that one could need some tweaking.