RfidResearchGroup/proxmark3 #1340: The usb serial device is lost

MengJueM, 2021-06-27:
Hi I flashed the official "**master branch**" version of proxmark3, the serial usb device is working good in my Win10. Then I compiled the iceman's "**55e1b6714 Release v4.13441 - midsummer**", the flashing is good, but the device is recognized as unknown, what else should I do? thanks!

iceman1001, 2021-06-27:
Did you follow the instructions for your host OS and Proxmark3 device ?

F9Alejandro, 2021-06-27:
^^ Make sure to flash all as the bootloader is different from the proxmark3 official since this is the Iceman fork there is separate work being done.

Be sure to read the firmware flashing instructions thoroughly as there are some tricks to get the firmware and bootloader on the device if it didn't flash properly.

iceman1001, 2021-07-15:
Closing since no feedback from OP,   assuming it all worked out in the end.

closed 2021-07-15 by iceman1001