RfidResearchGroup/proxmark3 #1341: Hitag detection fail on Android - RDV2

ysard, 2021-06-29:
**Describe the bug**
Hitag detection works on computer but not on Android device.

**To Reproduce**
Steps to reproduce the behavior:

1. Use the last current commit of master branch (35ddebc) on both platforms.
2. Follow the compilation [procedure](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/termux_notes.md#optional-building-proxmark3-client-from-source) especially for Android.
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/termux_notes.md#enable-the-driver

Note: There is a typo error in the command of this doc:
`pkg install make clang clang++ readline libc++ git`
The `clang++` packages doesn't exist. In fact, `clang` package also installs clang++. 

3. `make SKIPBT=1 PLATFORM=PM3GENERIC`
4. The command `lf search` reports "No known 125/134 kHz tags found!" on Android when the same tag (not moved)
is detected on computer side. 
The command `lf hitag reader --26` returns nothing on Android when the tag uid is found on computer side.

Android example:
```
[usb] pm3 --> lf hitag reader --26
[usb] pm3 --> lf search
[=] Checking for known tags...
[=]
[-] ⛔ No known 125/134 kHz tags found!
[=] Couldn't identify a chipset
```
PC example:
```
[usb] pm3 --> lf search
[=] Checking for known tags...
[=] 
[+]  UID: 126f436d

[+] Valid Hitag found!

[usb] pm3 --> lf hitag reader --26
[+]  UID: 126f436d
```

**Expected behavior**
The Hitag tag should be detected on both platforms.

**Desktop**
 - OS:
   - GNU/Linux Debian based
   - Android MIUI 11.0.2 / Android 7.1.2
   - Proxmark RDV2
 - hw version PC side
```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.13441-4-g35ddebc-dirty-unclean 2021-06-28 21:38:52
  compiled with GCC 4.9.2 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.13441-4-g35ddebc03 2021-06-28 22:04:36
       os: RRG/Iceman/master/v4.13441-4-g35ddebc03 2021-06-29 00:29:33
  compiled with GCC 6.3.1 20170620

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
  HF image built for 2s30vq100 on 2020-07-08 at 23:08:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23:08:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 53% used )
```
- hw version Android side
```
[ CLIENT ]
client: RRG/Iceman/master/v4.13441-4-g35ddebc03 2021-06-29 01:54:08
compiled with Clang/LLVM Clang 12.0.0 OS:Android ARCH:aarch64
```

 - hw status PC side
```
[#] Memory
[#]   BigBuf_size............. 44224
[#]   Available memory........ 44224
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 23
[#] Current FPGA image
[#]   mode.................... LF image built for 2s30vq100 on 2020-07-08 at 23:08:07
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] LF Sampling Stack
[#]   Max stack usage......... 3944 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 274432
[#]   Transfer Speed PM3 -> Client... 548864 bytes/s
[#] Various
[#]   Max stack usage......... 4104 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30563 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
```
- hw status Android side:
Identical except the Transfert Speed PM3 -> Client which is 252928 bytes/s

 - data tune PC side:
```
[=] ---------- LF Antenna ----------
[+] LF antenna: 25,18 V - 125,00 kHz
[+] LF antenna: 20,07 V - 134,83 kHz
[+] LF optimal: 26,00 V - 126,32 kHz
[+] Approx. Q factor (*): 6,9 by frequency bandwidth measurement
[+] Approx. Q factor (*): 7,6 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 28,82 V - 13.56 MHz
[+] Approx. Q factor (*): 8,4 by peak voltage measurement
[+] HF antenna is OK
```
- data tune Android side
```
[=] ---------- LF Antenna ----------
[+] LF antenna: 20.07 V - 125.00 kHz
[+] LF antenna: 16.29 V - 134.83 kHz
[+] LF optimal: 20.72 V - 127.66 kHz
[+] Approx. Q factor (*): 6.6 by frequency bandwidth measurement
[+] Approx. Q factor (*): 6.0 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------              
[+] HF antenna: 20.11 V - 13.56 MHz               
[+] Approx. Q factor (*): 5.8 by peak voltage measurement                                           
[+] HF antenna is OK                              
(*) Q factor must be measured without tag on the antenna                                                                                              
[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz
```
**Additional context**
By the way, while writing the issue I realize that the antenna voltage levels are much lower on Android.
Could it just be my phone's inability to supply enough power through the USB OTG port? 

Don't focus on the "dirty" flag of the git id on the computer. It's an old platform and I made some changes (gcc/ld flags in Makefiles) to successfully compile the client.

Also, I did not follow the project for 2 years and what was achieved during this time is enormous. Many thanks for all of this.

iceman1001, 2021-06-29:
Interesting,   there is not differences in source for compilation on Android vs PC.  So no real software issue.
I would believe like you mentioned your android phone which doesn't manage to power the device properly but even that is inconclusive to your experienced issue.   Try power it with a battery and retest on android.

Glad you noticed the massive improvements to the repository.   It is truly a different experience today than two years ago.
Nothings says thank you as much as a donation :)

Clang++ doesn't exists ?  @doegox  , that one is yours :)

doegox, 2021-06-29:
clang++: done, thank you !

ysard, 2021-06-29:
Hi thanks for the quick response, indeed, I should have tested with the battery of the rdv2, and it works now, with voltage levels similar to those on the computer.

Procedure:
- Power on the proxmark on it's battery
- Connect it on the phone via USB
- Start the client

So it's a power supply issue that is not on the proxmark side.
I can't test on another phone or with another OTG cable for now, but it should be interesting to test on a USB-C phone instead of on a Micro-USB one.

iceman1001, 2021-06-29:
Ok,  closing

closed 2021-06-29 by iceman1001