RfidResearchGroup/proxmark3 #1345: [idea] add sha1sum to created fw

iceman1001, 2021-07-01:
**Is your feature request related to a problem? Please describe.**

A way to verify that the created firmware files are the same as when the client was compiled from.

**Describe the solution you'd like**
At firmware creatation,  we add a sha1sum file for each of those files.
We add those hashes to the client like we add "version" data.

at flashing time, we can compare that either the file has the same hash,  and we can also match that client was compiled at the same time.

**Additional context**
Consider this extra layer of checks as optional.  If all hashes match,  all is fine,   but when mismatch,  we can inform the user that the firmware file isnt the same as expected and either give them option to flash regardless.   Or interrupt and let user verify their setup.   ie: _not_ mandatory,  in order to be flexible user experience.

iceman1001, 2022-02-11:
If this post strike you as odd, its in response to a dialogue done at the discord server.   Bringing it here instead.

I figured we take sha1sums of the generated files when compilation is finished.  Of firmware files and the client exe.   Save it to a file.   And when flashing we let the flash fct calc sha1sums of all files and compare.    If mismatch we warn about it.
Does that make sense?
And it wouldn’t require a list of know hashes but hashes from your own build.

Regarding when:
After compilation, generate  a hash file for each file.  Now, when to call the check.  That would be in pm3 client where we trigger the flashing.   Since we use pm3 client now to flash.  Inside client/dev/proxmark3.c  if I remember it memory.   The the first thing that fct would to call  the verification fct.

Regarding algo:
For this purpose it’s not super important which hashing algo but we used sha256 for other files so….

iceman1001, 2022-02-11:
When to create the hash files?   I would suggest to do it as a _post_make_ item,  preferably inside Makefile call a bash/python create hashes script.

doegox, 2022-02-14:
I'm not sure to understand which use case it's supposed to solve.
A user compiled a firmware some time ago then update the repo, compiles *only* the client and tries to flash the old firmware?
For me, the most frequent error case would be: Someone compiles the repo but forgets to flash the device. In this case it's more up to the client to detect that the proxmark runs an old firmware.

doegox, 2022-02-14:
I pushed a branch sha, it detects in the client when the current proxmark3 image is not corresponding to the ARM sources at the time the client was compiled.

To test: 
- build all, flash proxmark
- run the client -> all ok
- edit an ARM source file
- build all or just the client
- run the client -> it will tell about the mismatch
[image]

To fulfill your initial need, you can make a similar check in the flasher by comparing `g_version_information.armsrc` with the one contained in the image to be flashed.

iceman1001, 2022-02-14:
Interesting,   I had the idea that focus on verifying the binaries,    not the source. 
1. generate all hashes for the fullimage, bootrom, client etc in their own sha files.
2 generate a file with the all sha256 hashes created at the build time.   Where this would allow the client to "know" that another firmware file is used instead of the one that were made when the client was compiled.

.1 solves the file integrity. 
2. solves the client vs firmware mismatch

this way you don't need the source code,  it also will work for the binary distributions like proxspace and package installs.

doegox, 2022-02-15:
I added the support for your use case in the https://github.com/RfidResearchGroup/proxmark3/tree/sha branch.
Now the fullimage.elf and the client both contain the same sha (made from the srcs), without need for a separate hashes file.
So for binary distributions of compiled client and compiled fullimage, this works too. (including for Kali which delivers all precompiled firmware variations)
So now you get alerted:
- when you try to flash a firmware not in sync with current client (still a todo: you get a warning but you are not yet prompted to say ok or abort)
- when you connect a proxmark3 running a firmware not in sync with current client
As a bonus, the firmware generation was quite simplified: https://github.com/RfidResearchGroup/proxmark3/commit/8baedb30ac041fde2e5a5f236c1f52e03c0bb7c6

iceman1001, 2022-02-15:
awesome!,

closed 2022-02-25 by iceman1001