RfidResearchGroup/proxmark3 #1363: [idea]

scott4290, 2021-07-10:
**Is your feature request related to a problem? Please describe.**
lf t55xx bruteforce has some room for speed improvement.

**Describe the solution you'd like**
I would like a new switch to control how many data samples to use in the CmdT55xxBruteForce -> t55xx_try_one_password -> AcquireData function.    The current default/static value is 12,000.  I've tested with several values and found that I can reduce this value to 2,500 and cut the password testing time by about 3 minutes for every 10,000 checked.  The default can stay at 12,000, but I would like to have this be a configurable option.

**Describe alternatives you've considered**
I've considered bypassing the getSignalProperties check that identifies bad sample data, but the speed improvement was insignificant.

**Additional context**

iceman1001, 2021-07-10:
it depends on antenna coil.   Basically we need two proper reads from the tag.
Now tags has large coils,  then we have keyfobs with smaller antennas,  then we have rings, implants with even smaller antennas.   All commands must work on all different antenna couplings.   Next thing is the difference in modulation where we need more samples for PSK/FSK than ASK because general decoding is hard. 
So by good approximation the current number was chosen.  

When it comes to wishes,   there is a wish list in the wiki where we keep track of ideas that someone hasn't commited to,   You find it here: https://github.com/RfidResearchGroup/proxmark3/wiki/Wishlist   

Before you start removing/ sidepass things,  try understand that t5577 cards can be in many different configurations and the commands needs to work all time.  The signal properties is there to detect if there is a signal or not, and if there is a signal, then how _strong_ is it.   All relevant to all decoding functions in next step.

The idea is to know when you have found a valid password.   

So when you want to improve the bruteforce,    you need to verify it against all possible valid configurations a t5577 can have,  and take in consideration all different couplings signal strengths with different antenna sizes.

With that said, I am looking forward to see your contributions.

iceman1001, 2021-12-12:
time to close?

scott4290, 2021-12-20:
@iceman1001 Good points, thank you.  We can close this.