RfidResearchGroup/proxmark3 #1384: [Bug] the standalone mode LF_THAREXDE build failed

AxisRay, 2021-07-25:
Makefile.platform:
’# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
# Run 'make PLATFORM=' to get an exhaustive list of possible parameters for this file.

PLATFORM=PM3RDV4
#PLATFORM=PM3GENERIC
# If you want more than one PLATFORM_EXTRAS option, separate them by spaces:
PLATFORM_EXTRAS=BTADDON
STANDALONE=LF_THAREXDE
‘
Build Error：
`c:/proxspace/msys2/mingw64/bin/../lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld.exe: obj/lf_tharexde.o:(.bss.gPassword+0x0): multiple definition of `gPassword'; obj/em4x50.o:(.bss.gPassword+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
make[1]: *** [Makefile:193: obj/fullimage.stage1.elf] Error 1
make: *** [Makefile:134: armsrc/all] Error 2`

gPassword was defined twice.
the line 73 of lf_tharexde.c should be deleted.
https://github.com/RfidResearchGroup/proxmark3/blob/1affdfce64872306e61a12aab8d2fea1aa2d8e2f/armsrc/Standalone/lf_tharexde.c#L73

iceman1001, 2021-07-25:
Can't replicate the bug on latest source.

iceman1001, 2021-07-25:
On Proxspace it doesn't compile. 

fix https://github.com/RfidResearchGroup/proxmark3/commit/9564863d955b10e378cadf9e448537d09e76e617

closed 2021-07-25 by iceman1001