RfidResearchGroup/proxmark3 #1391: macOS does not recognize device with FW 4.13441

tengelmeier, 2021-07-27:
**Describe the bug**
After flashing the latest firmware for an updated proxmark3 with homebrew, the device does not appear as /dev/tty.... on MacOS Big Sur  

**To Reproduce**
- put an Proxmark3 Easy in bootloader mode
- run  pm3-flash-all    

[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodemiceman1
[=] Available memory on this board: 512K bytes
[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading ELF file /usr/local/Cellar/proxmark3/4.13441/bin/../share/proxmark3/firmware/bootrom.elf
[...]
................... OK
[+] All done

**Expected behavior**
Working Proxmark when disconnecting and reconnecting on macOS 

**Real behavior:**
Console Log;
kernel	281389.217950 AppleUSB20HubPort@14443000: AppleUSBHostPort::createDevice: failed to create device (0xe00002bc)
kernel	281389.217989 AppleUSB20HubPort@14443000: AppleUSB20HubPort::resetAndCreateDevice: failed to create device, disabling port

**Desktop (please complete the following information):**
 - OS: 
 macOS Big Sur 14.5

 - inside proxmark3 client run the following commands and paste the output here.
 Does not work any more

iceman1001, 2021-07-27:
Did you compile the fw for PM3GENERIC to match your device?

tengelmeier, 2021-07-27:
```
brew uninstall proxmark3
brew install proxmark3 --with-generic
pm3-flash-all
```

cured it. Thanks

closed 2021-07-27 by tengelmeier