RfidResearchGroup/proxmark3 #1411: [Bug]14a anti-collision not working

AxisRay, 2021-08-03:
**Describe the bug**
Anti-collision not working, because the BBC in uid_resp is always 0x00
```
[#] BCC0 incorrect, got 0x00, expected 0xdc
[#] Aborting
[usb] pm3 --> trace list -t 14a
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 71 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2116 |       4484 | Tag |04  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10564 |      16388 | Tag |99  bf! 3f! ff  de                                                       |     | 
     164480 |     168032 | Rdr |93  30  99                                                               | !crc| SELECT_XXX
     169140 |     173876 | Tag |ba  16  e9  dc                                                           |     | 
```
as the trace , tag have replied the remaining UID ba16e9  with BCC dc
But the pm3 print error log : BCC0 incorrect, got 0x00, expected 0xdc

**To Reproduce**
Steps to reproduce the behavior:
1. there are two mifare classic card with UID 99BA16E9 and 19B72F9F
2. put the two cards  on the pm3
3. execute `hf 14a info`
4. console output:
```
[#] BCC0 incorrect, got 0x00, expected 0xdc
[#] Aborting
```

**Expected behavior**
A clear and concise description of what you expected to happen.
```
[usb] pm3 --> hf 14a info

[+]  UID: 59 B7 2F 9F
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Magic capabilities : Gen 1a
[+] Prng detection: weak
[#] Auth error
[?] Hint: try `hf mf` commands
```
**Additional context**
Add any other context about the problem here.

AxisRay, 2021-08-03:
the collision position detecting is unstable.
first try:
```
[usb] pm3 --> hf 14a list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 59 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2116 |       4484 | Tag |04  00                                                                   |     |
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10564 |      16452 | Tag |59  bf! 2f  9f  5e                                                       |     |
      18176 |      22240 | Rdr |93  34  59  0f!                                                          | !crc| SELECT_XXX
```
Second try:
```
[usb] pm3 --> hf 14a list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 60 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2116 |       4484 | Tag |04  00                                                                   |     |
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10564 |      16452 | Tag |59  b7  3f! 9f  5e                                                       |     |
      18432 |      23776 | Rdr |93  45  59  b7  1f                                                       | !crc| SELECT_XXX
```

iceman1001, 2021-12-12:
I believe we only support _bytewise_ bit collision,   not the proper _bitwise_  one.  

A proper fix would to implement the needed bitwise solution.

iceman1001, 2022-02-04:
closing due to inactivity

closed 2022-02-04 by iceman1001