RfidResearchGroup/proxmark3 #1426: [BUG] hf mfu dump return cmd error:00

AxisRay, 2021-08-05:
try to dump tag of the xiaomi air purifier filter with `hf mfu dump`
the key is `bc77b6ed` (sniff)
cmd returned `Cmd Error: 00` 
```
[fpc] pm3 --> hf mfu dump -k bc77b6ed
[+] TYPE: NTAG UNKNOWN  
[+] Reading tag memory...
[#] Cmd Error: 00
[#] Read block 4 error
[=] MFU dump file information
[=] -------------------------------------------------------------
[=]       Version | 00 53 04 02 01 00 0F 03 
[=]         TBD 0 | 00 00 
[=]         TBD 1 | 00 
[=]     Signature | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[=]     Counter 0 | 00 00 00 
[=]     Tearing 0 | 00 
[=]     Counter 1 | 00 00 00 
[=]     Tearing 1 | 00 
[=]     Counter 2 | 00 00 00 
[=]     Tearing 2 | 00 
[=] Max data page | 2 (12 bytes)
[=]   Header size | 56
[=] -------------------------------------------------------------
[=] block#   | data        |lck| ascii
[=] ---------+-------------+---+------
[=]   0/0x00 | 53 5C 4D CA |   | S\M.
[=]   1/0x01 | EE 02 1F 00 |   | ....
[=]   2/0x02 | F3 48 00 00 |   | .H..
[=]   3/0x03 | E1 10 12 00 | 0 | ....
[=] ---------------------------------
[=] Using UID as filename
[+] saved 72 bytes to binary file hf-mfu-535C4DEE021F00-dump-8.bin
[+] saved to json file hf-mfu-535C4DEE021F00-dump-8.json
[!] ⚠️  Partial dump created. (4 of 42 blocks)
```
the trace shows that the pm3 didn't send READ BLOCK cmd
```
[fpc] pm3 --> trace list -t 14a
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 193 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
       2116 |       4484 | Tag |44  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10564 |      16388 | Tag |88  53  5c  4d  ca                                                       |     | 
      19200 |      29728 | Rdr |93  70  88  53  5c  4d  ca  3a  40                                       |  ok | SELECT_UID
      30788 |      34308 | Tag |04  da  17                                                               |     | 
      35712 |      38176 | Rdr |95  20                                                                   |     | ANTICOLL-2
      39236 |      45060 | Tag |ee  02  1f  00  f3                                                       |     | 
      47744 |      58272 | Rdr |95  70  ee  02  1f  00  f3  3f  f8                                       |  ok | SELECT_UID-2
      59332 |      62916 | Tag |00  fe  51                                                               |     | 
    3512448 |    3517216 | Rdr |3c  00  a2  01                                                           |  ok | READ SIG
    3518276 |    3557572 | Tag |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     | 
            |            |     |00  00  00  00  00  00  00  00  00  00  00  00  00  00  20  da           |  ok | 

```

but the cmd `hf mfu rdbl -b 4 -k bc77b6ed` works well
```
[fpc] pm3 --> hf mfu rdbl -b 4 -k bc77b6ed
[=] Using pwd BC 77 B6 ED 

[=] Block#  | Data        | Ascii
[=] -----------------------------
[=] 04/0x04 | 00 00 48 4B | ..HK
```

iceman1001, 2021-08-05:
looks like you had a bad coupling with the tag.  The tag is identified as NTAG Unknown.
try finding a better position and see if you get a different response.

AxisRay, 2021-08-05:
> looks like you had a bad coupling with the tag. The tag is identified as NTAG Unknown.
> try finding a better position and see if you get a different response.

I have tried.
but not working...

I am trying to figure it out with adding some debug log.

iceman1001, 2021-08-05:
The GetVersion data looks odd..
`00 53 04 02 01 00 0F 03` 

Something is going on.
Can you do a sniff of the tag and purifier?   save `trace save -f`  a file and share it here

AxisRay, 2021-08-06:
I did it before, but not saved as a file.
I will sniff again and save as a file later.
```
[fpc] pm3 --> trace list -t 14a
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 1073 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       1056 | Rdr |26(7)                                                                    |     | REQA
       2244 |       4612 | Tag |44  00                                                                   |     | 
     419552 |     422016 | Rdr |93  20                                                                   |     | ANTICOLL
     423204 |     429028 | Tag |88  53  5c  4d  ca                                                       |     | 
     839136 |     849664 | Rdr |93  70  88  53  5c  4d  ca  3a  40                                       |  ok | SELECT_UID
     850852 |     854372 | Tag |04  da  17                                                               |     | 
    1285856 |    1288320 | Rdr |95  20                                                                   |     | ANTICOLL-2
    1289524 |    1295348 | Tag |ee  02  1f  00  f3                                                       |     | 
    1705568 |    1716096 | Rdr |95  70  ee  02  1f  00  f3  3f  f8                                       |  ok | SELECT_UID-2
    1717300 |    1720884 | Tag |00  fe  51                                                               |     | 
    2111712 |    2119936 | Rdr |1b  bc  77  b6  ed  da  5b                                               |  ok | PWD-AUTH KEY: 0xbc77b6ed
    2121124 |    2125796 | Tag |00  00  a0  1e                                                           |     | 
    2517728 |    2522432 | Rdr |30  04  26  ee                                                           |  ok | READBLOCK(4)
    2523700 |    2544564 | Tag |00  00  48  4b  00  00  31  33  00  21  03  05  00  06  14  77  f0  dc   |  ok | 
    2937216 |    2941920 | Rdr |30  05  af  ff                                                           |  ok | READBLOCK(5)
    2943172 |    2964036 | Tag |00  00  31  33  00  21  03  05  00  06  14  77  d1  18  1a  00  d3  f4   |  ok | 
    3356592 |    3361360 | Rdr |30  06  34  cd                                                           |  ok | READBLOCK(6)
    3362548 |    3383412 | Tag |00  21  03  05  00  06  14  77  d1  18  1a  00  00  00  00  00  39  38   |  ok | 
    3776096 |    3780864 | Rdr |30  07  bd  dc                                                           |  ok | READBLOCK(7)
    3782068 |    3802932 | Tag |00  06  14  77  d1  18  1a  00  00  00  00  00  00  00  00  00  21  07   |  ok | 
    4195680 |    4200384 | Rdr |30  08  4a  24                                                           |  ok | READBLOCK(8)
    4201636 |    4222500 | Tag |d1  18  1a  00  00  00  00  00  00  00  00  00  00  00  00  00  4e  e3   |  ok | 
  167133552 |  167134608 | Rdr |26(7)                                                                    |     | REQA
  167135796 |  167138164 | Tag |44  00                                                                   |     | 
  167552960 |  167555424 | Rdr |93  20                                                                   |     | ANTICOLL
  167556628 |  167562452 | Tag |88  53  5c  4d  ca                                                       |     | 
  167972112 |  167982640 | Rdr |93  70  88  53  5c  4d  ca  3a  40                                       |  ok | SELECT_UID
  167983844 |  167987364 | Tag |04  da  17                                                               |     | 
  168418576 |  168421040 | Rdr |95  20                                                                   |     | ANTICOLL-2
  168422244 |  168428068 | Tag |ee  02  1f  00  f3                                                       |     | 
  168838000 |  168848528 | Rdr |95  70  ee  02  1f  00  f3  3f  f8                                       |  ok | SELECT_UID-2
  168849716 |  168853300 | Tag |00  fe  51                                                               |     | 
  169243760 |  169251984 | Rdr |1b  bc  77  b6  ed  da  5b                                               |  ok | PWD-AUTH KEY: 0xbc77b6ed
  169253188 |  169257860 | Tag |00  00  a0  1e                                                           |     | 
  169649664 |  169654368 | Rdr |30  04  26  ee                                                           |  ok | READBLOCK(4)
  169655620 |  169676484 | Tag |00  00  48  4b  00  00  31  33  00  21  03  05  00  06  14  77  f0  dc   |  ok | 
  170069264 |  170073968 | Rdr |30  05  af  ff                                                           |  ok | READBLOCK(5)
  170075220 |  170096084 | Tag |00  00  31  37! 00  21  03  05  00  06  14  77  d1  18  1a  00  d3  f4   | !crc| 
  170488640 |  170493408 | Rdr |30  06  34  cd                                                           |  ok | READBLOCK(6)
  170494612 |  170515476 | Tag |00  21  03  05  00  06  14  77  d1  18  1a  30  00  00  00  00  39  38   | !crc| 
[... cut ...]
  311874224 |  311878992 | Rdr |30  06  34  cd                                                           |  ok | READBLOCK(6)
  311880180 |  311901044 | Tag |00  21  03  05  00  06  14  77  d1  18  1a  00  00  00  00  00  39  38   |  ok | 
  312293264 |  312298032 | Rdr |30  07  bd  dc                                                           |  ok | READBLOCK(7)
  312712160 |  312716864 | Rdr |30  08  4a  24                                                           |  ok | READBLOCK(8)
  312718116 |  312738980 | Tag |d1  18  1a  00  00  00  00  00  00  00  00  00  00  00  00  00  4e  e3   |  ok | 
```

AxisRay, 2021-08-06:
```
[fpc] pm3 --> trace list -t 14a -f
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 359 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       1056 | Rdr |26(7)                                                                    |     | REQA
       1056 |       2244 |     |fdt (Frame Delay Time): 1188
       2244 |       4612 | Tag |44  00                                                                   |     | 
     419200 |     421664 | Rdr |93  20                                                                   |     | ANTICOLL
     421664 |     422852 |     |fdt (Frame Delay Time): 1188
     422852 |     428676 | Tag |88  53  5c  4d  ca                                                       |     | 
     838400 |     848928 | Rdr |93  70  88  53  5c  4d  ca  3a  40                                       |  ok | SELECT_UID
     848928 |     850116 |     |fdt (Frame Delay Time): 1188
     850116 |     853636 | Tag |04  da  17                                                               |     | 
    1284656 |    1287120 | Rdr |95  20                                                                   |     | ANTICOLL-2
    1287120 |    1288308 |     |fdt (Frame Delay Time): 1188
    1288308 |    1294132 | Tag |ee  02  1f  00  f3                                                       |     | 
    1703840 |    1714368 | Rdr |95  70  ee  02  1f  00  f3  3f  f8                                       |  ok | SELECT_UID-2
    1714368 |    1715556 |     |fdt (Frame Delay Time): 1188
    1715556 |    1719140 | Tag |00  fe  51                                                               |     | 
    2109536 |    2117760 | Rdr |1b  bc  77  b6  ed  da  5b                                               |  ok | PWD-AUTH KEY: 0xbc77b6ed
    2515200 |    2519904 | Rdr |30  04  26  ee                                                           |  ok | READBLOCK(4)
    2519904 |    2521156 |     |fdt (Frame Delay Time): 1252
    2521156 |    2542020 | Tag |00  00  48  4b  00  00  31  33  00  21  03  05  00  06  14  77  f0  dc   |  ok | 
    2934400 |    2939104 | Rdr |30  05  af  ff                                                           |  ok | READBLOCK(5)
    2939104 |    2940356 |     |fdt (Frame Delay Time): 1252
    2940356 |    2961220 | Tag |00  00  31  33  00  21  03  05  00  06  14  77  33  72  00  00  f9  67   |  ok | 
    3353600 |    3358368 | Rdr |30  06  34  cd                                                           |  ok | READBLOCK(6)
    3358368 |    3359556 |     |fdt (Frame Delay Time): 1188
    3359556 |    3380420 | Tag |00  21  03  05  00  06  14  77  33  72  00  00  00  00  00  00  d9  0d   |  ok | 
    3772768 |    3777536 | Rdr |30  07  bd  dc                                                           |  ok | READBLOCK(7)
    3777536 |    3778724 |     |fdt (Frame Delay Time): 1188
    3778724 |    3799524 | Tag |00  06  14  77  33  72  00  00  00  00  00  00  00  00  00  00  b8  0c   |  ok | 
    4191968 |    4196672 | Rdr |30  08  4a  24                                                           |  ok | READBLOCK(8)
    4196672 |    4197940 |     |fdt (Frame Delay Time): 1268
    4197940 |    4218740 | Tag |33  72  00  00  00  00  00  00  00  00  00  00  00  00  00  00  44  dd   |  ok | 

```
haha, so `-f` not means "save as a file"
it means "show frame delay times"

AxisRay, 2021-08-06:
[image]
the version seems okay 
just the vendor ID  is different
it is 53 (Shanghai Feiju Microelectronics Co. Ltd. China) instead of 04 (NXP Semiconductors)

AxisRay, 2021-08-06:
```
                if (memcmp(version, "\x00\x04\x03\x01\x01\x00\x0B", 7) == 0)      { tagtype = UL_EV1_48; break; }
                else if (memcmp(version, "\x00\x04\x03\x01\x02\x00\x0B", 7) == 0) { tagtype = UL_NANO_40; break; }
                else if (memcmp(version, "\x00\x04\x03\x02\x01\x00\x0B", 7) == 0) { tagtype = UL_EV1_48; break; }
                else if (memcmp(version, "\x00\x04\x03\x01\x01\x00\x0E", 7) == 0) { tagtype = UL_EV1_128; break; }
                else if (memcmp(version, "\x00\x04\x03\x02\x01\x00\x0E", 7) == 0) { tagtype = UL_EV1_128; break; }
                else if (memcmp(version, "\x00\x34\x21\x01\x01\x00\x0E", 7) == 0) { tagtype = UL_EV1_128; break; } // Mikron JSC Russia EV1 41 pages tag
                else if (memcmp(version, "\x00\x04\x04\x01\x01\x00\x0B", 7) == 0) { tagtype = NTAG_210; break; }
                else if (memcmp(version, "\x00\x04\x04\x01\x02\x00\x0B", 7) == 0) { tagtype = NTAG_210u; break; }
                else if (memcmp(version, "\x00\x04\x04\x01\x01\x00\x0E", 7) == 0) { tagtype = NTAG_212; break; }
                else if (memcmp(version, "\x00\x04\x04\x02\x01\x00\x0F", 7) == 0) { tagtype = NTAG_213; break; }
                else if (memcmp(version, "\x00\x04\x04\x02\x01\x01\x0F", 7) == 0) { tagtype = NTAG_213_C; break; }
                else if (memcmp(version, "\x00\x04\x04\x02\x01\x00\x11", 7) == 0) { tagtype = NTAG_215; break; }
                else if (memcmp(version, "\x00\x04\x04\x02\x01\x00\x13", 7) == 0) { tagtype = NTAG_216; break; }
                else if (memcmp(version, "\x00\x04\x04\x04\x01\x00\x0F", 7) == 0) { tagtype = NTAG_213_F; break; }
                else if (memcmp(version, "\x00\x04\x04\x04\x01\x00\x13", 7) == 0) { tagtype = NTAG_216_F; break; }
                else if (memcmp(version, "\x00\x04\x04\x02\x03\x00\x0F", 7) == 0) { tagtype = NTAG_213_TT; break; }
                else if (memcmp(version, "\x00\x04\x04\x05\x02\x01\x13", 7) == 0) { tagtype = NTAG_I2C_1K; break; }
                else if (memcmp(version, "\x00\x04\x04\x05\x02\x01\x15", 7) == 0) { tagtype = NTAG_I2C_2K; break; }
                else if (memcmp(version, "\x00\x04\x04\x05\x02\x02\x13", 7) == 0) { tagtype = NTAG_I2C_1K_PLUS; break; }
                else if (memcmp(version, "\x00\x04\x04\x05\x02\x02\x15", 7) == 0) { tagtype = NTAG_I2C_2K_PLUS; break; }
                else if (version[2] == 0x04) { tagtype = NTAG; break; }
                else if (version[2] == 0x03) { tagtype = UL_EV1; }
                break;
```
Looks like some improvements can be made

AxisRay, 2021-08-06:
I have add support for this tag,
but cmd error still there.

iceman1001, 2021-08-06:
yeah, now you have the max blocks etc..  to identify this fudan version use correct block limits....

iceman1001, 2021-08-08:
time to close?

AxisRay, 2021-08-08:
Yeah, all problems were solved.
Thanks!

closed 2021-08-08 by AxisRay