RfidResearchGroup/proxmark3 #1438: [BUG] The uid of the simulated card is 00000000 , which is simulated by hf_mfcsim

AxisRay, 2021-08-09:
```
[usb] pm3 --> hf 14a info
[#] ----------- BREAKING ----------
[#] Emulator stopped. Tracing: 1  trace length: 0 
[#] Simulation end!

[+]  UID: C1 F0 6A 3E 
[+] ATQA: 00 04
[+]  SAK: 28 [1]
[+] Possible types:
[+]    SmartMX with MIFARE Classic 1K
[=] -------------------------- ATS --------------------------
[+] ATS: 10 78 80 A0 02 20 90 00 00 00 00 00 C1 F0 6A 3E [ 43 00 ]
[=]      10...............  TL    length is 16 bytes
[=]         78............  T0    TA1 is present, TB1 is present, TC1 is present, FSCI is 8 (FSC = 256)
[=]            80.........  TA1   different divisors are NOT supported, DR: [], DS: []
[=]               A0......  TB1   SFGI = 0 (SFGT = (not needed) 0/fc), FWI = 10 (FWT = 4194304/fc)
[=]                  02...  TC1   NAD is NOT supported, CID is supported

[=] -------------------- Historical bytes --------------------
[+]    20900000000000C1F06A3E
[+] Prng detection: weak
[#] Auth error
[?] Hint: try `hf mf` commands

[#] 
[#] Stand-alone mode, no computer necessary
[#] Standalone mode MFCSIM started!
[#] Emulator memory initialized
[#] Buffer cleared (40504 bytes)
[#] Found dump file hf_mfcsim_dump.bin
[#] File size is 1024
[#] Read card data from input file
[#] Uploaded to emulator memory
[#] Emulator memory filled, simulation ready to start.
[#] Press button to abort simulation at anytime.
[#] Simulation start!
[#] Enforcing Mifare 1K ATQA/SAK
[#] 4B UID: 00000000
[#] ATQA  : 00 04
[#] SAK   : 08
[#] ----------- BREAKING ----------
[#] Emulator stopped. Tracing: 1  trace length: 0 
[#] Simulation end!
[#] 
[#] Stand-alone mode, no computer necessary
[#] Standalone mode MFCSIM started!
[#] Emulator memory initialized
[#] Buffer cleared (40504 bytes)
[#] Found dump file hf_mfcsim_dump.bin
[#] File size is 1024
[#] Read card data from input file
[#] Uploaded to emulator memory
[#] Emulator memory filled, simulation ready to start.
[#] Press button to abort simulation at anytime.
[#] Simulation start!
[#] Enforcing Mifare 1K ATQA/SAK
[#] 4B UID: 99ba16e9
[#] ATQA  : 00 04
[#] SAK   : 08
```
As the record above,
the problem will arise when some commands are executed before entering standalone mode
looks like the memory was modified by something.

AxisRay, 2021-08-09:
Need further analysis

AxisRay, 2021-08-09:
```
[offline] pm3 --> hw connect
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC

[#] 
[#] Stand-alone mode, no computer necessary
[#] Standalone mode MFCSIM started!
[#] Emulator memory initialized
[#] Buffer cleared (40504 bytes)
[#] Found dump file hf_mfcsim_dump.bin
[#] File size is 1024
[#] Read card data from input file
[#] Uploaded to emulator memory
[#] Emulator memory filled, simulation ready to start.
[#] Press button to abort simulation at anytime.
[#] Simulation start!
[#] ERROR: Invalid dump. UID/SAK/ATQA not found
[#] Simulation end!
[#] 
[#] Stand-alone mode, no computer necessary
[#] Standalone mode MFCSIM started!
[#] Emulator memory initialized
[#] Buffer cleared (40504 bytes)
[#] Found dump file hf_mfcsim_dump.bin
[#] File size is 1024
[#] Read card data from input file
[#] Uploaded to emulator memory
[#] Emulator memory filled, simulation ready to start.
[#] Press button to abort simulation at anytime.
[#] Simulation start!
[#] Enforcing Mifare 1K ATQA/SAK
[#] 4B UID: 99ba16e9
[#] ATQA  : 00 04
[#] SAK   : 08
```
After power on, without execute any command (except `hw connect`).

AxisRay, 2021-08-09:
damn it
my fault……
I should use `BigBuf_free_keep_EM()` instead of BigBuf_free()
it freed everything include emulator memory!

iceman1001, 2021-08-09:
time to close?

doegox, 2021-08-09:
I guess so

closed 2021-08-09 by doegox