RfidResearchGroup/proxmark3 #1467: [BUG] if you try to execute trace list -t 14a before pressing button to stop sniff, the pm3 device will be stuck

AxisRay, 2021-08-20:
1. Execute `hf 14a sniff` to sniff
```
[usb] pm3 --> hf 14a sniff

[#] Starting to sniff. Press PM3 Button to stop.
```
2. Don't press button to stop. Just execute `trace list -t 14a`  and you will see timeout error.
```
[usb] pm3 --> trace list -t 14a
[=] downloading tracelog data from device
[=] Waiting for a response from the Proxmark3...
[=] You can cancel this operation by pressing the pm3 button
[-] ⛔ Timed out while trying to download data from device
[!] ⚠️  timeout while waiting for reply.
[+] Recorded activity (trace len = 0 bytes)
```
3. Try again, you will see double free error.
```
[usb] pm3 --> trace list -t 14a
free(): double free detected in tcache 2
./pm3: line 249: 288500 Aborted                 (core dumped) $CLIENT "$@"
```
4. the pm3 device stuck, you must cut the power and reset.
```
# ./pm3
[=] Session log /root/.proxmark3/logs/log_20210820.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[!!] 🚨 ERROR: cannot communicate with the Proxmark

# ./pm3
[=] Session log /root/.proxmark3/logs/log_20210820.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
```

doegox, 2021-08-20:
It's by design, please follow the instructions.
When proxmark is busy sniffing, it can't check USB without losing sniffing frames.

closed 2021-08-20 by doegox

AxisRay, 2021-08-20:
But
After step3
Even if you press the button to stop sniffing, you still cannot connect the device.

iceman1001, 2021-08-21:
we have no good way of knowing when the device is busy in a loop,  especially if its a thight loop like sniffing where we don't check for usb commands.
Either we can "lock the client" from entering any commands during all commands but that makes it a bit less flexible.
or we can change all usb commands to enforce that the pm3 device always answers back with a status,  that we then would need to keep track off in the client.

If you find a simple solution for getting double free,  from not knowing if the device is busy or not,  feel free to contribute

PR #1469 (2021-08-21): Fix double free error #1467
Fixes  #1467 
Setting g_trace to NULL after free
to fix double free error

Before:
```
[fpc] pm3 --> hf 14a sniff

[#] Starting to sniff. Press PM3 Button to stop.
[fpc] pm3 --> trace list -t 14a
[=] downloading tracelog data from device
[=] Waiting for a response from the Proxmark3...
[=] You can cancel this operation by pressing the pm3 button
[-] ⛔ Timed out while trying to download data from device
[!] ⚠️  timeout while waiting for reply.
[+] Recorded activity (trace len = 0 bytes)
[fpc] pm3 --> 
[fpc] pm3 --> 
[fpc] pm3 --> trace list -t 14a
=================== [... cut]