RfidResearchGroup/proxmark3 #1485: Failed to build on macOS 12

lsylx, 2021-08-25:
***Compilation problems***
ld: warning: pointer not aligned at address 0x10025E9C2 (_mfu_ident_table + 258 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E9BA (_mfu_ident_table + 250 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E9B2 (_mfu_ident_table + 242 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E99A (_mfu_ident_table + 218 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E95E (_mfu_ident_table + 158 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E956 (_mfu_ident_table + 150 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E94E (_mfu_ident_table + 142 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E946 (_mfu_ident_table + 134 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E93E (_mfu_ident_table + 126 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E936 (_mfu_ident_table + 118 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8FA (_mfu_ident_table + 58 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8F2 (_mfu_ident_table + 50 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8EA (_mfu_ident_table + 42 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8E2 (_mfu_ident_table + 34 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8DA (_mfu_ident_table + 26 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8D2 (_mfu_ident_table + 18 from obj/cmdhfmfu.o)
ld: unaligned pointer(s) for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2

**Describe the bug**
When I tried to build this on my Mac I got errors like this above
I don't know what happened and how to fix this problem
All things works fine on macOS 11
but it didn't work on macOS 12

doegox, 2021-08-25:
which version of clang ?
can you compile with make V=1 and share the log ?

lsylx, 2021-08-25:
> which version of clang ?
> can you compile with make V=1 and share the log ?

===================================================================
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C client all DESTDIR=
===================================================================
Client platform:   Darwin
GUI support:       QT5 found, enabled ()
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 not found, disabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
compiler version:  Apple clang version 13.0.0 (clang-1300.0.29.3)
===================================================================
[*] MAKE deps/amiitool/libamiibo.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/amiitool all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/cliparser/libcliparser.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/cliparser all
PLATFORM Darwin
make[2]: Nothing to be done for `all'.
[*] MAKE deps/hardnested/libhardnested.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/hardnested all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/liblua/liblua.a for macosx
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/liblua macosx
/Library/Developer/CommandLineTools/usr/bin/make all SYSCFLAGS="-DLUA_USE_MACOSX"
make[3]: Nothing to be done for `all'.
[*] MAKE deps/jansson/libjansson.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/jansson all
make[2]: Nothing to be done for `all'.
[*] MAKE obj/libmbedtls.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ../common/mbedtls OBJDIR=../../client/obj BINDIR=../../client/obj all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/reveng/libreveng.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/reveng all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/tinycbor/tinycbor.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/tinycbor all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/whereami/libwhereami.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/whereami all
make[2]: Nothing to be done for `all'.
[=] LD proxmark3
g++  -framework Foundation -framework AppKit obj/aiddesfire.o obj/aidsearch.o obj/cmdanalyse.o obj/cmdcrc.o obj/cmddata.o obj/cmdflashmem.o obj/cmdflashmemspiffs.o obj/cmdhf.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf15.o obj/cmdhfcryptorf.o obj/cmdhfepa.o obj/cmdhfemrtd.o obj/cmdhffelica.o obj/cmdhffido.o obj/cmdhfcipurse.o obj/cmdhficlass.o obj/cmdhflegic.o obj/cmdhfjooki.o obj/cmdhflist.o obj/cmdhflto.o obj/cmdhfmf.o obj/cmdhfmfdes.o obj/cmdhfmfhard.o obj/cmdhfmfu.o obj/cmdhfmfp.o obj/cmdhfseos.o obj/cmdhfst.o obj/cmdhfst25ta.o obj/cmdhfthinfilm.o obj/cmdhftopaz.o obj/cmdhfwaveshare.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfawid.o obj/cmdlfcotag.o obj/cmdlfdestron.o obj/cmdlfem.o obj/cmdlfem410x.o obj/cmdlfem4x05.o obj/cmdlfem4x50.o obj/cmdlfem4x70.o obj/cmdlffdxb.o obj/cmdlfguard.o obj/cmdlfgallagher.o obj/cmdlfhid.o obj/cmdlfhitag.o obj/cmdlfidteck.o obj/cmdlfindala.o obj/cmdlfio.o obj/cmdlfjablotron.o obj/cmdlfkeri.o obj/cmdlfmotorola.o obj/cmdlfnedap.o obj/cmdlfnexwatch.o obj/cmdlfnoralsy.o obj/cmdlfpac.o obj/cmdlfparadox.o obj/cmdlfpcf7931.o obj/cmdlfpresco.o obj/cmdlfpyramid.o obj/cmdlfsecurakey.o obj/cmdlft55xx.o obj/cmdlfti.o obj/cmdlfviking.o obj/cmdlfvisa2000.o obj/cmdmain.o obj/cmdnfc.o obj/cmdparser.o obj/cmdscript.o obj/cmdsmartcard.o obj/cmdtrace.o obj/cmdusart.o obj/cmdwiegand.o obj/comms.o obj/crypto/asn1dump.o obj/crypto/asn1utils.o obj/crypto/libpcrypto.o obj/emv/cmdemv.o obj/emv/crypto.o obj/emv/crypto_polarssl.o obj/emv/dol.o obj/emv/emv_pk.o obj/emv/emv_pki.o obj/emv/emv_pki_priv.o obj/emv/emv_roca.o obj/emv/emv_tags.o obj/emv/emvcore.o obj/emv/emvjson.o obj/emv/tlv.o obj/emv/test/crypto_test.o obj/emv/test/cryptotest.o obj/emv/test/cda_test.o obj/emv/test/dda_test.o obj/emv/test/sda_test.o obj/fido/additional_ca.o obj/fido/cose.o obj/fido/cbortools.o obj/fido/fidocore.o obj/cipurse/cipursecore.o obj/cipurse/cipursecrypto.o obj/cipurse/cipursetest.o obj/fileutils.o obj/flash.o obj/generator.o obj/graph.o obj/jansson_path.o obj/iso7816/apduinfo.o obj/iso7816/iso7816core.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/elite_crack.o obj/loclass/ikeys.o obj/mifare/desfire_crypto.o obj/mifare/desfirecrypto.o obj/mifare/desfirecore.o obj/mifare/desfiresecurechan.o obj/mifare/desfiretest.o obj/mifare/mad.o obj/mifare/mfkey.o obj/mifare/mifare4.o obj/mifare/mifaredefault.o obj/mifare/mifarehost.o obj/nfc/ndef.o obj/pm3.o obj/pm3_binlib.o obj/pm3_bitlib.o obj/pref [... cut ...]
ld: warning: pointer not aligned at address 0x10025E8EA (_mfu_ident_table + 42 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8E2 (_mfu_ident_table + 34 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8DA (_mfu_ident_table + 26 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10025E8D2 (_mfu_ident_table + 18 from obj/cmdhfmfu.o)
ld: unaligned pointer(s) for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2

doegox, 2021-08-25:
Hmm ok clang13.
I tried clang13 on my Debian, I pushed 2 fixes unrelated but I don't get this linker error, still investigating...

doegox, 2021-08-25:
You didn't compile the latest proxmark3 code, which commit was it? The latest release ? Please try the latest comit.

lsylx, 2021-08-25:
> You didn't compile the latest proxmark3 code, which commit was it? The latest release ? Please try the latest commit.

===================================================================
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C client all DESTDIR=
===================================================================
Client platform:   Darwin
GUI support:       QT5 found, enabled ()
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 not found, disabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
compiler version:  Apple clang version 13.0.0 (clang-1300.0.29.3)
===================================================================
[*] MAKE deps/amiitool/libamiibo.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/amiitool all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/cliparser/libcliparser.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/cliparser all
PLATFORM Darwin
make[2]: Nothing to be done for `all'.
[*] MAKE deps/hardnested/libhardnested.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/hardnested all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/liblua/liblua.a for macosx
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/liblua macosx
/Library/Developer/CommandLineTools/usr/bin/make all SYSCFLAGS="-DLUA_USE_MACOSX"
make[3]: Nothing to be done for `all'.
[*] MAKE deps/jansson/libjansson.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/jansson all
make[2]: Nothing to be done for `all'.
[*] MAKE obj/libmbedtls.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ../common/mbedtls OBJDIR=../../client/obj/mbedtls BINDIR=../../client/obj all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/reveng/libreveng.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/reveng all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/tinycbor/tinycbor.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/tinycbor all
make[2]: Nothing to be done for `all'.
[*] MAKE deps/whereami/libwhereami.a
/Library/Developer/CommandLineTools/usr/bin/make --no-print-directory -C ./deps/whereami all
make[2]: Nothing to be done for `all'.
[=] LD proxmark3
g++  -framework Foundation -framework AppKit obj/mifare/aiddesfire.o obj/aidsearch.o obj/cmdanalyse.o obj/cmdcrc.o obj/cmddata.o obj/cmdflashmem.o obj/cmdflashmemspiffs.o obj/cmdhf.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf15.o obj/cmdhfcryptorf.o obj/cmdhfepa.o obj/cmdhfemrtd.o obj/cmdhffelica.o obj/cmdhffido.o obj/cmdhfcipurse.o obj/cmdhficlass.o obj/cmdhflegic.o obj/cmdhfjooki.o obj/cmdhflist.o obj/cmdhflto.o obj/cmdhfmf.o obj/cmdhfmfdes.o obj/cmdhfmfhard.o obj/cmdhfmfu.o obj/cmdhfmfp.o obj/cmdhfseos.o obj/cmdhfst.o obj/cmdhfst25ta.o obj/cmdhfthinfilm.o obj/cmdhftopaz.o obj/cmdhfwaveshare.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfawid.o obj/cmdlfcotag.o obj/cmdlfdestron.o obj/cmdlfem.o obj/cmdlfem410x.o obj/cmdlfem4x05.o obj/cmdlfem4x50.o obj/cmdlfem4x70.o obj/cmdlffdxb.o obj/cmdlfguard.o obj/cmdlfgallagher.o obj/cmdlfhid.o obj/cmdlfhitag.o obj/cmdlfidteck.o obj/cmdlfindala.o obj/cmdlfio.o obj/cmdlfjablotron.o obj/cmdlfkeri.o obj/cmdlfmotorola.o obj/cmdlfnedap.o obj/cmdlfnexwatch.o obj/cmdlfnoralsy.o obj/cmdlfpac.o obj/cmdlfparadox.o obj/cmdlfpcf7931.o obj/cmdlfpresco.o obj/cmdlfpyramid.o obj/cmdlfsecurakey.o obj/cmdlft55xx.o obj/cmdlfti.o obj/cmdlfviking.o obj/cmdlfvisa2000.o obj/cmdmain.o obj/cmdnfc.o obj/cmdparser.o obj/cmdscript.o obj/cmdsmartcard.o obj/cmdtrace.o obj/cmdusart.o obj/cmdwiegand.o obj/comms.o obj/crypto/asn1dump.o obj/crypto/asn1utils.o obj/crypto/libpcrypto.o obj/emv/cmdemv.o obj/emv/crypto.o obj/emv/crypto_polarssl.o obj/emv/dol.o obj/emv/emv_pk.o obj/emv/emv_pki.o obj/emv/emv_pki_priv.o obj/emv/emv_roca.o obj/emv/emv_tags.o obj/emv/emvcore.o obj/emv/emvjson.o obj/emv/tlv.o obj/emv/test/crypto_test.o obj/emv/test/cryptotest.o obj/emv/test/cda_test.o obj/emv/test/dda_test.o obj/emv/test/sda_test.o obj/fido/additional_ca.o obj/fido/cose.o obj/fido/cbortools.o obj/fido/fidocore.o obj/cipurse/cipursecore.o obj/cipurse/cipursecrypto.o obj/cipurse/cipursetest.o obj/fileutils.o obj/flash.o obj/generator.o obj/graph.o obj/jansson_path.o obj/iso7816/apduinfo.o obj/iso7816/iso7816core.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/elite_crack.o obj/loclass/ikeys.o obj/mifare/lrpcrypto.o obj/mifare/desfirecrypto.o obj/mifare/desfirecore.o obj/mifare/desfiresecurechan.o obj/mifare/desfiretest.o obj/mifare/mad.o obj/mifare/mfkey.o obj/mifare/mifare4.o obj/mifare/mifaredefault.o obj/mifare/mif [... cut ...]
ld: warning: pointer not aligned at address 0x10026691A (_mfu_ident_table + 34 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x100266912 (_mfu_ident_table + 26 from obj/cmdhfmfu.o)
ld: warning: pointer not aligned at address 0x10026690A (_mfu_ident_table + 18 from obj/cmdhfmfu.o)
ld: unaligned pointer(s) for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2

This one is the latest proxmark3 code
The problem still exists.

doegox, 2021-08-25:
please try again, I pushed a fix

lsylx, 2021-08-25:
> please try again, I pushed a fix

Problem fixed. Compiled successfully

doegox, 2021-08-25:
great, thanks for your cooperation

closed 2021-08-25 by doegox

berikv, 2021-11-29:
I'm trying to fix an error in a totally different project, but with the same title "ld: unaligned pointer(s) for architecture arm64".

To make it easier to find what has fixed this error it would be helpful to mention the commit that fixed it.
It looks like these commits are the ones fixing the error:

ba7f1b93b986a6b1c975258977209d2869bbdaf9
2b3cd6e604f821b7f7a5696e081456ace019ab70

alfonsrv, 2022-04-23:
Compilation still doesn't work on macOS Monterey with M1 Macs – unfortunately the Homebrew tap also seems broken.

<details>

<summary>make log</summary>

```
user@NullX proxmark3-master % make clean && make -j

===================================================================
Version info:      Iceman/master/release (git)
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
===================================================================
Version info:      Iceman/master/release (git)
Client platform:   Darwin
GUI support:       QT5 found, enabled (Qt version 6.2.3 in /usr/local/lib)
native BT support: Bluez not found, disabled
Jansson library:   system library found
Lua library:       system library not found, using local library
Python3 library:   Python3 not found, disabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
compiler version:  Apple clang version 13.1.6 (clang-1316.0.21.2)
===================================================================
PLATFORM Darwin
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE mf_nonce_brute/clean
[*] MAKE mfd_aes_brute/clean

===================================================================
Version info:      Iceman/master/release (git)
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
[*] MAKE mfkey/all
[*] MAKE nonce2key/all
[*] MAKE mf_nonce_brute/all
[*] MAKE mfd_aes_brute/all
[-] CC fpga_compress.c
[-] CC ../../common/lz4/lz4hc.c
[-] CC ../../common/lz4/lz4.c
[-] CC brute_key.c
[-] CC nonce2key.c
[-] CC mf_nonce_brute.c
[-] CC mfkey32.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crypto1.c
[... 1176 lines cut ...]
      _fPrintAndLog in ui.o
  "_rl_replace_line", referenced from:
      _fPrintAndLog in ui.o
  "_rl_reset_screen_size", referenced from:
      _print_progress in ui.o
  "_rl_restore_prompt", referenced from:
      _fPrintAndLog in ui.o
  "_rl_save_prompt", referenced from:
      _fPrintAndLog in ui.o
  "_rl_set_prompt", referenced from:
      _pm3line_update_prompt in pm3line.o
  "_rl_set_signals", referenced from:
      _pm3line_install_signals in pm3line.o
  "_using_history", referenced from:
      _pm3line_init in pm3line.o
  "_write_history", referenced from:
      _sigint_handler in pm3line.o
      _pm3line_flush_history in pm3line.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2
```

</details>

iceman1001, 2022-04-24:
CLANG 13...
seems to be annoyed with the OPENSSL ,  readline,  cbor , QT...
did you setup the devenv correct?

alfonsrv, 2022-04-24:
I installed the dependencies as outlined [here](https://github.com/Proxmark/proxmark3/wiki/MacOS) – however I worked around this by using the `brew tap`, which somehow failed due to an M1-specific OS upgrade error that's solved after rebooting.

CalfCrusher, 2023-09-16:
I tried everything but i'm still unable to compile on OSX M1 getting **ld: unaligned pointer(s) for architecture arm64**